esphome/issues #4561: TM1637 Documentation

stephenhouser, 2023-06-05:
### The problem

In the [documentation for the TMS1637](https://esphome.io/components/display/tm1637.html) is the section on connecting multiple displays.

"Connect multiple displays
To connect multiple TM1636 displays you need as many control lines as you have displays. This is achived by sharing control-lines for clk and dio between displays."...

1. It reads *TMS1636* here but should be TMS1637.
2. I can find no reference for the wiring shown in the diagrams for connecting multiple displays in this manner. If this is possible, there should be a corresponding configuration shown (yaml). All other references use a shared CLK and individual DIO lines for each display.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
display:
  - platform: tm1637
    id: right_display
    clk_pin: GPIO14
    dio_pin: GPIO27
    length: 4
    lambda: |-
      it.printf(0, "%4.1d", id(step_count));
  - platform: tm1637
    id: left_display
    clk_pin: GPIO14
    dio_pin: GPIO12
    lambda: |-
      it.printf(0, "%4.1d", id(step_count));
```

gitmarecki, 2023-09-13:
I was only able to force my multi-display setup to work thanks to your comment. And it worked without any issues. You just daisy-chain them as per the instruction in the [original post](https://esphome.io/components/display/tm1637.html), but you made a mistake in your code - 1st CLK pin becomes second's DIO pin, second's DIO is the third CLK and the third one's DIO loops back to the first's CLK pin. In your example you bridged CLKs together, which won't work. You only need two pins for two devices. Three for three and so on... 

My setup looks like this (ignore the GPIO numbers and look at D0, D1, D2 parameters - I am using ESP8266 here so numbers might be different on yours)

```yaml
display:
  - platform: tm1637 # RED FIRST
    clk_pin: GPIO16 #D0
    dio_pin: GPIO5 #D1
    intensity: 0
    update_interval: 500ms
    lambda: |-
# here goes the display function for the first node
  - platform: tm1637 # SECOND
    clk_pin: GPIO5 #D1
    dio_pin: GPIO4 #D2
    intensity: 3
    update_interval: 500ms
    lambda: |-
# here goes the display function for the second node
  - platform: tm1637 # THIRD
    clk_pin: GPIO4 #D2
    dio_pin: GPIO0 #D3
    intensity: 7
    update_interval: 500ms
    lambda: |-
# here goes the display function for the third node
  - platform: tm1637 # FOURTH
    clk_pin: GPIO0 #D3
    dio_pin: GPIO16 #D0
    intensity: 0
    update_interval: 500ms
    lambda: |-
# here goes the display function for the last node
```

stephenhouser, 2023-09-21:
I'll have to try that on my next project. The current one is now on a PCB and in operation. I don't really need 5 more boards :-)

I still did not find any reference to that wiring anywhere other than the ESPHome docs. So I remain curious where it came from.