esphome/issues #4563: Yaml '!secret' syntax does not match Home Assistant: Fails to climb up dir tree

disconn3ct, 2023-06-06:
### The problem

The first tip in the FAQ says ESPHome supports most of the yaml directives, including '!secret'. The HA documentation [says](https://www.home-assistant.io/docs/configuration/secrets/#debugging-secrets) secrets.yaml is checked for in current-file's folder, then parent folders until it reaches the main configuration file.

ESPHome only searches the current-file's folder.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

yaml

### Example YAML snippet

```yaml
secrets.yaml: (NOT in 'common')

devicename: potato
```
device.yaml: 
```
packages:
  core: !include common/core.yaml
```
common/core.yaml:
```
esphome:
  name: !secret devicename
```
```

### Anything in the logs that might be useful for us?

```txt
ERROR Error while reading config: Invalid YAML syntax:

Secret 'variable_here' not defined
  in "common/file.yaml", line 3, column 18:
      uuid: !secret variable_here
```
variable_here is defined in the secrets.yaml with the device file, not in `common/secrets.yaml`.
```

### Additional information

This may be a documentation bug (missing warning)

ssieb, 2023-06-07:
It supports the directive, but not the exact functionality.  The FAQ entry specifically says "within the directory where the configuration file is".

disconn3ct, 2023-06-07:
From the report:
> This may be a documentation bug (missing warning)

The docs state "this is supported, so you can use it in this way". That is not the same as "it supports **part of** this, so you can use it **only** in this way."

The rest of the answer is about **enhancements** above and beyond what HA supports. Limitations should be listed also.