esphome/issues #4564: Can't install firmware via web interface / How to work with the esphome repository directly

Andersama, 2023-06-06:
### The problem

I appear to be suffering the same problem as people dealing with #4525 and another issue of the same variety. The serial port appears, but instantly returns with an error. I know this isn't particularly helpful for debugging.

So instead I've attempted to use the https://github.com/esphome/esphome. Since it appears the repository is using platform.io I figured I'd just install the firmware that way. Which appears to have worked. However I have no idea what I'm doing with the repository as the documentation appears to all deal with having installed the firmware from the web interface.

Currently looking at the serial output I'm getting a repeating chain of "Found networks: ", followed by a "No matching network found!" which I'm guessing is because I'm meant to write the configuration file somewhere with my wifi credentials and where the web interface likely would've done that for me and put it somewhere the firmware knows to access I'm not sure where that is, or what the file would be called.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.5.4

### What platform are you using?

ESP32

### Board

nodemcu-32s

closed 2023-06-07 by ssieb

Andersama, 2023-06-07:
After another hour of playing around, I tried the docker install process but on a windows 10 machine. That worked, not exactly smoothly but at least getting past the serial error everything now seems to be fine.

ssieb, 2023-06-07:
Did you try looking at the documentation?
https://esphome.io/guides/getting_started_command_line.html

ssieb, 2023-06-07:
But as I mentioned before, this is not the place for support questions.  I will not continue answering here.

Andersama, 2023-06-09:
@ssieb yes, that was the page I was following along with which did not work on my linux machine, but did work on windows. I'm not going to waste time trying to debug whatever was going on, I'm not sure it's related to this project, it could easily a problem with the browser or some permissions problem.

If the original project developer takes a look here, I'd suggest turning that dummy_main application into what `https://web.esphome.io` flashes for "prepare for first use"*

ssieb, 2023-06-09:
No, you're just doing it completely wrong...

Andersama, 2023-06-12:
Lets be crystal clear here I'm not going to join some random discord to bother people who're busy with their hobby projects in hopes a developer comes around to someone having an already known open issue just mysteriously solve it out of the blue. Not what I expected to happen. At best I was hoping the author might pop by and explain what exactly the repo was for/how it was set up, at worst I let them know there's yet another person with that same issue.

You're not trying to be helpful anyway, you clearly didn't want to be helpful to start and it's not like your responses were exactly probing to find out what may be going wrong. The original issue I'm sure is a lot more helpful to most. But even the limited amount of effort I put in to even try this on a different operating system is a lot more than you're putting in, and I know what I wrote here isn't going to be resolving the issue. If you're one of the members handling this repo, you don't have to "help" just because. I get that, I've got a repo myself which has people pop up every once in a while expecting me to be able to magically fix things. I don't tell them they're doing it "wrong" because guess what, computers are an absolute mess and shit just doesn't always work.