esphome/issues #4565: ESPHome container stuck on "Starting dashboard web server"

kidzblvck, 2023-06-08:
### The problem

I'm not sure if this was caused by an update or just something random, but my ESPHome container fails to start, the only log after starting the container is:
`esphome    | 2023-06-08 00:47:40,480 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir /config...`

It has been working without any issues for the last year or so.

This is my docker compose:
`version: '3'
services:
  esphome:
    container_name: esphome
    image: ghcr.io/esphome/esphome:latest
    volumes:
      - /home/user/docker/esphome/config:/config
      - /etc/localtime:/etc/localtime:ro
    restart: always
    privileged: true
    network_mode: host
~`

### Which version of ESPHome has the issue?

latest from docker hub

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

ssieb, 2023-06-08:
That is the only log line.  The dashboard is started.

zibous, 2023-06-08:
Try this:

```bash
#!/bin/bash

# ---------------------------------------------
# bash esphome.sh
# see: https://hub.docker.com/r/esphome/esphome
# ----------------------------------------------

CURRENTURL=http://$(hostname)
DOCKER_TIMEZONE=Europe/Berlin
DOCKERIMAGE=esphome/esphome:latest
CONTAINERLABEL=esphome
APPSDATA="/home/user/docker/esphome/config"

echo "Try to remove previuos installation ${CONTAINERLABEL}..."
docker stop ${DOCKERIMAGE} && docker rm ${DOCKERIMAGE}

echo "Install Docker container ${CONTAINERLABEL}."
docker run --detach \
  --name ${CONTAINERLABEL} \
  --privileged=true \
  --env TZ=${DOCKER_TIMEZONE} \
  --restart unless-stopped \
  --volume /run/udev:/run/udev \
  --volume /etc/localtime:/etc/localtime:ro \
  --volume ${APPSDATA}/config:/config \
  --network host \
  ${DOCKERIMAGE}

echo "Docker container ${CONTAINERLABEL} ready."
echo "Run ESPHOME Dashboard: ${CURRENTURL}:6052"
```

closed 2023-06-08 by kidzblvck