esphome/issues #4567: ESP32 RMT LED strip component does not support setting RGB color/brightness

khenderick, 2023-06-08:
### The problem

It's not possible to set RGB colors/brightness on the ESP32 RMT LED strip component.

### Which version of ESPHome has the issue?

2022.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

esp32_rmt_led_strip

### Example YAML snippet

```yaml
light:
  - platform: esp32_rmt_led_strip
    name: "Light"
    id: bathroom_light
    rgb_order: GRB
    is_rgbw: true
    pin: GPIO21
    num_leds: 60
    rmt_channel: 0
    chipset: SK6812
```
```

### Anything in the logs that might be useful for us?

```txt
[22:03:38][I][app:102]: ESPHome version 2023.5.5 compiled on Jun  8 2023, 22:02:42
...
[22:03:38][C][esp32_rmt_led_strip:170]: ESP32 RMT LED Strip:
[22:03:38][C][esp32_rmt_led_strip:171]:   Pin: 21
[22:03:38][C][esp32_rmt_led_strip:172]:   Channel: 0
[22:03:38][C][esp32_rmt_led_strip:197]:   RGB Order: GRB
[22:03:38][C][esp32_rmt_led_strip:198]:   Max refresh rate: 0
[22:03:38][C][esp32_rmt_led_strip:199]:   Number of LEDs: 60
[22:03:38][C][light:103]: Light 'Light'
[22:03:38][C][light:105]:   Default Transition Length: 1.0s
[22:03:38][C][light:106]:   Gamma Correct: 2.80
...
[22:04:03][W][light:167]: 'Light' - This color mode does not support setting RGB brightness!
[22:04:03][W][light:175]: 'Light' - This color mode does not support setting RGB color!
[22:04:03][D][light:035]: 'Light' Setting:
[22:04:03][D][light:062]:   White: 0%
[22:04:03][D][light:084]:   Transition length: 1.0s
...
```
```

### Additional information

Relates to https://github.com/esphome/issues/issues/4536 & https://github.com/esphome/esphome/pull/4886

ssieb, 2023-06-08:
What are you doing to trigger that message?

khenderick, 2023-06-09:
I open up the light in HA, turn it on, and then click on the colour pallette icon to open up below interface;

[image]

Then I increase the RGB brightness (yields both error messages), and select a color (also yields both error messages)

[image]