esphome/issues #4568: mlx90614 Report wrong Object temperature and Ambient=nan°C

mishomvp, 2023-06-08:
### The problem

Hello,

As soon as I saw that esphome supports mlx90614 I decided to test one such sensor that I had available. Everything was working and you are showing real temperature values.
But after I updated esphome to 2023.5.5 the sensor now shows negative temperature ([D][mlx90614:112]: Got Temperature=-111.2°C Ambient=nan°C).
I also checked in the history back and just when I updated to 2023.5.5 on May 29th the readings of this sensor are different.
I don't see any sensor reading errors in the logs.
Thanks in advance!

### Which version of ESPHome has the issue?

ESPHome version 2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.0

### What platform are you using?

ESP32

### Board

esp32dev 

### Example YAML snippet

```yaml
i2c:
  sda: 21
  scl: 22
  frequency: 100kHz
sensor:

- platform: mlx90614
    ambient:
      name: Ambient
    object:
      name: Object
    update_interval: 120s
```

commentsonly, 2023-06-09:
strange, works here. I just updated my ESPhome (which I didn't since 2023.3.2 I think and found the MLX90614 being supported (unbelievable I was thinking it'll never happen within the next 10 years since it's not a year of speach component. hahaha

Ok, I just modified my code (includes a .h files + libraries: spo, wire, Adafruit BusIO and Adafruit MLX90614 lib), eg. commented these lines and compiled the ESP code to see if I could validate that problem.
And it simply works, means both solutions show the absolute same temperatures.
I do use the MLX pointing up into the sky to get an idea about the cloud coverage (clouds give warmer object temps but a clear sky). And before I used the new native MLX support it was 25,5°C on the rooftop (ambient) and 4,5°C object (sky). After recompiling with the native implementation it was 25,4°C rooftop and 4,5°C sky .. so it works as designed I'd say.

@mishomvp if you fancy I could let you have the non-native code snippet so you could do a cross check, since if that also fails on your side the problem must be located somewhere else but not within the MLX90614 ESPhome implementation.

Or you do another check with the actual 2023.6.1 which is what I use right now. Perhaps you issue got fixed?

mishomvp, 2023-06-09:
Hello,

And my idea is the same. To measure the temperature of the sky and monitor whether the weather is cloudy or clear.
I did some more experiments today. It turned out that after I did: "Clean build files".
And now everything is fine!
[mlx90614:112]: Got Temperature=23.6°C Ambient=23.9°C
It also turned out that if I change the configurations and do not "Clean build files" again it shows strange values. I can't figure out the dependency yet, but I'll experiment.
It works for now.
We will see after ESPHOME version 2023.06 is released if there will be a change.
I also tried with different speeds (10,50 and 100khz) on the i2c bus and it all works.

commentsonly, 2023-06-10:
cleaning build files is always a good idea. I wished for it would auto happen each time a compile run would succeed. But I bet it won't happen with respect to the ones still believing a Pi is a computer where reality showing differnt it's slower but an old mobile phone.
2 things for your project: try to get an MLX version with a smaller detection angle, works more realiable on greater distances like pointing to the sky.
Try to calibrate the object temperature at clear blue sky with information for example given from windy.com which allows to define the differnt height for the information you get. Means you could get not only temparature information not only for the ground but also for 1000m and more.
My lambda for the cloud coverage in % goes like this: 20 is something which needs adjustment. a=Tambient, o=Tobject and the check < 0 is simply because I wanted to disply the coverage from 0 ... 100% in an apexchart card in HA.
```
  lambda: |-
    if ((1 - ((id(a).state - id(o).state) / 20)) < 0) {
      return 0;
      }
    else {
      return (1 - ((id(a).state - id(o).state) / 20)) * 100;
    }
```

But nice to hear it works at your end

mishomvp, 2023-06-14:
Hello!
Thanks for all the guidance. I currently have the MLX90614ESF purchased. But I need to make a box and buy some quartz glass or something similar that lets in the full spectrum of light and protects the sensor from the weather. Looks like JGS3 glass is perfect for this. Can you share your experience ;)

commentsonly, 2023-06-14:
Get yourself the Datasheet from Melexis (google is your friend) @ 10.2 they do describe the FOV (field of view) which is a good explanation about the differnt angles the various versions are designed for.
Means an MLX90614ESF-DCI might be ideal.
For more conversation about the usecases better write a pm on the community forum to my alias 'justone' since these questions are surely not to be disgussed here on github, since not being part with your initial problem.

closed 2023-10-28 by github-actions[bot]

khenderick, 2023-12-02:
I have an MLX90614 as well, and I have the same issue:

```
[10:08:21][D][mlx90614:112]: Got Temperature=-124.2°C Ambient=nan°C
[10:08:21][D][sensor:094]: 'IR object temperature': Sending state -124.18999 °C with 2 decimals of accuracy
```

I can fix this by just re-installing until it starts reporting a correct value again, but it might be suddenly broken again when there's an ESPHome update.

Do I have to create a new ticket, or can/will this be re-opened?

andrelung, 2024-01-18:
Same here  and also got some observations:
- the temperature does change between -40 and -170,9 when I move hand in front of it (while my room is at ~20°C)
- I got an aliexpress version called GY-906
- running the latest esphome (2023.12.7)
- cleaning build cache resolves the issue, too

kj4dxk, 2024-03-20:
Same thing happening here with ESPhome 2024.3.1

kj4dxk, 2024-03-20:
Where do you see the option to Clean Build Files?  I am not seeing that, so my sensor is still not working.

mishomvp, 2024-03-20:
Check this: 

[image]

Skons, 2024-03-20:
Clean build files didn't work for me, it has something to do in the way it comes back after a reboot. I just created this script so it's fairly new. It already made the sensor work again after it showed `[D][mlx90614:112]: Got Temperature=-115.2°C Ambient=nan°C`, but it took about 5 reboots.

```yaml
sensor:
  - platform: mlx90614
    ambient:
      name: Weatherstation Ambient
      id: Ambient
    object:
      name: Weatherstation Object
      id: Object
    update_interval: 5s

globals:
- id: nan_detected
  type: int
  initial_value: "0"

switch:
  - platform: restart
    id: weatherstationrestart
    name: "Weatherstation Restart"

interval:
  - interval: 10s
    then:
      - script.execute: ambient_nan_detector

script:
  - id: ambient_nan_detector
    then:
      - lambda: |-
          if (isnan(id(Ambient).raw_state)) {
            ESP_LOGD("main", "NAN detected in sensor Ambient");
            id(nan_detected) = (id(nan_detected) + 1);
          }
          if (id(nan_detected) > 4) {
            ESP_LOGD("main", "Ambient is NAN for over 30 seconds, reboot");
            id(weatherstationrestart).turn_on(); 
          }
```

Skons, 2024-06-23:
> cleaning build files is always a good idea. I wished for it would auto happen each time a compile run would succeed. But I bet it won't happen with respect to the ones still believing a Pi is a [...]

To have only 1 calculation, i did this:
```
lambda: |-
      float result = (1 - ((id(Ambient).state - id(Object).state) / 20));
      if (result < 0) {
        return 0;
      }
      else {
        result = result * 100;
        if (result > 100) {
          return 100;
        }
        else {
          return result;
        }
      }
```

Can you elaborate on the 20? How did you came to that number?