esphome/issues #4573: MEMS microphone (INMP441) throws interrupt errors on ESP32-CAM

Jpsy, 2023-06-11:
### The problem

My MEMS microphone (INMP441) throws interrupt errors on ESP32-CAM MCUs while working normally on a standard ESP32 MCU (MH-ET ESP32 MiniKIT). The error is:
```
[17:33:16][D][voice_assistant:105]: Requesting start...
[17:33:16][D][voice_assistant:085]: Starting...
[17:33:16][V][esp-idf:000]: E (127282) I2S: i2s_dma_intr_init(538): Register I2S Interrupt error
[17:33:16][V][esp-idf:000]: E (127284) I2S: i2s_driver_install(1950): I2S interrupt initailze failed
```
It happens at the very moment, I start the recording for Voice Assist and is immediatelly followed by a crash & reboot of the MCU.

On ESP32-CAM the available GPIOs are quite limited due to the many GPIOs used by the camera. But I tried many possible variations of available GPIOs and they all fail. The GPIOs I tried are: 2, 12, 13, 14, 15, 1 (TX), 3 (RX) in arbitrary combinations. They all fail in the same way.

The exactly same ESPHome YAML config works on a normal ESP32 (with adapted GPIOs):

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.1

### What platform are you using?

ESP32

### Board

ESP32-CAM (AI Thinker Clone)

### Component causing the issue

i2s_audio

### Example YAML snippet

```yaml
# switch off logging through RX/TX so we can use them as GPIOs
logger:
  baud_rate: 0

# MEMS microphone INMP441
i2s_audio:
  i2s_lrclk_pin: GPIO14  # LRCLK, WS, FS
  i2s_bclk_pin: GPIO13   # BLCK, SCK
microphone:
  - platform: i2s_audio
    id: mic_i2s
    i2s_din_pin: GPIO2  # DIN, SDIN, SD, SDATA, ADCDATA
    adc_type: external
    pdm: false     # quite sure that INMP441 is NOT PDM!
    channel: left  # open or gnd on L/R input of INMP441 => left, high level => right
voice_assistant:
  microphone: mic_i2s

binary_sensor:    
  - platform: gpio
    pin: 
      number: GPIO12
      inverted: true
      mode:
        input: true
        pullup: true
    name: Talk Switch
    internal: true
    on_press:
      - voice_assistant.start:
    on_release:
      - voice_assistant.stop:
```

### Anything in the logs that might be useful for us?

```txt
[17:33:16][D][voice_assistant:105]: Requesting start...
[17:33:16][D][voice_assistant:085]: Starting...
[17:33:16][V][esp-idf:000]: E (127282) I2S: i2s_dma_intr_init(538): Register I2S Interrupt error
[17:33:16][V][esp-idf:000]: E (127284) I2S: i2s_driver_install(1950): I2S interrupt initailze failed
```

closed 2023-10-24 by github-actions[bot]

Jpsy, 2023-10-24:
This issue is still valid.

haldi4803, 2023-12-06:
Yeah... i got the exact same [issue](https://community.home-assistant.io/t/working-esphome-config-for-ttgo-esp32-camera-board-with-microphone/126231/21?u=haldi) on TTGO-Camera V1.6.2 with Mic.

Start push to Talk, mic doesn't start. i2s interrupt, reboots.

```
[04:04:48][D][binary_sensor:036]: 'ttgocam2 Button': Sending state ON
[04:04:48][D][voice_assistant:422]: State changed from IDLE to START_PIPELINE
[04:04:48][D][voice_assistant:428]: Desired state set to START_MICROPHONE
[04:04:48][D][voice_assistant:124]: microphone not running
[04:04:48][D][voice_assistant:206]: Requesting start...
[04:04:48][D][voice_assistant:422]: State changed from START_PIPELINE to STARTING_PIPELINE
[04:04:48][D][voice_assistant:443]: Client started, streaming microphone
[04:04:48][D][voice_assistant:422]: State changed from STARTING_PIPELINE to START_MICROPHONE
[04:04:48][D][voice_assistant:428]: Desired state set to STREAMING_MICROPHONE
[04:04:48][D][voice_assistant:159]: Starting Microphone
[04:04:48][D][voice_assistant:422]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[04:04:48][D][esp-idf:000]: E (107291) I2S: i2s_dma_intr_init(538): Register I2S Interrupt error

[04:04:48][D][esp-idf:000]: E (107303) I2S: i2s_driver_install(1950): I2S interrupt initailze failed

WARNING esphome-web-454f70: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esphome-web-454f70
WARNING Disconnected from API
INFO Successfully connected to esphome-web-454f70 in 0.004s
```

haldi4803, 2023-12-06:
Crap... i think this guy here is on to something!
https://github.com/espressif/esp-idf/issues/10994

because if i delete the Camera part from the config.yaml and reflash that it's ONLY Microphone, then it works!

```
[04:45:29][I][app:102]: ESPHome version 2023.11.6 compiled on Dec  6 2023, 04:44:29
[04:45:29][C][wifi:559]: WiFi:
[04:45:29][C][wifi:391]:   Local MAC: 08:3A:F2:45:4F:70
[04:45:29][C][wifi:396]:   SSID: [redacted]
[04:45:29][C][wifi:397]:   IP Address: 192.168.1.236
[04:45:29][C][wifi:399]:   BSSID: [redacted]
[04:45:29][C][wifi:400]:   Hostname: 'esphome-web-454f70'
[04:45:29][C][wifi:402]:   Signal strength: -59 dB ▂▄▆█
[04:45:29][C][wifi:406]:   Channel: 12
[04:45:29][C][wifi:407]:   Subnet: 255.255.255.0
[04:45:29][C][wifi:408]:   Gateway: 192.168.1.1
[04:45:29][C][wifi:409]:   DNS1: 192.168.1.1
[04:45:29][C][wifi:410]:   DNS2: 0.0.0.0
[04:45:29][C][logger:416]: Logger:
[04:45:29][C][logger:417]:   Level: DEBUG
[04:45:29][C][logger:418]:   Log Baud Rate: 115200
[04:45:29][C][logger:420]:   Hardware UART: UART0
[04:45:29][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'ttgocam2 PIR'
[04:45:29][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[04:45:29][C][gpio.binary_sensor:016]:   Pin: GPIO19
[04:45:29][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'ttgocam2 Button'
[04:45:29][C][gpio.binary_sensor:016]:   Pin: GPIO15
[04:45:29][C][captive_portal:088]: Captive Portal:
[04:45:29][C][mdns:115]: mDNS:
[04:45:29][C][mdns:116]:   Hostname: esphome-web-454f70
[04:45:29][C][ota:097]: Over-The-Air Updates:
[04:45:29][C][ota:098]:   Address: esphome-web-454f70.local:3232
[04:45:29][C][api:139]: API Server:
[04:45:29][C][api:140]:   Address: esphome-web-454f70.local:6053
[04:45:29][C][api:142]:   Using noise encryption: YES
[04:45:29][C][wifi_signal.sensor:009]: WiFi Signal 'ttgocam2 WiFi Signal'
[04:45:29][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[04:45:29][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[04:45:29][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[04:45:29][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[04:45:29][D][api:102]: Accepted 192.168.1.1
[04:45:29][W][component:214]: Component api took a long time for an operation (0.05 s).
[04:45:29][W][component:215]: Components should block for at most 20-30ms.
[04:45:29][D][api.connection:1089]: Home Assistant 2023.11.3 (192.168.1.1): Connected successfully
[04:45:48][D][binary_sensor:036]: 'ttgocam2 PIR': Sending state ON
[04:45:52][D][binary_sensor:036]: 'ttgocam2 PIR': Sending state OFF
[04:45:54][D][binary_sensor:036]: 'ttgocam2 PIR': Sending state ON
[04:45:56][D][binary_sensor:036]: 'ttgocam2 Button': Sending state ON
[04:45:56][D][voice_assistant:422]: State changed from IDLE to START_PIPELINE
[04:45:56][D][voice_assistant:428]: Desired state set to START_MICROPHONE
[04:45:56][D][voice_assistant:124]: microphone not running
[04:45:56][D][voice_assistant:206]: Requesting start...
[04:45:56][D][voice_assistant:422]: State changed from START_PIPELINE to STARTING_PIPELINE
[04:45:56][D][voice_assistant:443]: Client started, streaming microphone
[04:45:56][D][voice_assistant:422]: State changed from STARTING_PIPELINE to START_MICROPHONE
[04:45:56][D][voice_assistant:428]: Desired state set to STREAMING_MICROPHONE
[04:45:56][D][voice_assistant:159]: Starting Microphone
[04:45:56][D][voice_assistant:422]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[04:45:56][D][voice_assistant:529]: Event Type: 1
[04:45:56][D][voice_assistant:532]: Assist Pipeline running
[04:45:56][D][voice_assistant:422]: State changed from STARTING_MICROPHONE to STREAMING_MICROPHONE
[04:45:56][D][voice_assistant:529]: Event Type: 3
[04:45:56][D][voice_assistant:543]: STT started
[04:45:57][D][binary_sensor:036]: 'ttgocam2 Button': Sending state OFF
[04:45:57][D][voice_assistant:522]: Signaling stop...
[04:45:57][D][voice_assistant:422]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[04:45:57][D][voice_assistant:428]: Desired state set to IDLE
[04:45:57][D][voice_assistant:422]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[04:45:57][D][voice_assistant:422]: State changed from STOPPING_MICROPHONE to IDLE
[04:46:04][D][binary_sensor:036]: 'ttgocam2 PIR': Sending state OFF
[04:46:11][D][sensor:094]: 'ttgocam2 WiFi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
```