esphome/issues #4574: ESP32-CAM: Not possible to change contrast, brightness, saturation

Jpsy, 2023-06-11:
### The problem

Several methods of the esp32_camera component do not result in any visible change, i.e. (but not complete):
* set_contrast() + update_camera_parameters()
* set_brightness() + update_camera_parameters()
* set_saturation() + update_camera_parameters()

The reason is probably an oversimplified approach of the ESP32-CAM driver that is used by the component.
The central point here is the "SDE indirect register" of the OV2640, which contains a bit mask of enable flags for changes to contrast, brightness, saturation, color tint, negative and others. With each call to the driver's set_contrast / set_brightness / set_... the enable bit for the requested effect is set and all others are zeroed. Thus only the last changed effect remains active and all others are switched off again.

Now, the esp32_camera component's set_xyz() methods do not call the corresponding driver methods at all. 
Instead they store the requested effect strength parameter in an internal status var. 
All these status vars are then written to the OV2640 by a single call to update_camera_parameters(). This fires a series of method calls of the driver (including set_contrast, set_brightness, set_saturation and others). As described, only the last of these wins, which is probably the color tint change, as this seems to work. With this flow the brightness/saturation/contrast will never work.

I see two possible solutions here:

1. Simple but limited solution
Move the set_xyz() calls from update_camera_parameters() directly into the corresponding set_xyz() calls of the ESPHome component. This would at least allow to set all of the effects separately, but not in combination.

2. Take control of the enable flags in SDE indirect register
This would allow for all combinations of effects, but I see several comments on the web, that there are interferences between different effects that might need to be taken care of. I.e. brightness and contrast seem to interfere. But then again, this could also be handled by the user by sending adjusting method calls herself.

Solution 1. seems easy to implement and I think I can file a PR for that.
Solution 2. is beyond my reach. I tried but got stuck because I needed the write_SCCB() method of the driver. This method is currently not available in the ESPHome esp32_camera component and I have no idea how I could get hold of that. If someone could provide me a way to get hold of this method, I could also try to work on 2.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.1

### What platform are you using?

ESP32

### Board

ESP32-CAM (AI Thinker Clone)

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
esp32_camera:
  id: espcam
  ...
  ...

number:
  - platform: template
    name: Kontrast
    id: cam_contrast
    optimistic: true
    initial_value: 0
    min_value: -2
    max_value: 2
    step: 1
    set_action:
      - lambda: |-
          id(espcam).set_contrast(x);
          id(espcam).update_camera_parameters();
  - platform: template
    name: Helligkeit
    id: cam_brightness
    optimistic: true
    initial_value: 0
    min_value: -2
    max_value: 2
    step: 1
    set_action:
      - lambda: |-
          id(espcam).set_brightness(x);
          id(espcam).update_camera_parameters();
  - platform: template
    name: Sättigung
    id: cam_saturation
    optimistic: true
    initial_value: 0
    min_value: -2
    max_value: 2
    step: 1
    set_action:
      - lambda: |-
          id(espcam).set_saturation(x);
          id(espcam).update_camera_parameters();
```

Jpsy, 2023-10-24:
This issue is still valid.

Rudd-O, 2024-02-15:
This issue is still valid!  I just got bitten by it.

fornellas, 2024-02-17:
+1 here, this should be reopened. CC @OttoWinter for signal boost.