esphome/issues #4577: Fresh install in Docker with Error: - sh: 1: xtensa-esp32-elf-g++: not found

nkgiovannivl, 2023-06-14:
### The problem

After fresh install in docker and try "Install", ESPHome can not compile and install due a error related to "sh: 1: xtensa-esp32-elf-g++: not found "

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu

### Example YAML snippet

```yaml
esphome:
  name: regado-plantas-balcon

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "hidden_info_for_issue_report"

ota:
  password: "hidden_info_for_issue_report"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Regado-Plantas-Balcon"
    password: "hidden_info_for_issue_report"

captive_portal:
    
i2c:
  sda: GPIO21
  scl: GPIO22

switch:
  - platform: gpio
    pin: GPIO33
    name: dosifica_a_out
    id: dosifica_a_out
    on_turn_on:
    - delay: 600000ms
    - switch.turn_off: dosifica_a_out

  - platform: gpio
    pin: GPIO25
    name: dosifica_b_out
    id: dosifica_b_out
    on_turn_on:
    - delay: 600000ms
    - switch.turn_off: dosifica_a_out
    
sensor:
  - platform: adc
    pin: GPIO36
    name: "Humedad_A1_In"
    update_interval: 60s   

  - platform: adc
    pin: GPIO39
    name: "Humedad_A2_In"
    update_interval: 60s 

  - platform: adc
    pin: GPIO34
    name: "Humedad_A3_In"
    update_interval: 60s   

  - platform: adc
    pin: GPIO35
    name: "Lluvia_In"
    update_interval: 60s

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO27
      mode:
        input: true
        pullup: true
      inverted: true
    filters:
      - delayed_on: 30ms
      - delayed_off: 30ms    
    name: nivel-agua-in        

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.5.5
INFO Reading configuration /config/regado-plantas-balcon.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing regado-plantas-balcon (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40400.0
INFO Installing platformio/tool-esptoolpy @ ~1.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
INFO tool-esptoolpy@1.40400.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40400.0
INFO Installing platformio/tool-scons @ ~4.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-scons@4.40400.0 has been installed!
INFO tool-scons@4.40400.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 2.0.0
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/adc/adc_sensor.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/api_server.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/list_entities.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/adc/adc_sensor.cpp.o] Error 127
Compiling .pioenvs/regado-plantas-balcon/src/esphome/components/api/proto.cpp.o
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/api_server.cpp.o] Error 127
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/list_entities.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/regado-plantas-balcon/src/esphome/components/api/proto.cpp.o] Error 127
========================= [FAILED] Took 675.50 seconds =========================
```

### Additional information

Docker compose:

```
version: '3'

services:
    esphome:
        container_name: esphome
        image: ghcr.io/esphome/esphome:stable
        ports:
          - "6052:6052"
        volumes:
          - '/share/App_Data_Favernet/Docker/esphome/config:/config:rw'
          - '/etc/localtime:/etc/localtime:ro'
        devices:
          - '/dev/ttyUSB0:/dev/ttyUSB0'
        networks:
            macvlan-favernet:
                ipv4_address: 192.168.1.15
        privileged: true
        restart: always

networks:
  macvlan-favernet:
    external: true
    name: macvlan-favernet
```

ssieb, 2023-06-14:
Please come to the esphome discord server so we do this more interactively.

top4ek, 2023-06-14:
Not fresh installation, but the same. 
Raspbian on Pi4. Linux hassio 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

nkgiovannivl, 2023-06-15:
Related to a volumes: section in docker compose, this volume path belongs to a CIFS shared folder, so I changed to a local-persist folder and it does not work, but when I changed to a local folder compiling works without problems, I'm not sure if the problem could be symlinks issue or an access rigth issue or other issue, because I don't know how to check it.....

closed 2023-06-15 by nkgiovannivl

top4ek, 2023-06-15:
Hm... I'm using home Assistant(supervised) add-on and no network mounts. Even 8266 builds are broken, devices ran for ages.