esphome/issues #4580: Colors inverted on ili9488

duckfullstop, 2023-06-15:
### The problem

With a fully wired ILI9488 connected over 4-wire SPI, any colors are rendered inverse to their definitions (i.e pink is green, white is black, black is white, etc.)

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Wemos S2 mini

### Component causing the issue

ili9xxx

### Example YAML snippet

```yaml
esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

font:
  - file: 'fonts/OpenSans-Regular.ttf'
    id: font_opensans_regular
    size: 28
  

spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

color:
  - id: c_primary
    red: 0%
    blue: 100%
    green: 0%

display:
  - id: disp
    platform: ili9xxx
    model: "ILI9488"
    cs_pin: GPIO9
    dc_pin: GPIO10
    reset_pin: GPIO11
    update_interval: 5s
    color_palette: NONE
    lambda: |-
      it.fill(Color::WHITE);
      it.print(0, 0, id(font_opensans_regular), id(c_primary), TextAlign::TOP_LEFT, "Hello World!");
```

### Anything in the logs that might be useful for us?

```txt
[17:26:48][C][spi:101]: SPI bus:
[17:26:48][C][spi:102]:   CLK Pin: GPIO14
[17:26:48][C][spi:103]:   MISO Pin: GPIO12
[17:26:48][C][spi:104]:   MOSI Pin: GPIO13
[17:26:48][C][spi:106]:   Using HW SPI: YES
[17:26:49][C][ili9xxx:047]: ili9xxx
[17:26:49][C][ili9xxx:047]:   Rotations: 0 °
[17:26:49][C][ili9xxx:047]:   Dimensions: 480px x 320px
[17:26:49][C][ili9xxx:050]:   Color mode: 8bit Indexed
[17:26:49][C][ili9xxx:060]:   18-Bit Mode: YES
[17:26:49][C][ili9xxx:063]:   Reset Pin: GPIO11
[17:26:49][C][ili9xxx:064]:   DC Pin: GPIO10
[17:26:49][C][ili9xxx:070]:   Update Interval: 5.0s
[17:26:49][C][psram:020]: PSRAM:
[17:26:49][C][psram:021]:   Available: YES
[17:26:49][C][psram:024]:   Size: 2047 KB
```

### Additional information

My gut says that simply doing `this->invert_display_(true);` in the ili9488 setup routine ([ili9xxx_display.cpp#403](https://github.com/esphome/esphome/blob/467e42d8aa5c1dd8a6ec13b09075117d2f209e6a/esphome/components/ili9xxx/ili9xxx_display.cpp#L403)) should be sufficient - but I'm hesitating on bringing up a PR for this just in case people have perfectly colorful implementations (and we end up inverting everyone's except mine!)

Perhaps a YAML flag to invert colours would be a better idea, with a suggested default for the ili9488 (and m3stack) of `true`?

guillempages, 2023-06-22:
There are a lot of displays using some ILI chip, and each of them has a different configuration of color inversion, RGB vs BGR, mirroring,...
Either we need to support the different devices as such, or we should really add the option to set inversion and such things on the configuration.
I had a discussion with @jesserockz  and @kbx81 regarding that one or two weeks ago, and at that point the consensus was to add new devices, and not a generic config, btu this might need to be revisited.

duckfullstop, 2023-06-22:
Would a good compromise be that setting the `device` sets some default configuration observed on the majority of that flavour of device, that can then be overriden manually on a case-by-case basis?

Having three or four revisions of each `device` for different dialects would probably get pretty messy, pretty quickly - and having the option to override lets users quickly iterate and find a working configuration without relying on spinning custom components or raising PRs.

kbx81, 2023-07-05:
Is this fixed by https://github.com/esphome/esphome/pull/5027 ?