esphome/issues #4581: total_daily_energy (and other components?) needs updated for new ESPTime in core (PR #4926)

bkaufx, 2023-06-15:
### The problem

I believe total_daily_energy needs updated after this: https://github.com/esphome/esphome/pull/4926

I'd do a pull request myself but I don't fully understand what needs to be done TBH.

### Which version of ESPHome has the issue?

2023.6.0b1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Component causing the issue

total_daily_energy

### Example YAML snippet

```yaml
https://raw.githubusercontent.com/athom-tech/athom-configs/main/athom-smart-plug-v2.yaml
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/total_daily_energy/total_daily_energy.h:7,
                 from src/esphome/components/total_daily_energy/total_daily_energy.cpp:1:
src/esphome/core/time.h:11:13: warning: 'bool esphome::is_leap_year(uint32_t)' declared 'static' but never defined [-Wunused-function]
   11 | static bool is_leap_year(uint32_t year);
      |             ^~~~~~~~~~~~
src/esphome/core/time.h:13:16: warning: 'uint8_t esphome::days_in_month(uint8_t, uint16_t)' declared 'static' but never defined [-Wunused-function]
   13 | static uint8_t days_in_month(uint8_t month, uint16_t year);
      |                ^~~~~~~~~~~~~
```

ssieb, 2023-06-16:
This has nothing to do with the sensor.  It's something specific to ESP8266 in the `time.h` file.  It doesn't happen with ESP32.

droscy, 2023-06-16:
@ssieb it happens also on ESP32 with esp-idf framework.

```
Compiling .pioenvs/esp07/src/esphome/components/wifi/wifi_component_esp_idf.o
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/time/real_time_clock.cpp:1:
src/esphome/core/time.h:11:13: warning: 'bool esphome::is_leap_year(uint32_t)' declared 'static' but never defined [-Wunused-function]
 static bool is_leap_year(uint32_t year);
             ^~~~~~~~~~~~
src/esphome/core/time.h:13:16: warning: 'uint8_t esphome::days_in_month(uint8_t, uint16_t)' declared 'static' but never defined [-Wunused-function]
 static uint8_t days_in_month(uint8_t month, uint16_t year);
                ^~~~~~~~~~~~~
```

ssieb, 2023-06-16:
I wonder why it doesn't happen with an arduino compile.

bkaufx, 2023-06-16:
OK, good title change.  The problem is definitely a general one and not specific to total_daily_energy.  Even just using this as the entire config still gives the same warnings.

YAML config:
```
esphome:
  name: asdf

esp8266:
  board: esp01_1m
```

Warnings:
```
Compiling .pioenvs/asdf/FrameworkArduino/cont_util.cpp.o
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/core/time.h:11:13: warning: 'bool esphome::is_leap_year(uint32_t)' declared 'static' but never defined [-Wunused-function]
   11 | static bool is_leap_year(uint32_t year);
      |             ^~~~~~~~~~~~
Compiling .pioenvs/asdf/FrameworkArduino/core_esp8266_app_entry_noextra4k.cpp.o
src/esphome/core/time.h:13:16: warning: 'uint8_t esphome::days_in_month(uint8_t, uint16_t)' declared 'static' but never defined [-Wunused-function]
   13 | static uint8_t days_in_month(uint8_t month, uint16_t year);
      |                ^~~~~~~~~~~~~
```

tommyguuuun, 2023-06-22:
I got the same warnings since the newest update

stackner, 2023-06-22:
Same happening here

RobvanHamersveld, 2023-06-22:
Same here, I looks like everything works fine though.

mspiridakis, 2023-06-22:
Same here, it must be a bug. I don't see any changes in docs for time component.

closed 2023-06-23 by jesserockz

KitHubek, 2023-06-23:
after update from 2023.6 to 2023.6.1 problem still exsist, at log:

```
Compiling /data/wt32-eth01-bt-proxy/.pioenvs/wt32-eth01-bt-proxy/bootloader_support/src/flash_qio_mode.o
In file included from src/esphome.h:69,
                 from src/main.cpp:3:
src/esphome/core/time.h:11:13: warning: 'bool esphome::is_leap_year(uint32_t)' declared 'static' but never defined [-Wunused-function]
 static bool is_leap_year(uint32_t year);
             ^~~~~~~~~~~~
src/esphome/core/time.h:13:16: warning: 'uint8_t esphome::days_in_month(uint8_t, uint16_t)' declared 'static' but never defined [-Wunused-function]
 static uint8_t days_in_month(uint8_t month, uint16_t year);
                ^~~~~~~~~~~~~
Compiling /data/wt32-eth01-bt-proxy/.pioenvs/wt32-eth01-bt-proxy/bootloader_support/src/bootloader_flash_config_esp32.o
```

mspiridakis, 2023-06-23:
> after update from 2023.6 to 2023.6.1 problem still exsist, at log:
>
> ```
> Compiling /data/wt32-eth01-bt-proxy/.pioenvs/wt32-eth01-bt-proxy/bootloader_support/src/flash_qio_mode.o
> In file included from [...]

issue will be resolved in 2023.6.2 according to https://github.com/esphome/esphome/pull/4993

KitHubek, 2023-06-23:
yes, ok. thanks

stackner, 2023-06-23:
It has been fixed in 2023.6.2 no more errors.