esphome/issues #4582: Pulse does not allow min/max_brightness params

Normanras, 2023-06-16:
### The problem

According to the [Pulse Light effect](https://esphome.io/components/light/index.html#pulse-effect) in the documentation, you can pass a `min_brightness` and `max_brightness`. However, when I try to do this in ESPhome, I get the error: 
```[min_brightness] is an invalid option for [pulse]. Please check the indentation.```

I've looked around at various tutorials or other pieces of documentation, but nothing points toward what the correct indentation should be or if this was deprecated but not yet removed from the docs. Is this parameter still possible?

What led me here is trying to create a pulse effect without the light turning completely off between intervals. 

### Which version of ESPHome has the issue?

2023.5.5 (I believe, docker with latest tag)

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP32

### Board

ESP32-DEVKITV1

### Component causing the issue

neopixel lights

### Example YAML snippet

```yaml
- pulse:
          name: "Fast Pulse"
          transition_length: 0.5s
          update_interval: 0.5s
          min_brightness: 0%
          max_brightness: 100%
```

ssieb, 2023-06-16:
Check your esphome version.  You almost certainly don't have the latest version.  Those options were added in 2023.5.0.

Normanras, 2023-06-16:
Shouldn't the `latest` tag for docker update it correctly? I couldn't easily find how to find my version from the web-ui.

Normanras, 2023-06-16:
For this specific ticket, it can be closed. There may be a separate issue with the `latest` tag not pulling the actual latest build, but when I specifically ran `docker pull esphome/esphome:2023.5.0`, the `min` and `max_brightness` worked as expected in the docs.

closed 2023-06-16 by Normanras