esphome/issues #4585: Rename hostname delete the configuration yaml

spad, 2023-06-17:
### The problem

I used Rename Hostname function to rename a device from minir4_mydevice to minir4-mydevice.
The new firmware was successfully uploaded but the configuration was deleted and no longer available.

I recovered it from a backup, compiled it again and re-uploaded in order to have it again in my panel

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.06.2

### What platform are you using?

ESP32

### Board

sonoff mini r4 extreme

closed 2023-10-31 by github-actions[bot]

Brockhold, 2024-02-14:
I've just experienced the same behavior, I used the esphome rename feature to change the name of a config for a device that was offline (mDNS cannot find it, but direct IP can), and after successfully compiling and installing, the config file was deleted.
Recovering the config from `.esphome/build/<device>/src/main.cpp` is possible, but a lot of work!

alpharesearch, 2024-04-26:
This just happened to me too, remove the rename button!

Codex-, 2024-06-21:
Also encountered this today, and no temp files to recover it from :/

alpharesearch, 2024-06-21:
> Also encountered this today, and no temp files to recover it from :/

If you have a full backup you can pull just that file on your PC out of the folder and copy the content over.

Codex-, 2024-06-21:
> If you have a full backup you can pull just that file on your PC out of the folder and copy the content over.

None unfortunately for this particular config, didn't think esphome would nuke it. 

Managed to restitch it from a temp autogen'd C entrypoint which is better than nothing