esphome/issues #4587: SSD1306 contrast issue since 2023.06 dev

zed984, 2023-06-20:
### The problem

I have two devices with the same LCD driver SSD1306. One via I2C and one via SPI. Other sensors share the bus and work fine.
The LCD was working fine before I update to the 2022.06 dev branch. Updated today again to the latest 2022.07 dev release for the same

### Which version of ESPHome has the issue?

2023-07-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

ESP32-C3-DEVKITM

### Component causing the issue

display

### Example YAML snippet

```yaml
esphome:
  name: esp-sense-test-2
  # https://github.com/esphome/issues/issues/2931
  platformio_options:
    board_build.flash_mode: dio  

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "secret"

ota:
  password: "secret"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Esp-Sense-Test-2"
    password: "y3jrdldaRInb"
    
i2c:
  sda: GPIO0
  scl: GPIO1
  scan: false
  id: bus_a

sensor:
  - platform: bme680 #bme680_bsec
    temperature:
      name: "Office temperature"
      id: "office_temperature"
    pressure:
      name: "Office pressure"
      id: "office_pressure"
    humidity:
      name: "Office humidity"
      id: "office_humidity"
    gas_resistance:
      name: "Office Gas Resistance"
      id: "office_gas_resistance"
    address: 0x77
    update_interval: 60s

  - platform: bh1750
    name: "Office Illuminance"
    id: "office_illuminance"
    address: "0x23"
    update_interval: 15s

binary_sensor:
  - platform: gpio
    pin: GPIO10
    name: "Office motion"
    id: "office_motion"
    device_class: motion

font:
  - file: "gfonts://Roboto@700"
    id: roboto_large
    size: 35
  - file: "gfonts://Roboto"
    id: roboto_small
    size: 20

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    id: "lcd"
    address: 0x3C

    lambda: |-
      it.printf(it.get_width()/2, 4, id(roboto_large), TextAlign::TOP_CENTER, "%.1f°C", office_temperature->state);
      it.printf(0, it.get_height(), id(roboto_small), TextAlign::BOTTOM_LEFT, "%.0f %%", office_humidity->state);
      it.printf(it.get_width(), it.get_height(), id(roboto_small), TextAlign::BOTTOM_RIGHT, "%5.0f lx", office_illuminance->state);
```

### Anything in the logs that might be useful for us?

```txt
[21:22:15][C][ssd1306_i2c:023]: I2C SSD1306
[21:22:15][C][ssd1306_i2c:023]:   Rotations: 0 °
[21:22:15][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[21:22:15][C][ssd1306_i2c:024]:   Address: 0x3C
[21:22:15][C][ssd1306_i2c:025]:   Model: SSD1306 128x64
[21:22:15][C][ssd1306_i2c:027]:   External VCC: NO
[21:22:15][C][ssd1306_i2c:028]:   Flip X: YES
[21:22:15][C][ssd1306_i2c:029]:   Flip Y: YES
[21:22:15][C][ssd1306_i2c:030]:   Offset X: 0
[21:22:15][C][ssd1306_i2c:031]:   Offset Y: 0
[21:22:15][C][ssd1306_i2c:032]:   Inverted Color: NO
[21:22:15][C][ssd1306_i2c:033]:   Update Interval: 1.0s
```

### Additional information

[image]

MOST2K2, 2023-06-21:
Same here, please fix it asap, with invert=true i get a black screen.

zed984, 2023-06-21:
>  with invert=true i get a black screen.

Confirmed that invert=true gives a black screen as well.
Also tried various contrast settings which has limited effect.

jimmy-edison, 2023-06-22:
Same issue here with SSD1306 and SH1106
- platform: ssd1306_i2c
    model: "SSD1306 128x64"

- platform: ssd1306_i2c
    model: "SH1106 128x64"

erik3rik, 2023-06-22:
Same issue here. Using a 
```
display:
  - platform: ssd1306_spi
    model: "SH1106 128x64"
```
Invert=true gives black screen. Invert=false gives a completely white screen. I'm using a quite low contrast of 10%.

Worked on 2023.5.5. Broke when updated to 2023.6.0. Started working again when reverting back to 2023.5.5 and re-installed the code on the ESP-32.

I notice that there are quite a few updates that mention "display" in the release docs. The only breaking one is:

`display: add BaseImage and provide only Image::get_pixel method`

But that is not something I use. I don't even use images on it, just text.

Electronlibre2012, 2023-06-22:
Hello,

i thought that ```invert=true``` will solved the issue because before the font is black and the text is white (or whenever color you choose to buy) and with this option the font is white and the text is black but its so bad that i cant read anything on my tiny 128x64 pixels ssd1306...

Wait for a fix please

rubendediegomartinez, 2023-06-22:
Same issue, using:

randybb, 2023-06-22:
@zed984 There is no contrast issue - only your OLED is burn-in like an old CRT... But the problem is that the whole screen is white and not showing anything. Would be good to remove that "contrast" from description/issue name and close the other issues that people created without checking other issues first.

randybb, 2023-06-22:
As [khenderick](https://github.com/khenderick) mentioned, display component _inkplate6_ (model _inkplate_6_plus_) is also affected.

ggottwald, 2023-06-22:
waveshare_epaper is also affected

erik3rik, 2023-06-22:
I saw this issue was labeled `ssd1306_i2c`. FYI it also affects my display using `ssd1306_spi` i.e. also over SPI.

adryzzts, 2023-06-22:
Same here, after upgrade at 2023.6.0, two display ko, screen all white

mariusbaumann, 2023-06-22:
Same here with Inkplate 6. It stays white since the 2023.6.0 update. But maybe helpful observation: The graphs are being plotted correctly. 
[image]

billmaterial, 2023-06-22:
Exactly the same issue also, no text shown on the screen.

platform: ssd1306_i2c
model: "SSD1306 128x64"
address: 0x3C

The "Invert" function seems to be opposite now too.

guillempages, 2023-06-22:
Could you try explicitly setting a black text? (I.e. passing "Color::BLACK" as a color parameter on your printed texts)
if this does not work, Color::WHITE might do the trick (though I hope not ;-) )
Or just explicitly "Color(0,0,0,0)"?
We are working on a fix, but would need confirmation that the suspected issue is actually the root cause.

@ayufan, FYI

ayufan, 2023-06-22:
@mariusbaumann Can you test this fix https://github.com/esphome/issues/issues/4587?

daernsinstantfortress, 2023-06-22:
> @guillempages As a quick test, could you try explicitly setting a black text? (I.e. passing "Color::BLACK" as a color parameter on your printed texts)
> if this does not work, could you try an explicit "Color(0,0,0,0)"?

On the Waveshare ePaper 2.9", the whole display has gone black (it would normally be white), but forcing the text to black does indeed draw white text on the black screen:

```
it.print(10, 10, id(main_sensor_unit), Color::BLACK, "CUPRA");
```
[image]

ayufan, 2023-06-22:
@daernsinstantfortress It is the same problem. The `waveshare` does have inverted logic for pixels:

```cpp
  if (!color.is_on()) {
    this->buffer_[pos] |= 0x80 >> subpos;
  } else {
    this->buffer_[pos] &= ~(0x80 >> subpos);
  }
```

guillempages, 2023-06-22:
@daernsinstantfortress the "quickest" fix/workaround would be to add an explicit "it.fill(Color::WHITE);" at the begining of your display lambda, until the fix gets merged and released.
Or just wait for the latter.

rubendediegomartinez, 2023-06-22:
> @zed984 There is no contrast issue - only your OLED is burn-in like an old CRT... But the problem is that the whole screen is white and not showing anything. Would be good to remove that "contrast" [...]

I deleted the contrast line and the problem remains.  Previously to the ESPHome 2023.6.0 the display runs perfectly

I can appreciate some characters but are ilegibles.
BR

billmaterial, 2023-06-22:
I can also confirm Guillempages fix of placing "it.fill(Color::BLACK);" at the beginning of the lambda section works with my display:

platform: ssd1306_i2c
model: "SSD1306 128x64"
address: 0x3C

randybb, 2023-06-22:
Mentioned fix works with SSD1306, can somebody test it with other displays?
```yaml
external_components:
  - source: github://pr#4991
    components: display
```

mariusbaumann, 2023-06-22:
> ```yaml
> external_components:
>   - source: github://pr#4991
>     components: display
> ```

Getting errors with the lines above:
[logs_inkplate_run.txt](https://github.com/esphome/issues/files/11834233/logs_inkplate_run.txt)

daernsinstantfortress, 2023-06-22:
> Mentioned fix works with SSD1306, can somebody test it with other displays?
> 
> ```yaml
> external_components:
>   - source: github://pr#4991
>     components: display
> ```

Yup, working fine here with this on Waveshare epaper 👍

mrzottel, 2023-06-22:
Tested with SSD1306 and works fine

ayufan, 2023-06-22:
@mariusbaumann This is change targeting development version, not what is in `master`, so there might be some compilation errors. This will not compile with `graph` when using this PR.

mrzottel, 2023-06-22:
@mariusbaumann 
I had the same issue. Try "Clean build files" and build again

mariusbaumann, 2023-06-22:
👍 Without graph, it's working fine!

zatricky, 2023-06-22:
Confirming the `it.fill()` workaround works with `model: "SH1106 128x64"`

alexandreflach, 2023-06-22:
it works for me as well:  it.fill(Color::BLACK); 

- platform: ssd1306_i2c
    model: "SSD1306 128x64"

Hufnagels, 2023-06-22:
It works for me too, with followings with version 2023.6

`
######################################
# 0.96" DISPLAY SECTION
######################################
display:
  - platform: ssd1306_i2c
    model: "SSD1306 64x48"
    i2c_id: i2cbus_ssd1306
    reset_pin: D0
    address: 0x3C
    id: mini_oled
    contrast: 60%
    invert: true
    pages:
      - id: page1
        lambda: |-
          // LINE 1
          // Manual/Auto state
          it.print(0, 0, id(roboto_regular_9), Color::BLACK, id(state_auto_mode) ? "AUTO":"MANUAL");
`
So, this way is screen normally BLACK and font is WHITE

And there is another issue when compiling:
`In file included from src/esphome/components/display/display_buffer.h:9,
                 from src/esphome.h:23,
                 from src/main.cpp:3:
src/esphome/core/time.h:11:13: warning: 'bool esphome::is_leap_year(uint32_t)' declared 'static' but never defined [-Wunused-function]
   11 | static bool is_leap_year(uint32_t year);
      |             ^~~~~~~~~~~~
src/esphome/core/time.h:13:16: warning: 'uint8_t esphome::days_in_month(uint8_t, uint16_t)' declared 'static' but never defined [-Wunused-function]
   13 | static uint8_t days_in_month(uint8_t month, uint16_t year);`

jesserockz, 2023-06-22:
Fixed in https://github.com/esphome/esphome/releases/tag/2023.6.1

closed 2023-06-22 by jesserockz

MOST2K2, 2023-06-23:
Fixed with latest update :) (I did no code changes!) Thanks!

billmaterial, 2023-06-23:
FYI, installed ESPhome 2023.6.1 and this has fixed the issue.