esphome/issues #4589: ESP32 crashes when playing AAC radio streams

RoboMagus, 2023-06-20:
### The problem

I've got an ESP32-S3 operating as a media player using a pcm5102 I2S audio component.
When playing mp3 streams, everything is fine. But when loading an AAC stream the device crashes and reboots.

### Which version of ESPHome has the issue?

2023.7.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/a

### What platform are you using?

ESP32

### Board

AI-S3 / ESP32-S3-DevKitC-1

### Component causing the issue

media_player, i2s_audio

### Example YAML snippet

```yaml
esp32:
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  framework:
    type: arduino
    version: 2.0.8    
    platform_version: 6.3.1

i2s_audio:
  i2s_lrclk_pin: GPIO13
  i2s_bclk_pin: GPIO14

media_player:
- platform: i2s_audio
  name: S3 Media Player
  dac_type: external
  i2s_dout_pin: GPIO12
  mode: stereo
```

### Anything in the logs that might be useful for us?

```txt
[D][media_player:066]:   Media URL: https://playerservices.streamtheworld.com/api/livestream-redirect/TLPSTR21AAC.aac
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060930  A0      : 0x8201dfd5  A1      : 0x3fceb7d0  
A2      : 0x3fcab70c  A3      : 0x3fc9bce0  A4      : 0x3fceb810  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x006e776f  A8      : 0x8201b383  A9      : 0x3fceb710  
A10     : 0x3fcaaf70  A11     : 0x3c0d6730  A12     : 0x000000a0  A13     : 0x00000000  
A14     : 0x46494441  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  

Backtrace: 0xfffffffd:0x3fceb7d0 0x4201dfd2:0x3fcebad0 0x4201e0e7:0x3fcebb10 0x4202166a:0x3fcebb30 0x4202175f:0x3fcebdb0 0x42008e57:0x3fcebdd0 0x420c5c15:0x3fcebdf0 0x420c5cd5:0x3fcebe10 0x4200f1b5:0x3fcebe30 0x42011592:0x3fcebe60 0x4202becd:0x3fcebe80

ELF file SHA256: a46154f60a7898ff

E (16392) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420c7bb6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (232) esp_core_dump_flash: No core dump partition found!
E (232) esp_core_dump_flash: No core dump partition found!
[I][logger:262]: Log initialized
[C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
```

### Additional information

URL used for testing: https://playerservices.streamtheworld.com/api/livestream-redirect/TLPSTR21AAC.aac

RoboMagus, 2023-06-20:
Related issue on used library: schreibfaul1/ESP32-audioI2S#439

xkuyax, 2023-10-25:
Hey @RoboMagus how did you get the PCM5102 working with esphome?
I have tried something similar and with a different project (using bt as a receiver) and with tasmota and their i2s implementation it works, but i cant get it to work with esphome

RoboMagus, 2023-10-26:
@xkuyax, the relevant part of the config is included in the yaml shown above. Other than that it's just making sure the pins of the audio module are propperly wired to the configured pins on the ESP board. Could you share what's not working for you?

fborangic, 2023-11-11:
I'm also encountering this issue on a ESP32 device. If I try to play any aac radio stream the device reboots.

I see this in the log output:
`[23:37:31][D][media_player:066]:   Media URL: http://astreaming.virginradio.ro:8000/EuropaFM_aac
[23:37:31][W][component:214]: Component api took a long time for an operation (0.21 s).
[23:37:31][W][component:215]: Components should block for at most 20-30ms.
WARNING media-player-veranda.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for media-player-veranda.local
WARNING Disconnected from API`

koriwi, 2023-12-24:
i tried this and it seams to work.
```yaml
esphome:
  name: esp32-audio-test
  friendly_name: ESPHome Audio Test
  includes: 
  libraries: 
    - WiFi
    - WiFiClientSecure
    - "ESP32-audioI2S=https://github.com/RoboMagus/esphome-ESP32-audioI2S.git#merge_upstream_master"
```

chess-levin, 2024-01-13:
Hi guys, I've a ESP32-S3 DevKitC1 N16R8 and I'm not sure which are the I2S pin I should use to connect my PCM5102 to. Could you help me?

heibertelf, 2024-04-08:
> i tried this and it seams to work.

My esphome speaker has stopped playing anything at all, even mp3s