esphome/issues #4594: Compiling errors after 2023.6 release related to time functions

pyrodex, 2023-06-22:
### The problem

The following errors in compiling showed up after upgrading the docker container published to 2023.6. Code still compiles but just throwing errors.

```
Updating /config/barfridge-smartplug-cbdc36.yaml
------------------------------------------------------------

INFO ESPHome 2023.6.0
INFO Reading configuration /config/barfridge-smartplug-cbdc36.yaml...
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Compiling app...
Processing barfridge-smartplug-cbdc36 (board: esp8285; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-scons @ ~4.40502.0
INFO Installing platformio/tool-scons @ ~4.40502.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40502.0 has been installed!
INFO tool-scons@4.40502.0 has been installed!
Tool Manager: Removing tool-scons @ 4.40400.0
INFO Removing tool-scons @ 4.40400.0
Tool Manager: tool-scons@4.40400.0 has been removed!
INFO tool-scons@4.40400.0 has been removed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/cse7766/cse7766.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/dashboard_import/dashboard_import.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/factory_reset/button/factory_reset_button.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/addressable_light.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/automation.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/esp_color_correction.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/esp_hsv_color.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/esp_range_view.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/light_call.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/light_json_schema.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/light_output.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/light/light_state.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/barfridge-smartplug-cbdc36/src/esphome/components/mdns/mdns_esp32.cpp.o
[... 354 lines cut ...]

mdns:
  disabled: false

web_server:
  port: 80
  local: true

captive_portal:

sensor:
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal Sensor"
    update_interval: 60s

time:
  - platform: sntp
    id: eastern
    timezone: "Amercia/New_York"
    servers:
      - 192.168.24.1
    on_time_sync:
      then:
        - logger.log: "Synchronized system clock"
```

pyrodex, 2023-06-22:
Dup

randybb, 2023-06-22:
Only scary warnings... anyway, it is a duplicate of #4581

closed 2023-06-22 by randybb