esphome/issues #4595: Power consumption in deep sleep

maframan, 2023-06-22:
### The problem

High power consumptation in deep sleep mode.
By the esp32 documentation, the power consumptation must be between 0.15 mA (when the ULP coprocessor is on) and 10 µA.
(reference https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/#:~:text=one%20by%20one.-,ESP32%20Active%20Mode,about%20240%20mA%20of%20power.)
In my case with last esphome firmware the consumption is about 5,4mA

[image]

### Which version of ESPHome has the issue?

ESPHome 2023.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.6.2 Supervisor 2023.06.2 Frontend 20230608.0 - latest

### What platform are you using?

ESP32

### Board

esp32-s3

### Component causing the issue

N/A

### Example YAML snippet

```yaml
esphome:
  name: power-test
  friendly_name: Power Test

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "MuOrmqcbCsc0Fzys5vl6xb8HC5day3wUK/0bQ+Mug20="

ota:
  password: "ebbce1662db9203bb2ea675c96c29e6a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Power-Test Fallback Hotspot"
    password: "xxxxxxxxxxx"

captive_portal:

deep_sleep:
  run_duration: 2min
  wakeup_pin:
    number: GPIO04
    inverted: False
    mode: INPUT_PULLUP
  wakeup_pin_mode: KEEP_AWAKE
  id: deep_sleep_1
```

### Anything in the logs that might be useful for us?

```txt
N/A
```

### Additional information

N/A

maframan, 2023-06-22:
Details about ESP32-S3 component: amazon link https://www.amazon.it/dp/B0BVQLW2S6?psc=1&ref=ppx_yo2ov_dt_b_product_details

maframan, 2023-06-23:
Any people help me? Thanks a lot.

vondraussen, 2023-07-13:
If you want to achieve the lowest possible power consumption with a dev board, you have to make sure, that you power the esp32 directly via the pin header with 3.3V. You might even need to remove one solder bridge.
Is it the devkit-c you're using?

maframan, 2023-07-14:
> If you want to achieve the lowest possible power consumption with a dev board, you have to make sure, that you power the esp32 directly via the pin header with 3.3V. You might even need to remove one solder bridge. Is it the devkit-c you're using?

Hi, this is my devkit AZDelivery 3 x Scheda di Sviluppo ESP-32 NodeMCU Dev Kit C V4 WiFi  https://www.amazon.it/gp/product/B08BZGC22Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
What'is the bridge, the CP2102 component?

vondraussen, 2023-07-14:
How do you supply power to the board? Via USB or the pin header? If the pin header, which pins?