esphome/issues #4596: Unable to use rp2040_pio_led_strip building on Linux x86_64

Kummallinen, 2023-06-22:
### The problem

When updating to ESPHome 2023.6.0 I saw "Rp2040 pio ledstrip esphome#4818" listed as a new integration. When trying it I get this error:
`INFO ESPHome 2023.6.0
INFO Reading configuration /config/esphome/cubelight.yaml...
INFO Generating C++ source...
INFO Generating PIO assembly code
INFO Assembling PIO assembly code
Tool Manager: Installing earlephilhower/tool-pioasm-rp2040-earlephilhower
INFO Installing earlephilhower/tool-pioasm-rp2040-earlephilhower
Error: Could not find the package with 'earlephilhower/tool-pioasm-rp2040-earlephilhower' requirements for your system 'linux_x86_64'
ERROR PIO assembly failed`

Maybe related but I found issue #4541, which suggests its broken for all environments? If so then I suggest the integration isn't listed in the release notes.

### Which version of ESPHome has the issue?

2023.6.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RP2040

### Board

PicoW

### Component causing the issue

rp2040_pio_led_strip

### Example YAML snippet

```yaml
light:
  - platform: rp2040_pio_led_strip
    name: cube_light
    id: cube_light
    pin: GPIO15
    num_leds: 50
    pio: 0
    rgb_order: GRB
    chipset: WS2812B
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.0
INFO Reading configuration /config/esphome/cubelight.yaml...
INFO Generating C++ source...
INFO Generating PIO assembly code
INFO Assembling PIO assembly code
Tool Manager: Installing earlephilhower/tool-pioasm-rp2040-earlephilhower
INFO Installing earlephilhower/tool-pioasm-rp2040-earlephilhower
Error: Could not find the package with 'earlephilhower/tool-pioasm-rp2040-earlephilhower' requirements for your system 'linux_x86_64'
ERROR PIO assembly failed
```

ssieb, 2023-06-22:
The component was added before platformio broke it.  Hopefully there's a workaround. Otherwise we'll add a note to the doc page.

ssieb, 2023-06-23:
@Kummallinen please try adding the `external_components:` section from the linked PR and see if that works for you.

Kummallinen, 2023-06-23:
@ssieb that gets it much further but it then fails at:
```
src/main.cpp: In function 'void setup()':
src/main.cpp:169:67: error: 'rp2040_pio_led_strip_pixels_program' was not declared in this scope
  169 |   rp2040_pio_led_strip_rp2040pioledstriplightoutput->set_program(&rp2040_pio_led_strip_pixels_program);
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:170:72: error: 'rp2040_pio_led_strip_driver_pixels_init' was not declared in this scope
  170 |   rp2040_pio_led_strip_rp2040pioledstriplightoutput->set_init_function(rp2040_pio_led_strip_driver_pixels_init);
      |                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Archiving /data/cubelight/.pioenvs/cubelight/lib1c6/liblwIP_CYW43.a
Compiling /data/cubelight/.pioenvs/cubelight/libce7/lwIP_Ethernet/LwipIntf.cpp.o
Compiling /data/cubelight/.pioenvs/cubelight/libce7/lwIP_Ethernet/LwipIntfCB.cpp.o
*** [/data/cubelight/.pioenvs/cubelight/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.07 seconds ==========================
```

ssieb, 2023-06-23:
Add `refresh: 1min` to the `external_components:` entry and try again.

closed 2023-06-23 by jesserockz

Kummallinen, 2023-06-23:
@ssieb That worked. Thank you for the very fast fix!

ssieb, 2023-06-23:
Thank Jesse.  But also, remove that section from your config now since it's in the latest release.

Kummallinen, 2023-06-23:
Updated to 2023.6.2, removed that section and did a clean build and it works.

Thank you all again for the fast fix