esphome/issues #4598: In 2023.6.0, AirThings BLE config reported as invalid despite being identical to documentation

strikeir13, 2023-06-22:
### The problem

My yaml config for Airthings Wave Plus is identical to the yaml in the Airthings documentation, but there is an error given that prevents compiling: `At least one of 'id:' or 'name:' is required!.` My yaml config is included below.

### Which version of ESPHome has the issue?

2023.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.2

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
#Airthing Wave Plus BLE tracker
  - platform: airthings_wave_plus
    ble_client_id: airthings01
    update_interval: 5min # default
    temperature:
      name: "WavePlus Temperature"
    radon:
      name: "WavePlus Radon"
    radon_long_term:
      name: "WavePlus Radon Long Term"
    pressure:
      name: "WavePlus Pressure"
    humidity:
      name: "WavePlus Humidity"
    co2:
      name: "WavePlus CO2"
    tvoc:
      name: "WavePlus VOC"

ble_client:
  - mac_address: xx:xx:xx:xx:xx:xx [redacted but config worked prior to 2023.6.x]
    id: airthings01

esp32_ble_tracker:
```

### Anything in the logs that might be useful for us?

```txt
The error given is: "At least one of 'id:' or 'name:' is required!."
```

dasb00ter, 2023-06-26:
I have same also with previous yaml and just trying to update:
`esphome:
 name: ble-hub

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

sensor:
  - platform: airthings_wave_plus
    ble_client_id: airthings01
    update_interval: 30min # default
    temperature:
      name: "WavePlus Temperature"
    radon:
      name: "WavePlus Radon"
    radon_long_term:
      name: "WavePlus Radon Long Term"
    pressure:
      name: "WavePlus Pressure"
    humidity:
      name: "WavePlus Humidity"
    co2:
      name: "WavePlus CO2"
    tvoc:
      name: "WavePlus VOC"

ble_client:
  - mac_address: redacted
    id: airthings01

esp32_ble_tracker:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  networks:
   - ssid: !secret wifi_ssid1
     password: !secret wifi_password1
   - ssid: !secret wifi_ssid
     password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ble-Hub Fallback Hotspot"
    password: "redacted"

captive_portal:
    `

aldalen, 2023-06-26:
I am seeing the same issue, adding a name/id attribute to the base sensor config solved the problem.   Seems like something in the base sensor/binary_sensor config added the requirement for a name/id attribute.

dasb00ter, 2023-06-26:
> I am seeing the same issue, adding a name/id attribute to the base sensor config solved the problem. Seems like something in the base sensor/binary_sensor config added the requirement for a name/id attribute.

Sample yaml for my slow brain. What you say makes some sense but was it binary? There was a breaking change I remember and I had to move a sensor to its own "designation"

strikeir13, 2023-09-11:
Closing because this is no longer an issue as of 2023.8.x (potentially earlier versions of ESPHome, I'm not sure when it was resolved, just that it was).

closed 2023-09-11 by strikeir13