esphome/issues #4599: Waveshare 7.5 " keeps black after update to 2023.6.0

TomTesner, 2023-06-22:
### The problem

Waveshare display keeps complete black  after update to 2023.6.0. Display is trying to refresh but keeps black.
Tested on two seperate displays - issue the same. Esp32 device is reachable all the time.
Display model: Waveshare_epaper 7.50inV2

### Which version of ESPHome has the issue?

Firmware: 2023.6.0 (Jun 22 2023, 19:09:19)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.6.2

### What platform are you using?

ESP32

### Board

ESP 32 esp-wrover-kit

### Component causing the issue

Display

### Example YAML snippet

```yaml
esphome:
  name: mcu-displej75
  platform: ESP32
  board: esp-wrover-kit

spi:
  clk_pin: 13
  mosi_pin: 14

i2c:
  sda: 19
  scl: 21

display:
  - platform: waveshare_epaper
    id: displej_75
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 7.50inV2
    update_interval: 600s
```

Progaros, 2023-06-22:
same here
```
display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 4.20in
    rotation: 180°
    update_interval: never
    id: eink
    lambda: |
      if (id(text).state == "wifiqr") {
        it.qr_code(90, 40, id(wifiqr), Color(255,255,255), 8);
      }
      else if (id(text).state
```

matzman666, 2023-06-22:
I have the same issue (except that the display shows a white instead of a black screen) with my Inkplate 6. It works with 2023.5.5 but not with 2023.6.0.

domgrimm, 2023-06-22:
I'm seeing the same issue, doesn't matter what lambda I use.

khenderick, 2023-06-22:
See #4587, work on a fix seems to be in progress.

jesserockz, 2023-06-22:
Fixed in https://github.com/esphome/esphome/releases/tag/2023.6.1

closed 2023-06-22 by jesserockz

kbarnes3, 2023-06-22:
The quick workaround from the other thread is to add a black fill to the beginning of your display lambda:
```
it.fill(Color::BLACK);
```