esphome/issues #4602: Web server does not show AC modes anymore

WorlockM, 2023-06-23:
### The problem

Since yesterday I can't see any controls for my AC's. Serveral users reported this. I only see state "24" and not the controls like Cooling/Heating etc. I did not update the ESPHome version but all stopped working at the same time.

I have no problems with Home Assistant.

### Which version of ESPHome has the issue?

All versions

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

Tarik2142, 2023-06-26:
problem with this script
[image]

jamexcb, 2023-06-26:
I have the same issue too.

CamielFolmer, 2023-06-26:
Same issue here

wilfredktlr, 2023-06-28:
This is the same problem as I described in my issue (https://github.com/esphome/issues/issues/4606). The cause of the bug can be found under the `Additional information` section.

closed 2023-06-28 by jesserockz