esphome/issues #4603: ili9488 white looks Green/blueish

Wummeke, 2023-06-23:
### The problem

I just installed 2023.6.1, which includes [esphome#4849](https://github.com/esphome/esphome/pull/4849)
I actually never had white balance or color issues before, now my display looks like what this pull request actually should fix.
In other words: what wasn't broken before in my case, now is.
it worked fine with ESPHome 2023.5.5

### Which version of ESPHome has the issue?

2023.6.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

lolin_d32_pro

### Component causing the issue

ILI9xxx

### Example YAML snippet

```yaml
display:
  - id: my_display
    platform: ili9xxx
    model: ili9488
    cs_pin: 14
    dc_pin: 27
    reset_pin: 33
    auto_clear_enabled: true
```

lnicolas83, 2023-06-28:
Same problem with ugly display and colors no longer matching (white is no longer white).

This is apparently caused by the PR https://github.com/esphome/esphome/pull/4849

Please revert this bad commit.

Wummeke, 2023-06-30:
Making your settings optional would be a good solution I guess

closed 2023-07-05 by kbx81