esphome/issues #4604: Error at logs updating from 2023.6 to 2023.6.1

KitHubek, 2023-06-23:
### The problem

board: esp32dev; 
framework: espidf; 
platform: platformio/espressif32@5.3.0

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

Updating via HA addon ESPhome from 2023.6 to 2023.6.1

### Which version of ESPHome has the issue?

2023.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6

### What platform are you using?

ESP32

### Board

esp32dev - nodemcu

### Anything in the logs that might be useful for us?

```txt
Attempting uninstall: six
    Found existing installation: six 1.16.0
    Not uninstalling six at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'six'. No files were found to uninstall.

AND

  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.1.0
    Uninstalling pyparsing-3.1.0:
      Successfully uninstalled pyparsing-3.1.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.6.1 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Successfully installed cachecontrol-0.13.1 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 filelock-3.12.2 future-0.18.3 idf-component-manager-1.2.3 idna-3.4 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-1.26.16
Reading CMake configuration...
Generating assembly for certificate bundle...

AND

Compiling /data/esp32-bluetooth-proxy/.pioenvs/esp32-bluetooth-proxy/bootloader_support/src/bootloader_flash_config_esp32.o
In file included from src/esphome.h:71,
                 from src/main.cpp:3:
src/esphome/core/time.h:11:13: warning: 'bool esphome::is_leap_year(uint32_t)' declared 'static' but never defined [-Wunused-function]
 static bool is_leap_year(uint32_t year);
             ^~~~~~~~~~~~
src/esphome/core/time.h:13:16: warning: 'uint8_t esphome::days_in_month(uint8_t, uint16_t)' declared 'static' but never defined [-Wunused-function]
 static uint8_t days_in_month(uint8_t month, uint16_t year);
                ^~~~~~~~~~~~~
Compiling /data/esp32-bluetooth-proxy/.pioenvs/esp32-bluetooth-proxy/bootloader_support/src/bootloader_efuse.o
```

ssieb, 2023-06-23:
These are known issues that have already been filed.  The last part is fixed in 2023.6.2.

KitHubek, 2023-06-28:
sorry, but i now update from 2023.6.2 to 2023.6.3 and still see this error at log:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
Successfully installed cachecontrol-0.13.1 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 filelock-3.12.2 future-0.18.3 idf-component-manager-1.2.3 idna-3.4 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.31.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-1.26.16
esphome 2023.6.3 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
```

ssieb, 2023-06-28:
Duplicate of https://github.com/esphome/issues/issues/4517

closed 2023-06-28 by ssieb