esphome/issues #4606: Column "Actions" not showing with button control

wilfredktlr, 2023-06-23:
### The problem

When the only action you have is a button the `Actions` column does not show in the webserver GUI.

[image]

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.2

### What platform are you using?

ESP8266

### Board

Wemos D1 mini v4

### Component causing the issue

https://oi.esphome.io/v2/www.js

### Additional information

In https://github.com/esphome/esphome-webserver/blob/main/v2/esp-entity-table.ts#L62 the check for `this.control(entity).length` will result in `undefined` because https://github.com/esphome/esphome-webserver/blob/main/v2/esp-entity-table.ts#L204 does not return an array.

wilfredktlr, 2023-06-23:
FYI. this problem is caused by this commit (https://github.com/esphome/esphome-webserver/pull/29/commits/af61e52ff31d32d1d21a8d258eb495e09bf1c80d).

screek-workshop, 2023-06-27:
same problem here. 
i wonder why not make www.js to local inside esp32?

nijsni, 2023-06-27:
I have the same problems. Before version 2023.6.0 I was able to use the buttons in actions and now I can't see actions anymore.

nijsni, 2023-06-27:
My configuration 
  platform: ESP8266
  board: d1_mini

button:
  - platform: template
    name: "hybridepomp + 0,001"
    on_press:
      then:
        lambda: id(totalhybridepompusage) += 0.001;

 Home Assistant 2023.6.3 Supervisor 2023.06.2 Operating System 10.3 Frontend-versie: 20230608.0 - latest

nijsni, 2023-06-27:
Sorry, Trip5.It's not working for me. I'm not that skilled in this area. I need to save a file somewhere on my raspberry but where do I get it from.
 INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/watermeter.yaml...
INFO Detected timezone 'Europe/Amsterdam'
Failed config

web_server: [source /config/esphome/watermeter.yaml:38]
  port: 80
  ota: True
  include_internal: True
  
  Could not find file '/config/esphome/www/v2_www.js'. Please make sure it exists (full path: /config/esphome/www/v2_www.js).
  js_include: www/v2_www.js
  js_url: ''
  Can you help me further.

trip5, 2023-06-27:
Yeah you need to download that file (view as raw), save it somewhere, rename it as 'v2_www.js' then copy it over to the appropriate folder on your Pi.  Login to your Pi using WinSCP or similar.  Find your esphome folder.  For you looks like /config/esphome.  Make a folder called 'www' and then copy the file there.

nijsni, 2023-06-27:
Thanks, it works.

closed 2023-06-28 by jesserockz