esphome/issues #4607: Updated to the latest version but it will no longer compile.

paulkidd68, 2023-06-23:
### The problem

src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);

see below for compiler output

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.6.2 Supervisor 2023.06.2 Operating System 10.3 Frontend 20230608.0 - latest

### What platform are you using?

ESP32

### Board

esp32 dev

### Component causing the issue

Nextion display

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/panel.yaml...
INFO Detected timezone 'Europe/London'
INFO Generating C++ source...
INFO Compiling app...
Processing nspanel (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.4
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_commands.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_upload.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/sensor/nextion_sensor.cpp.o
In file included from src/esphome/components/nextion/nextion_commands.cpp:1:
src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);
                                   ^~~~~~~
In file included from src/esphome/components/nextion/nextion_upload.cpp:1:
src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);
                                   ^~~~~~~
src/esphome/components/nextion/nextion_commands.cpp:223:42: error: variable or field 'set_nextion_rtc_time' declared void
 void Nextion::set_nextion_rtc_time(time::ESPTime time) {
                                          ^~~~~~~
src/esphome/components/nextion/nextion_commands.cpp:223:42: error: 'ESPTime' is not a member of 'esphome::time'
src/esphome/components/nextion/nextion_commands.cpp:223:42: note: suggested alternative:
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/nextion/nextion.h:21,
                 from src/esphome/components/nextion/nextion_commands.cpp:1:
src/esphome/core/time.h:12:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_commands.cpp.o] Error 1
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_upload.cpp.o] Error 1
In file included from src/esphome/components/nextion/nextion.cpp:1:
src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);
                                   ^~~~~~~
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion.cpp.o] Error 1
========================= [FAILED] Took 13.90 seconds =========================
```

ssieb, 2023-06-23:
You didn't include your config.  You are almost certainly using an external component for the nextion which is no longer compatible.  (I see the "nspanel" in the path.)  You only needed that for one boot anyway.

closed 2023-06-23 by ssieb

paulkidd68, 2023-06-23:
Yes it was an external component causing the issue