esphome/issues #4608: Esp32_Improv not compiling

raidnet-ms, 2023-06-23:
### The problem

Esp32-c3 won't compile with 

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6

### What platform are you using?

ESP32-IDF

### Board

lolin_c3_mini

### Component causing the issue

esp32_improv

### Example YAML snippet

```yaml
esp32_improv:
  authorizer: none
```
```

### Anything in the logs that might be useful for us?

```txt
Dependency Graph
|-- noise-c @ 0.1.4
|-- Improv @ 1.2.3
Linking /data/esp32-c3/.pioenvs/esp32-c3/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/esp32-c3/.pioenvs/esp32-c3/src/esphome/components/esp32_ble/ble_advertising.o: in function `esphome::esp32_ble::BLEAdvertising::start()':
/config/esphome/build/esp32-c3/src/esphome/components/esp32_ble/ble_advertising.cpp:71: undefined reference to `esp_ble_gap_config_adv_data'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/build/esp32-c3/src/esphome/components/esp32_ble/ble_advertising.cpp:78: undefined reference to `esp_ble_gap_config_adv_data'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/build/esp32-c3/src/esphome/components/esp32_ble/ble_advertising.cpp:89: undefined reference to `esp_ble_gap_start_advertising'
collect2: error: ld returned 1 exit status
*** [/data/esp32-c3/.pioenvs/esp32-c3/firmware.elf] Error 1
========================= [FAILED] Took 65.20 seconds =========================
```

ssieb, 2023-06-24:
Try doing a clean build.  If that doesn't work, then paste your full config.

raidnet-ms, 2023-06-24:
> Try doing a clean build.  If that doesn't work, then paste your full config.

Hi Ssieb, I already found this suggestion in the forum so I already tryied to do so without success.

ssieb, 2023-06-24:
So paste your config.  I don't have any problem compiling it...

raidnet-ms, 2023-06-24:
I also have problems with i2c with idf but I will open another report later

raidnet-ms, 2023-06-24:
> So paste your config. I don't have any problem compiling it...

If I remove esp32_improv then the config works.

The board is the new esp32-c mini by wemos.

Mat931, 2023-06-25:
I can reproduce the compile error

ssieb, 2023-06-25:
I missed that it was for `esp-idf`.  I can reproduce it now, but it doesn't happen with the original ESP32, only the C3 and S3.

ssieb, 2023-06-25:
That specific error is only for improv, but adding the bluetooth proxy gives other missing symbols.

jesserockz, 2023-06-25:
Try add this to your YAML until it is released
```yaml
external_components:
  - source: github://pr#5008
    components: [esp32_ble]
```

raidnet-ms, 2023-06-26:
> Try add this to your YAML until it is released
> 
> ```yaml
> external_components:
>   - source: github://pr#5008
>     components: [esp32_ble]
> ```

this works great, thank you.

raidnet-ms, 2023-06-26:
Waiting for the release

closed 2023-06-26 by raidnet-ms