esphome/issues #4609: ESP32 doesn't complete boot with esp32_ble_tracker and baud_rate:0 on ESP32 SOLO

silverchris, 2023-06-24:
### The problem

Ran into a weird issue.
With the following config, my ESP32 (tested on a ESP32 solo, as well as an ESP32 with CONFIG_FREERTOS_UNICORE)
esphome fails to start, it does not respond to pings or trying to get the logs over API.

### Which version of ESPHome has the issue?

2023.6.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

esp-wrover-kit

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    sdkconfig_options:
       CONFIG_FREERTOS_UNICORE: y

wifi:
  ssid: ""
  password: ""

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
```

### Anything in the logs that might be useful for us?

```txt
Can't access the logs during this bug :)
```

### Additional information

To Troubleshoot, I started commenting parts of esp32_ble_tracker.cpp out. Everything seems fine until this line.
https://github.com/esphome/esphome/blob/eb145757e5240e60044305d7b57ce3b73c9d7f87/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp#L68

It looks like it might be some issue with xSemaphoreTake and only using a single core?

silverchris, 2023-06-24:
Using an on_boot automation seems to work around this issue, if you default continuous to off for the tracker, and then turn continuous on after everything is booted

```YAML
esphome:
  name: test

  on_boot:
    - priority: -100
      then:
        - esp32_ble_tracker.start_scan:
            continuous: true

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

wifi:
  ssid: ""
  password: ""

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
  scan_parameters:
    continuous: false
```

LordMyschkin, 2023-06-27:
I think I am onto something: I do not get any values from my ATC-thermohygrometers with this version; tried your workaround, but the scan_parameters.continuous: false gets ignored; 
is there a way to force that parameter

```
esp32_ble_tracker:  
  scan_parameters:  
    continuous: false
    active: false
  on_ble_advertise:
    - mac_address: A4:C1:38:C9:DD:16 #wohnz
      then:
        - lambda: |-
            for (auto data : x.get_manufacturer_datas()) {
              if(data.data.size()==6) {
                const int basenum = (int16_t(data.data[1]) << 16) + (int16_t(data.data[2]) << 8) + int16_t(data.data[3]);
                const float temperature = (basenum / 10000.0f);
                const float humidity = (basenum % 1000) / 10.0f;
                const float battery_level = uint16_t(data.data[4]) / 1.0f;
                int16_t rssi=x.get_rssi();
                id(wohnz_humidity).publish_state(humidity);
                id(wohnz_temperature).publish_state(temperature);
                id(wohnz_rssi).publish_state(rssi);
                id(wohnz_battery).publish_state(battery_level);
              }
            }
```

but the output keeps being
```
...
22:07:49][C][esp32_ble:238]: ESP32 BLE:
[22:07:49][C][esp32_ble:240]:   MAC address: 78:21:84:89:41:16
[22:07:49][C][esp32_ble:241]:   IO Capability: none
[22:07:49][C][esp32_ble_tracker:590]: BLE Tracker:
[22:07:49][C][esp32_ble_tracker:591]:   Scan Duration: 300 s
[22:07:49][C][esp32_ble_tracker:592]:   Scan Interval: 320.0 ms
[22:07:49][C][esp32_ble_tracker:593]:   Scan Window: 30.0 ms
[22:07:49][C][esp32_ble_tracker:594]:   Scan Type: PASSIVE
[22:07:49][C][esp32_ble_tracker:595]:   Continuous Scanning: True
[22:07:49][C][captive_portal:088]: Captive Portal:
[22:07:49][C][mdns:108]: mDNS:
[22:07:49][C][mdns:109]:   Hostname: ble-bridge
[22:07:49][C][ota:093]: Over-The-Air Updates:
[22:07:49][C][ota:094]:   Address: ble-bridge.local:3232
[22:07:49][C][api:138]: API Server:
[22:07:49][C][api:139]:   Address: ble-bridge.local:6053
[22:07:49][C][api:143]:   Using noise encryption: NO
```

myhomeiot, 2024-05-11:
@silverchris Can you try to set `CONFIG_ESP_TASK_WDT_TIMEOUT_S` option as enplaned [here](https://digiblur.com/wiki/ha/esphome-bluetooth-proxy-esp32solo). I have same problem with `Xiaomi Mi LED Desk Lamp 1S (MJTD01SYL)` but can't check it now because in case of failure it's will require to disassemble the lamp :-(
Thanks in advance.

```yaml
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```