esphome/issues #4610: upload-port option not working anymore

jlo88, 2023-06-24:
### The problem

Hi, 

I used the have this script to update all my nodes: 

```
f_list=$(ls *.yaml | xargs)
esphome run $f_list --no-logs --upload-port OTA &> update_all.log
```

Is there an alternative?

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.2

### What platform are you using?

ESP8266

ssieb, 2023-06-24:
I think you've mixed that up a bit.  You've never been able to put multiple config files on the command line.  What are you doing with `xargs` there?

jlo88, 2023-06-24:
I use the xargs to create a file-list with only spaces in between, like this: 

```
cat_fountain.yaml kitchen_light_counter_1.yaml kitchen_light_counter_2.yaml secrets.yaml shelly_bathroom.yaml shelly_bedroom.yaml shelly_diningroom.yaml shelly_hallway.yaml shelly_kitchen.yaml shelly_livingroom.yaml shelly_office.yaml shelly_toilet.yaml storage_room_light.yaml sunscreen.yaml
```

esphome starts to build with `esphome run $f_list --no-logs` but stops for each yaml file to ask me how it should upload, via USB or OTA. I would like to say beforehand always OTA. This did work in previous releases with the `--upload-port OTA` option if I remember correctly.

jlo88, 2023-06-25:
Using `esphome run $f_list --no-logs --device OTA` works.

closed 2023-06-25 by jlo88