esphome/issues #4612: Text_sensor state.c_str() doesn't work

paki81, 2023-06-24:
### The problem

i have 2 text_sensor and this two sensors not showing state on display after update to 2023.6.2

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.2

### What platform are you using?

ESP32

### Board

esp32-s3-devkitc-1

### Component causing the issue

text_sensor

### Example YAML snippet

```yaml
sensor:
  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.ingresso_sensore_temperature

  - platform: homeassistant
    id: outside_temperature
    entity_id: sensor.desk_am2301_temperature

  - platform: homeassistant
    id: batteria_posta_percent
    entity_id:  sensor.posta_livello_batteria_posta
    

text_sensor:
  - platform: homeassistant
    entity_id: binary_sensor.posta_mailbox
    name: "Posta State"
    id: posta
  - platform: homeassistant
    id: posta_oggi
    entity_id: counter.conta_posta
  - platform: homeassistant
    id: segnale_posta
    entity_id: sensor.mailbox_wifi_signal_percent
  

time:
  - platform: homeassistant
    id: esptime

i2c:
  sda: GPIO17
  scl: GPIO18
  scan: false
  frequency: 700khz

font:
  - file: 'slkscr.ttf'
    id: font1
    size: 12

  - file: 'BebasNeue-Regular.ttf'
    id: font2
    size: 30

  - file: 'BebasNeue-Regular.ttf'
    id: font4
    size: 12

  - file: 'arial.ttf'
    id: font3
    size: 14

  - file: 'materialdesignicons-webfont.ttf'
    id: icon_font_20
    size: 12
    glyphs: [
      "󰅣", # online
      "󰅤", # offline
      “󰔨“, #posta in arrivo 
      “󰶌“, #posta chiusa
      "󰶈", #posta aperta
      "󰍀", #lucchetto aperto
      "󰍁", #lucchetto chiuso
      "󰑫", #wifi
      "󰁺", #10
      "󰁻", #20
      "󰁼", #30
      "󰁽", #40
      "󰁾", #50
      "󰁿", #60
      "󰂀", #70
      "󰂁", #80
      "󰂂", #90
      "󰁹", #100
      "󰂃", #batt alert
      ]

  - file: 'materialdesignicons-webfont.ttf'
    id: icon_font_30
    size: 25
    glyphs: [
      "󰅣", # online
      "󰅤", # offline
      “󰔨“, #posta in arrivo 
      “󰶌“, #posta chiusa
      "󰶈", #posta aperta
      "󰍀", #lucchetto aperto
      "󰍁", #lucchetto chiuso
      "󰑫", #wifi
      "󰁺", #10
      "󰁻", #20
      "󰁼", #30
      "󰁽", #40
      "󰁾", #50
      "󰁿", #60
      "󰂀", #70
      "󰂁", #80
      "󰂂", #90
      "󰁹", #100
      "󰂃", #batt alert
      ]   

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: GPIO21
    address: 0x3C
    id: my_display
    lambda: |-

     
      it.printf(0, 60, id(font2), TextAlign::BASELINE_LEFT,  "Posta:");

      if (id(posta).state == "unavailable") {
        it.printf(110, 3, id(icon_font_20), TextAlign::TOP_LEFT, "󰅤" );
      } else {
         it.printf(110, 3, id(icon_font_20), TextAlign::TOP_LEFT, "󰅣" ); 
      }
      it.printf(54, 3, id(icon_font_20), TextAlign::TOP_LEFT, "󰑫" ); 

       if (id(segnale_posta).state == "unavailable") {
        it.printf(70, 0, id(icon_font_20), TextAlign::TOP_LEFT, "󰅤" );
      } 
        it.printf(70, 0, id(font3), TextAlign::TOP_LEFT , "%s%%", id(segnale_posta).state.c_str());
      
      
      it.printf(75, 60, id(font2), TextAlign::BASELINE_LEFT , "%s", id(posta_oggi).state.c_str());

      it.line(45, 0, 45, 20);
      it.line(102, 0, 102, 20);
      it.line(0, 20, 128, 20);

      // Print time in HH:MM format
      it.strftime(0, 0, id(font3), TextAlign::TOP_LEFT, "%H:%M", id(esptime).now());

      // Print inside temperature (from homeassistant sensor)
      it.printf(122, 23, id(font3), TextAlign::TOP_RIGHT ,"%.0f" , id(batteria_posta_percent).state);
      

      if (isnan(id(batteria_posta_percent).state)) {
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰂃" );
      } else {
      int percentuale = id(batteria_posta_percent).state;
      if (percentuale < 1 || percentuale > 110) {
         it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰂃" );
      } else if (percentuale <= 10) {
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁺" );
      } else if (percentuale <= 20) {
        // La percentuale è compresa tra 11 e 20
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁻" );
      } else if (percentuale <= 30) {
        // La percentuale è compresa tra 21 e 30
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁼" );
      } else if (percentuale <= 40) {
        // La percentuale è compresa tra 31 e 40
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁽" );
      } else if (percentuale <= 50) {
        // La percentuale è compresa tra 41 e 50
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁾" );
      } else if (percentuale <= 60) {
        // La percentuale è compresa tra 51 e 60
         it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁿" );
      } else if (percentuale <= 70) {
        // La percentuale è compresa tra 61 e 70
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰂀" );
      } else if (percentuale <= 80) {
        // La percentuale è compresa tra 71 e 80
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰂁" );
      } else if (percentuale <= 90) {
        // La percentuale è compresa tra 81 e 90
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰂂" );
      } else {
        // La percentuale è compresa tra 91 e 110
        it.printf(127, 60, id(icon_font_30), TextAlign::BASELINE_RIGHT, "󰁹" );
      }
      }
```

ssieb, 2023-06-25:
Are the other sensors working?  What do you see?  What's in the logs?

paki81, 2023-06-25:
No errors in the logs

The others sensor are working.

In the display i see everything but not this 2 line of the text_sensor:

```
it.printf(70, 0, id(font3), TextAlign::TOP_LEFT , "%s%%", id(segnale_posta).state.c_str());

it.printf(75, 60, id(font2), TextAlign::BASELINE_LEFT , "%s", id(posta_oggi).state.c_str());
```
[image]

ssieb, 2023-06-25:
`nan` doesn't look like it's working.  Did you add the device as an HA integration?

paki81, 2023-06-25:
Is the battery sensor and in the pic i powered the ESP from the cable without the battery.

ssieb, 2023-06-25:
But I don't see any indication there that your other sensors are working.  Did you add the device as an HA integration?

paki81, 2023-06-25:
> But I don't see any indication there that your other sensors are working. Did you add the device as an HA integration?

I added the device from the ADOPT button in ESPHome and then i write wirelesy the firmware.

Before the 2023.5.5 everything worked perfectly.

ssieb, 2023-06-25:
You added it to the dashboard, but you haven't added it to HA yet.  If you need more help, come ask on the Discord server.

closed 2023-06-25 by ssieb

paki81, 2023-06-25:
You right! Thank you. That's solved my problem.