esphome/issues #4613: [ESP8285/ESP8285] final linking failure: liblto_plugin.so: file too short

hekmon, 2023-06-24:
### The problem

I can not update my smart plug firmware for 2 versions now (2023.5.x and 2023.6.x). Final linking fails on ` liblto_plugin.so` with the error `file too short`. More details below.

### Which version of ESPHome has the issue?

at least v2023.5.X & v2023.6.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

esp8285

### Example YAML snippet

```yaml
substitutions:
  devicename: "smart-plug"
  friendlyname: "Smart Plug"
  # Higher value gives lower watt readout
  current_res: "0.00237"
  # Lower value gives lower voltage readout
  voltage_div: "731"

esphome:
  name: $devicename
  friendly_name: $friendlyname
  comment: "Nous Smart Wifi Socket A1T (Tasmota)"
  project:
    name: "NOUS.Smart-Wifi-Socket"
    version: "A1T"

esp8266:
  board: esp8285
  restore_from_flash: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<redacted>"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicename}-AP"
    password: !secret wifi_ap_password

captive_portal:

# see: https://esphome.io/components/time.html
time:
  - platform: homeassistant
    id: homeassistant_time

# Enable Web server
[... 387 lines cut ...]
Compiling .pioenvs/smart-plug/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/sqrt32.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/stdlib_noniso.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/time.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/uart.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pioenvs/smart-plug/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pioenvs/smart-plug/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pioenvs/smart-plug/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pioenvs/smart-plug/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving .pioenvs/smart-plug/libFrameworkArduino.a
Linking .pioenvs/smart-plug/firmware.elf
/config/.esphome/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /config/.esphome/platformio/packages/toolchain-xtensa/bin/../libexec/gcc/xtensa-lx106-elf/10.3.0/liblto_plugin.so: error loading plugin: /config/.esphome/platformio/packages/toolchain-xtensa/bin/../libexec/gcc/xtensa-lx106-elf/10.3.0/liblto_plugin.so: file too short
collect2: error: ld returned 1 exit status
*** [.pioenvs/smart-plug/firmware.elf] Error 1
========================= [FAILED] Took 152.44 seconds =========================
```

### Additional information

The board is currently running with an ESPHome firmware of version 2023.4.4.

Mat931, 2023-06-25:
Can't reproduce the error. You can try "Clean build files" and delete the `.esphome` folder.

hekmon, 2023-06-25:
I already tried the clean build files with no luck. But deleting the entire `.esphome` did the trick, thx !

closed 2023-06-25 by hekmon