esphome/issues #4614: use_address not honored when mDNS disabled

phoenixtechnam, 2023-06-24:
### The problem

Thanks for all the great work so far.

After updating ESPHome, I am unable to update my devices because of IP resolution issues.

mDNS is disabled in all my configs. Instead I utliize the use_address option for Ethernet and WIFI.

But use_address is not honored anymore with the recent update. Instead, ESPHome tries to get the IP from MQTT which fails because the devices do not yet make use of the topic esphome/ping/device_name. 

I updated one device manually now and afterwards it subscribes to esphome/ping/device_name and IP resolution succeeds.

Nevertheless, I believe use_address config option for ethernet: and wifi: should still be honored with top priority. 

Furthermore, I cannot find any documentation about the available esphome/* topics, maybe someone can help me there ?

### Which version of ESPHome has the issue?

2023.6.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

mDNS, MQTT, Ethernet, WIFI

### Example YAML snippet

```yaml
ethernet:
  domain: .iot
  use_address: 10.1.2.3
  type: LAN8720
  mdc_pin: GPIO16
  mdio_pin: GPIO17
  clk_mode: GPIO0_IN
  phy_addr: 0

mqtt:
  broker: !secret mqtt_ip
  username: !secret mqtt_username
  password: !secret mqtt_password
  client_id: $device_name
  discovery: true
  discovery_retain: true
  topic_prefix: $device_name
```

DJ1975-SE, 2023-07-02:
Seems to remain in 2023.6.3

# Which version of ESPHome has the issue?
2023.6.3

# What type of installation are you using?
python venv / esphome 2023.6.3 via pip on fedora 38

#Which version of Home Assistant has the issue?
Not used

#What platform are you using?
ESP32

#Board
AI-Thinker ESP32CAM

#Component causing the issue
suspecting config parsing in main esphome binary

Example YAML snippet (esphome config output using 2023.5.5)

substitutions:
  devicename: scrubbed
  cameraname: scrubbed
  hass_api_name: scrubbed
  my_ip: scrubbed
  my_gateway: scrubbed
  my_subnet: scrubbed
  mytopic: esp32camsleep/scrubbed
  my_topic_prefix: esp32camsleep
  my_otatopic: esp32camsleep/ota_mode
  runduration: 2min
  sleepduration: 13min
  my_payload: scrubbed
  swname: scrubbed
esphome:
  name: scrubbed
  project:
    name: scrubbed
    version: scrubbed
  build_path: .esphome/build/scrubbed
  friendly_name: ''
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
  min_version: 2023.5.5
esp32:
  board: esp32cam
  framework:
    version: 2.0.5
    source: ~3.20005.0
    platform_version: platformio/espressif32@5.3.0
    type: arduino
  variant: ESP32
deep_sleep:
  run_duration: 2min
  sleep_duration: 13min
  id: powersaver
wifi:
  fast_connect: true
  manual_ip:
    static_ip: scrubbed
    gateway: scrubbed
    subnet: scrubbed
    dns1: 0.0.0.0
    dns2: 0.0.0.0
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  passive_scan: false
  networks:
  - ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    priority: 0.0
  use_address: scrubbed
mdns:
  disabled: true
  services: []
logger:
  level: INFO
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  logs: {}
api:
  encryption:
    key: !secret 'api_encrkey'
  reboot_timeout: 480min
  id: scrubbed
  port: 6053
  password: ''
ota:
  safe_mode: true
  password: !secret 'ota_password'
  on_begin:
  - then:
    - logger.log:
        format: OTA start
        level: DEBUG
        tag: main
        args: []
    - deep_sleep.prevent:
        id: powersaver
  on_progress:
  - then:
    - logger.log:
        format: OTA progress %0.1f%%
        args:
        - !lambda |-
          x
        level: DEBUG
        tag: main
  on_error:
  - then:
    - logger.log:
        format: OTA update error %d
        args:
        - !lambda |-
          x
        level: DEBUG
        tag: main
  on_end:
  - then:
    - logger.log:
        format: OTA end
        level: DEBUG
        tag: main
        args: []
  port: 3232
  reboot_timeout: !secret 'updateinterval'
  num_attempts: 10
esp32_camera:
  name: scrubbed
  external_clock:
    pin: 0
    frequency: 8000000.0
  i2c_pins:
    sda: 26
    scl: 27
  data_pins:
  - 5
  - 18
  - 19
  - 21
  - 36
  - 39
  - 34
  - 35
  vsync_pin: 25
  href_pin: 23
  pixel_clock_pin: 22
  power_down_pin: 32
  resolution: 1280X1024
  max_framerate: 1.0
  idle_framerate: 3.0e-05
  internal: false
  jpeg_quality: 10
  contrast: 0
  brightness: 0
  saturation: 0
  special_effect: NONE
  aec_mode: AUTO
  ae_level: 0
  aec_value: 300
  agc_mode: AUTO
  agc_value: 0
  wb_mode: CLOUDY
  disabled_by_default: false
  vertical_flip: true
  horizontal_mirror: true
  aec2: false
  agc_gain_ceiling: 2X
  test_pattern: false
sensor:
- platform: wifi_signal
  name: WiFi RSSI
  update_interval: !secret 'updateinterval'
  internal: true
  disabled_by_default: false
  force_update: false
  unit_of_measurement: dBm
  accuracy_decimals: 0
  device_class: signal_strength
  state_class: measurement
  entity_category: diagnostic
text_sensor:
- platform: wifi_info
  mac_address:
    name: ESP Wifi MAC
    internal: false
    retain: false
    disabled_by_default: false
    entity_category: diagnostic
mqtt:
  broker: !secret 'mqtt_brokerip'
  username: scrubbed
  password: !secret 'mqtt_password'
  discovery: false
  topic_prefix: esp32camsleep
  birth_message:
    topic: esp32camsleep/scrubbed
    payload: scrubbed
    qos: 0
    retain: true
  will_message: null
  shutdown_message: null
  reboot_timeout: 0s
  keepalive: 30s
  log_topic:
    topic: esp32camsleep
    level: NONE
    qos: 0
    retain: true
  on_message:
  - topic: esp32camsleep/ota_mode
    payload: 'ON'
    then:
    - deep_sleep.prevent:
        id: powersaver
    qos: 0
  port: 1883
  discovery_retain: true
  discovery_prefix: homeassistant
  discovery_unique_id_generator: legacy
  discovery_object_id_generator: none
  use_abbreviations: true
button:
- platform: restart
  name: scrubbed
  disabled_by_default: false
  entity_category: config
  device_class: restart

afestbaum, 2023-07-10:
Gonna 2nd that issue. This is blocking me from remotely updating my devices right now.

randybb, 2023-07-10:
Don't understand what are you trying to achieve. `use_address:` is only for esphome dashboard - so if mdns is not available, then you need to use this parameter to tell dashboard which address it has to use for that device. If esphome device gets a different address from DHCP and it is not matching, then it will not work.

phoenixtechnam, 2023-07-10:
> Don't understand what are you trying to achieve. `use_address:` is only for esphome dashboard - so if mdns is not available, then you need to use this parameter to tell dashboard which address it [...]

you are missing the point that OTA update fails because ESPHome tries to lookup the IP address via MQTT when MQTT is enabled. But the MQTT lookups fail because older firmware did not support PING via MQTT. The use_address parameter is not utilized for OTA anymore if MQTT is enabled which I think is a problem (e.g. I cannot update my devices currently via ESPHome OTA)

randybb, 2023-07-10:
ESPHome Dashboard is not using MQTT. It just connects to the set address via port 8266.

phoenixtechnam, 2023-07-10:
> ESPHome Dashboard is not using MQTT. It just connects to the set address via port 8266.

you are missing the issue again: 
the Dashboard reports the devices as ONLINE, but OTA update from dashboard fails because the OTA procedure does not honor use_address and instead tries to (re-)discover the IPs via MQTT Ping.

randybb, 2023-07-10:
Ahhh. Totally missed that https://github.com/esphome/esphome/pull/3887 which implemented mqtt for the dashboard 😬

Did some testing and indeed, this is not working as expected as `use_address:` has to override discovery via mqtt (and mdns where it works).

Have added
```yaml
wifi:
  use_address: 1.1.1.1
```
but ota and logging keeps using the discovered address via mqtt
```
========================= [SUCCESS] Took 34.08 seconds =========================
INFO Successfully compiled program.
INFO Starting looking for IP in topic esphome/discover/mcu-atoms3
INFO Connected to MQTT broker!
INFO Send discover via MQTT broker topic: esphome/ping/mcu-atoms3
INFO Found IP: 10.17.240.52
INFO Connecting to 10.17.240.52
INFO Uploading /data/mcu-atoms3/.pioenvs/mcu-atoms3/firmware.bin (1449104 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting looking for IP in topic esphome/discover/mcu-atoms3
INFO Connected to MQTT broker!
INFO Send discover via MQTT broker topic: esphome/ping/mcu-atoms3
INFO Found IP: 10.17.240.52
INFO Starting log output from 10.17.240.52 using esphome API
INFO Successfully connected to 10.17.240.52
[00:47:42][I][app:102]: ESPHome version 2023.7.0-dev compiled on Jul 11 2023, 00:46:54
[00:47:42][C][wifi:543]: WiFi:
[00:47:42][C][wifi:379]:   Local MAC: DC:54:75:C7:0C:C0
[00:47:42][C][wifi:380]:   SSID: [redacted]
[00:47:42][C][wifi:381]:   IP Address: 10.17.240.52
```
Only way is to "enable" mdns so dashboard stops using mqtt discovery and it finally takes the address configured via `use_address:`.

MoHf7f4, 2023-07-11:
I'm having the same problem.  What am I supposed to publish to mqtt so I can use mqtt again?  I needed to remove it to get it to install. Couldn't find anything in the documentation.

jesserockz, 2023-07-12:
I have made a [PR](https://github.com/esphome/esphome/pull/5096) (that unfortunately cannot be used as an external_component due to the actual changes) that fixes this.

<details>
<summary>With use_address</summary>

```yaml
esphome:
  name: mqtt-ip-test

esp32:
  board: nodemcu-32s

ota:
logger:

mdns:
  disabled: true

mqtt:
  broker: 1.1.1.1
  username: username
  password: password

wifi:
  ssid: "ssid"
  password: "password"
  use_address: 10.99.99.99
```

```
================================================================================ [SUCCESS] Took 3.01 seconds ================================================================================
INFO Successfully compiled program.
INFO Connecting to 10.99.99.99
```
</details>

<details>
<summary>Without use_address</summary>

```yaml
esphome:
  name: mqtt-ip-test

esp32:
  board: nodemcu-32s

ota:
logger:

mdns:
  disabled: true

mqtt:
  broker: 1.1.1.1
  username: username
  password: password

wifi:
  ssid: "ssid"
  password: "password"
```

```
================================================================================ [SUCCESS] Took 6.53 seconds ================================================================================
INFO Successfully compiled program.
INFO Starting looking for IP in topic esphome/discover/mqtt-ip-test
```

</details>

closed 2023-07-13 by kbx81