esphome/issues #4615: pi pico w doesn't read DHT22

SDMatt2020, 2023-06-24:
### The problem

My pi pico w isn't getting data from a DHT22.  If I put micropython on the pi pico w, it reads the sensor as expected.

This is on ESPHome 2023.6.2

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

RP2040

### Board

pi pico w

### Component causing the issue

dht sensor

### Example YAML snippet

```yaml
sensor:
  - platform: dht
    pin: 5
    temperature:
      name: "DEV - Living Room Temperature"
    humidity:
      name: "DEV - Living Room Humidity"
    model: DHT22
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/pico-dev.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing pico-dev (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.4
|-- MD5Builder @ 1.0.0
Compiling /data/pico-dev/.pioenvs/pico-dev/src/esphome/components/api/api_connection.cpp.o
Compiling /data/pico-dev/.pioenvs/pico-dev/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/pico-dev/.pioenvs/pico-dev/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/pico-dev/.pioenvs/pico-dev/src/esphome/components/api/api_pb2_service.cpp.o
[... 135 lines cut ...]
[16:11:00][D][sensor:093]: 'DEV - Living Room Temperature': Sending state inf °C with 1 decimals of accuracy
[16:11:00][D][sensor:093]: 'DEV - Living Room Humidity': Sending state inf % with 0 decimals of accuracy
```

### Additional information

here's the micropython code I used to check the functionality of the setup

from machine import Pin
from time import sleep
import dht

sensor = dht.DHT22(Pin(5))

while True:
    try:
        sleep(1)
        sensor.measure()
        # Retrieves measurements from the sensor
        print(f"Temperature : {sensor.temperature():.1f}")
        print(f"Humidity    : {sensor.humidity():.1f}")
        # Transmits the temperature to the terminal
    except OSError as e:
        print('Failed reception')
        # If the pico does not receive the measurements from the sensor

SDMatt2020, 2023-06-24:
I forgot to mention.  I've tried this with no resister, a 5K resister, and a 10K resistor (as the pull up).  I've also tried running this at 3.3 and 5 volts, as I came across a post somewhere that suggested the DHT22 might run better at 5v.

Mat931, 2023-06-25:
Looks like the delayMicroseconds function is the problem.

https://forum.seeedstudio.com/t/ds18b20-temperature-sensor-issue/263525

Try adding this to your config:
```
external_components:
  - source: github://Mat931/esphome@rp2040-dht-fix
    components: [ dht ]
    refresh: always
```

SDMatt2020, 2023-06-26:
> Looks like the delayMicroseconds function is the problem.
> 
> https://forum.seeedstudio.com/t/ds18b20-temperature-sensor-issue/263525
> 
> Try adding this to your config:
> 
> ```
> external_components:
>   - source: github://Mat931/esphome@rp2040-dht-fix
>     components: [ dht ]
>     refresh: always
> ```

Thanks, but that didn't help.

Mat931, 2023-06-26:
Thanks for testing. I'm still waiting for my Pico W. Once it arrives I can take a look with the logic analyzer

Mat931, 2023-06-28:
My Pico W just arrived and it can read the DHT22 (AM2302) sensor just fine, both with and without the "fix". I'm using a 10k pull-up resistor.

SDMatt2020, 2023-06-28:
I wonder if there's some manufacturing variance in the DHT22s, and perhaps the micropython library is more forgiving.  I'll also put a meter on the resister to verify its resistance, but I doubt the resistor is the issue here.

Where did you get your DHT22?  If it's a generic Amazon/AliExpress part can you post a link (unless that's against some Github policy)?

Mat931, 2023-06-28:
Sorry, my sensor is at least 5 years old and I have no clue where I bought it. I can do some more testing with MicroPython vs. ESPHome later. By the way I saw the same error message for a while and thought I reproduced the issue but turns out the sensor didn't make good contact to the wires.

Mat931, 2023-06-28:
MicroPython:
[image]

ESPHome:
[image]

In MicroPython there's a longer delay before the data is transmitted

Mat931, 2023-06-28:
@SDMatt2020 Can you try configuring other DHT models in yaml?
Available models are: DHT11, DHT22, DHT22_TYPE2, AM2302, RHT03, SI7021

SDMatt2020, 2023-06-29:
I just tried all those models without success.

SDMatt2020, 2023-07-03:
At this point I'm planning on buying some more sensors.  I'm betting there's something weird about these DHT22s that I have.  I have an application that should work fine with the cheaper DHT11s, so I plan on buying some from a reputable seller in hopes they'll work with ESPHome.

The current DHT22s I have were sold under the DWEII brand name.  They are 3 pin sensors, rather than the 4 pin that you commonly see from sellers like adafruit.

Fixitman333, 2023-07-04:
I'm thinking this is related to manufacturing differences vs. delay before data transmission. Some sensors "wake up" quickly, while others are slow.

I just tested my DHT11 with micropython, and it wouldn't work. I added a 2 second delay after creating the sensor object, and it worked normally. I haven't found a way to add a delay using esphome, so my testing has come to a halt.

kdpuvvadi, 2024-06-16:
Seems like it's still issue. Works file with micropython library but but throwing error on esphome. though.

closed 2025-02-16 by jesserockz