esphome/issues #4616: ESPHome Firmware Upgrade causing time parsing issues

evakq8r, 2023-06-25:
### The problem

Upgraded ESPHome from 2023.5.5 to 2023.6.2 in Home Assistant, and my m5stack-esp32 board no longer compiles code successfully.

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

11.0.dev20230622

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

powerpal_ble

### Example YAML snippet

```yaml
esphome:
  name: "powerpal-gateway"

esp32:
  board: esp32dev

external_components:
  - source:
      type: local
      path: my_components
    components: [powerpal_ble]
# Enable logging
logger:
 level: VERY_VERBOSE
 logs:
  esp32_ble_tracker: WARN
# Enable Home Assistant API
api:
  encryption:
    key: <key>

ota:

#esp32_ble_tracker:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# optional requirement used with daily energy sensor
time:
  - platform: homeassistant
    id: homeassistant_time

esp32_ble_tracker:

ble_client:
  - mac_address: !secret powerpal_mac_address
    id: powerpal

#http_request:
 #id: powerpal_cloud_uploader

switch:
  - platform: restart
    name: "Powerpal Monitor Reboot"

sensor:
  - platform: powerpal_ble
    ble_client_id: powerpal
    power:
      name: "Powerpal Power"
    daily_energy:
      name: "Powerpal Daily Energy"
    energy:
      name: "Powerpal Total Energy"
    battery_level:
      name: "Powerpal Battery"
    watt_hours:
       name: "Powerpal Watt Hours_json"
    cost:
       name: "Powerpal Cost_json"
       unit_of_measurement: "$/kWh"
    timestamp:
       name: "Powerpal Timestamp_json"
    pulses:
       name: "Powerpal Pulses_json"
    pairing_code: !secret powerpal_pairing_code
    notification_interval: 1
    pulses_per_kwh: 1000
    #http_request_id: powerpal_cloud_uploader
    time_id: homeassistant_time # daily energy still works without a time_id, but recommended to include one to properly handle daylight savings, etc.
    cost_per_kwh: <number> #dollars per kWh
    
    #powerpal_device_id:  #optional, component will retrieve from your Powerpal if not set
    #powerpal_apikey:  #optional, component will retrieve from your Powerpal if not set
```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/powerpal_ble/powerpal_ble.cpp: In member function 'void esphome::powerpal_ble::Powerpal::parse_measurement_(const uint8_t*, uint16_t)':
src/esphome/components/powerpal_ble/powerpal_ble.cpp:107:13: error: 'ESPTime' is not a member of 'esphome::time'
       time::ESPTime date_of_measurement = time_->now();
             ^~~~~~~
src/esphome/components/powerpal_ble/powerpal_ble.cpp:107:13: note: suggested alternative:
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/homeassistant/time/homeassistant_time.h:4,
                 from src/esphome.h:30,
                 from src/esphome/components/powerpal_ble/powerpal_ble.h:8,
                 from src/esphome/components/powerpal_ble/powerpal_ble.cpp:1:
src/esphome/core/time.h:12:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
src/esphome/components/powerpal_ble/powerpal_ble.cpp:108:11: error: 'date_of_measurement' was not declared in this scope
       if (date_of_measurement.is_valid()) {
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/powerpal_ble/powerpal_ble.cpp:108:11: note: suggested alternative: 'parse_measurement_'
       if (date_of_measurement.is_valid()) {
           ^~~~~~~~~~~~~~~~~~~
           parse_measurement_
Compiling /data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/switch/switch.cpp.o
Compiling /data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/time/automation.cpp.o
Compiling /data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/time/real_time_clock.cpp.o
*** [/data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/powerpal_ble/powerpal_ble.cpp.o] Error 1
========================== [FAILED] Took 9.81 seconds ==========================
```

### Additional information

OS: Unraid 6.12.1
Home Assistant Setup: HAOS on an Unraid VM

HA prompted an upgrade to 2023.6.2 (was on 2023.5.5). I have 14 ESPHome power plugs (Athom Tech) that upgraded successfully, however my m5stack-esp32 board refuses to due to the errors above. The full YAML used (minus the encryption key) is posted in this thread.

Board: https://shop.m5stack.com/products/atom-lite-esp32-development-kit

The m5stack board is used for power monitoring through Powerpal. Setup guide here: https://tatham.blog/2022/07/16/smart-home-energy-monitoring-in-melbourne-australia/

Was working fine until ESPHome upgraded to 2023.6.2. 

Logs from HA:

```
Logger: homeassistant.components.esphome.update
Source: components/esphome/update.py:179
Integration: ESPHome ([documentation](https://next.home-assistant.io/integrations/esphome), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+esphome%22))
First occurred: 13:40:44 (1 occurrences)
Last logged: 13:40:44
```

Error OTA updating powerpal-gateway.yaml. Try again in ESPHome dashboard for error`

Logs from ESPHome:

```
src/esphome/components/powerpal_ble/powerpal_ble.cpp: In member function 'void esphome::powerpal_ble::Powerpal::parse_measurement_(const uint8_t*, uint16_t)':
src/esphome/components/powerpal_ble/powerpal_ble.cpp:107:13: error: 'ESPTime' is not a member of 'esphome::time'
       time::ESPTime date_of_measurement = time_->now();
             ^~~~~~~
src/esphome/components/powerpal_ble/powerpal_ble.cpp:107:13: note: suggested alternative:
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/homeassistant/time/homeassistant_time.h:4,
                 from src/esphome.h:30,
                 from src/esphome/components/powerpal_ble/powerpal_ble.h:8,
                 from src/esphome/components/powerpal_ble/powerpal_ble.cpp:1:
src/esphome/core/time.h:12:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
src/esphome/components/powerpal_ble/powerpal_ble.cpp:108:11: error: 'date_of_measurement' was not declared in this scope
       if (date_of_measurement.is_valid()) {
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/powerpal_ble/powerpal_ble.cpp:108:11: note: suggested alternative: 'parse_measurement_'
       if (date_of_measurement.is_valid()) {
           ^~~~~~~~~~~~~~~~~~~
           parse_measurement_
Compiling /data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/switch/switch.cpp.o
Compiling /data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/time/automation.cpp.o
Compiling /data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/time/real_time_clock.cpp.o
*** [/data/powerpal-gateway/.pioenvs/powerpal-gateway/src/esphome/components/powerpal_ble/powerpal_ble.cpp.o] Error 1
========================== [FAILED] Took 9.81 seconds ==========================
```

Mat931, 2023-06-25:
Try removing `time::` in front of `time::ESPTime` in the file `powerpal_ble.cpp`. If there are still any errors please share the code of your powerpal_ble component.

randybb, 2023-06-25:
It was mentioned as a breaking change in [2023.6.0](https://esphome.io/changelog/2023.6.0.html#id1), so the component needs to be updated.

evakq8r, 2023-06-25:
> Try removing `time::` in front of `time::ESPTime` in the file `powerpal_ble.cpp`. If there are still any errors please share the code of your powerpal_ble component.

That did the job, thanks very much! Firmware upgrade has now completed successfully.

evakq8r, 2023-06-25:
> It was mentioned as a breaking change in [2023.6.0](https://esphome.io/changelog/2023.6.0.html#id1), so the component needs to be updated.

Ah, I see. I did glance through the change notes but overlooked this. Thanks for the info.

evakq8r, 2023-06-25:
Closing issue as https://github.com/esphome/issues/issues/4616#issuecomment-1605995950 fixed the problem. :)

closed 2023-06-25 by evakq8r