esphome/issues #4617: ESPHome Firmware upgrade - ESPTime issue

stahelp, 2023-06-25:
### The problem

Upgraded ESPHome from 2023.5.5 to 2023.6.2 in Home Assistant, and my code for the nspanel is brocken.

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32

### Board

ESP32 in NSPanel EU version

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/nspanel.yaml...
INFO Detected timezone 'Europe/Zurich'
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing nspanel (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_commands.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_upload.cpp.o
Compiling /data/nspanel/.pioenvs/nspanel/src/esphome/components/ntc/ntc.cpp.o
In file included from src/esphome/components/nextion/nextion_commands.cpp:1:
src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);
                                   ^~~~~~~
src/esphome/components/nextion/nextion_commands.cpp:223:42: error: variable or field 'set_nextion_rtc_time' declared void
 void Nextion::set_nextion_rtc_time(time::ESPTime time) {
                                          ^~~~~~~
src/esphome/components/nextion/nextion_commands.cpp:223:42: error: 'ESPTime' is not a member of 'esphome::time'
src/esphome/components/nextion/nextion_commands.cpp:223:42: note: suggested alternative:
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/nextion/nextion.h:21,
                 from src/esphome/components/nextion/nextion_commands.cpp:1:
src/esphome/core/time.h:12:8: note:   'esphome::ESPTime'
 struct ESPTime {
        ^~~~~~~
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_commands.cpp.o] Error 1
In file included from src/esphome/components/nextion/nextion.cpp:1:
src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);
                                   ^~~~~~~
In file included from src/esphome/components/nextion/nextion_upload.cpp:1:
src/esphome/components/nextion/nextion.h:324:35: error: 'esphome::time::ESPTime' has not been declared
   void set_nextion_rtc_time(time::ESPTime time);
                                   ^~~~~~~
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion.cpp.o] Error 1
*** [/data/nspanel/.pioenvs/nspanel/src/esphome/components/nextion/nextion_upload.cpp.o] Error 1
========================== [FAILED] Took 8.22 seconds ==========================
```

krouton24, 2023-06-25:
Getting the same issue with v2023.6.2

```
src/esphome/components/tuya/tuya.cpp: In member function 'void esphome::tuya::Tuya::send_local_time_()':
src/esphome/components/tuya/tuya.cpp:465:9: error: 'ESPTime' is not a member of 'esphome::time'; did you mean 'esphome::ESPTime'?
  465 |   time::ESPTime now = time_id->now();
      |         ^~~~~~~
In file included from src/esphome/components/time/real_time_clock.h:8,
                 from src/esphome/components/tuya/tuya.h:9,
                 from src/esphome/components/tuya/tuya.cpp:1:
src/esphome/core/time.h:12:8: note: 'esphome::ESPTime' declared here
   12 | struct ESPTime {
      |        ^~~~~~~
src/esphome/components/tuya/tuya.cpp:466:7: error: 'now' was not declared in this scope; did you mean 'pow'?
  466 |   if (now.is_valid()) {
      |       ^~~
      |       pow
*** [/data/bedskyfanesp01/.pioenvs/bedskyfanesp01/src/esphome/components/tuya/tuya.cpp.o] Error 1
```

Mat931, 2023-06-25:
Can you share the yaml code?

stahelp, 2023-06-25:
substitutions:
  name: nspanel
  friendly_name: "NSPanel Loft"
  location: Loft

esphome:
  name: $name
  comment: touchscreen loft

esp32:
  board: esp32dev

# Wifi settings. Add these to your secrets.yaml. fast_connect must be true for a hidden ssid.
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
 services:
    # Service to play a song
  - service: play_rtttl
    variables:
      song_str: string
    then:
      - rtttl.play:
          rtttl: !lambda 'return song_str;'
  # Service to upload a new screen file (TFT)
  - service: upload_tft
    then:
      - lambda: 'id(disp1)->upload_tft();'
  - service: send_command
    variables:
      cmd: string
    then:
      - lambda: 'id(disp1).send_command_printf("%s", cmd.c_str());'
  # Service to control the visible page from Home Assistant
  - service: set_page
    variables:
      page: int
    then:
      - lambda: 'id(disp1).send_command_printf("page %i", page);'
  # Service to show a notification on the screen for 20 s. Tap or wait to close
  - service: notification
    variables:
      header: string
      message: string
    then:
      - lambda: |-
          id(disp1).set_component_text_printf("notification.head", "%s", header.c_str());
          id(disp1).set_component_text_printf("notification.body", "%s", message.c_str());
          id(disp1).send_command_printf("page 9");
      - rtttl.play: "twobits:d=16,o=5,b=180:c,e,g,c6,8p,c,16p,e,c"
      - delay: 20s
      - lambda: id(disp1).send_command_printf("page 0");

ota:
  password: !secret ota_pwd

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
[... 1168 lines cut ...]
            int symbol=56;
            if (id(zigbee_status).state == "on") {
              symbol=56;
            } else {
              symbol=55;
            }
            id(disp1).send_command_printf("security.p_zigbee.pic=%i", symbol);

# Configure the screen itself
display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: http://192.168.1.215:8123/local/hmi.tft
    on_setup:
      then:
        - wait_until:
            api.connected
        - switch.template.publish:
            id: nextion_init
            state: on
        - delay: 5s
        - switch.template.publish:
            id: nextion_init_slow
            state: on

Mat931, 2023-06-25:
Sorry, that is unreadable. Please make sure there is a ` ``` ` at the beginning and end of the code block and nowhere else.
Then click Preview and make sure it looks nice before posting.

Mat931, 2023-06-25:
You are using an external `nextion` component that wasn't updated to work with the latest ESPHome version.

Remove the `external_components` block from your config and add [this](https://github.com/esphome/esphome/pull/2956) instead: 
```
...

esphome:
  name: $name
  comment: touchscreen loft
  on_boot:
    priority: 601
    then:
      - lambda: id(my_nextion).send_command_printf("DRAKJHSUYDGBNCJHGJKSHBDN");

...
```

stahelp, 2023-06-25:
That fixed my problem. Thank you so much.
Issue can be closed

krouton24, 2023-06-25:
Is this a universal fix/workaround? 

```on_boot:
    priority: 601
    then:
      - lambda: id(disp1).send_command_printf("DRAKJHSUYDGBNCJHGJKSHBDN");```

ssieb, 2023-06-25:
> Is this a universal fix/workaround?

That's the equivalent to the PR in the external component.  And you only need to run it once anyway.

Btw, that also means you've been missing out on all the nextion component improvements since then as well.

closed 2023-06-25 by ssieb