esphome/issues #4619: Flashing ESP32 Error: The program size is greater than maximum allowed

guidokieboom, 2023-06-25:
### The problem

When comp

### Which version of ESPHome has the issue?

ESPHome 2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.06

### What platform are you using?

ESP32

### Board

nodemcu ESP32 240MHz, 320KB RAM, 4MB Flash

### Component causing the issue

ota, api, esp32_ble_tracker, bluetooth_proxy, i2s_audio, media_player

### Example YAML snippet

```yaml
i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    dac_type: external
    i2s_dout_pin: GPIO25 # changed from 22 to 25
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
Compiling /data/esp32-03/.pioenvs/esp32-03/src/main.cpp.o
Linking /data/esp32-03/.pioenvs/esp32-03/firmware.elf
RAM:   [==        ]  16.9% (used 55360 bytes from 327680 bytes)
Flash: [==========]  106.2% (used 1947941 bytes from 1835008 bytes)
Error: The program size (1947941 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 299.28 seconds =========================
```

George1422, 2023-06-25:
In my experience ble and media player will not fit into an ESP32 with 2MB of flash. You need a variant that has 4MB (like ESP32-WROOM-32UE) of  flash to fit both ble and media player into the same device.

guidokieboom, 2023-06-25:
Thanks George for sharing. Indeed when I remove the BLE and only use for the media_player, then I can successfully flash (Flash: [=======   ]  66.7% (used 1223973 bytes from 1835008 bytes)). 

I am using a ESP-WROOM-32 with 4MB flash RAM already. But indeed the error message mentions I only have 1835008 bytes. Would you know if I should do anything to activate the use of 4MB?

esphome:
  name: esp32-03
  platform: ESP32
  board: nodemcu-32s

George1422, 2023-06-25:
Well initially I thought you are right why is it limiting your image size to 2MB, your esp32 has 4MB, so why doesn't the compiler use larger than 2MB. Now I understand ,for OTA updates you need to have 2 time your image size. 2MB for your current image and 2 MB for the image you will upload next. So to have an image size of 4MB, your CPU needs to have 8MB of flash. I never took the time to figure this out I just knew that ble and mediaplayer won't fit on the esp32s I've been purchasing.

I have not tested this but if you wanted to give up on OTA you might be able to use more flash by removing the ota section in your config yaml file. But that means you will need to use usb to update your gadget in the future.