esphome/issues #4621: Get volume, player status and URL from the media player

RoelHermus, 2023-06-25:
### The problem

Ik wil het volume, de status en de afspeel-URL teruglezen
In het YAML-bestand heb ik de volgende verklaringen:

          id(volume) = to_string((float)(id(player).volume));
          id(aktie)  = to_string((int)  (id(player).state));
          id(song)   = id(player).media_url;

Works ok for readback the volume and the  status of the player. But the readback of the media_url give a compilation error:

/config/esphome/huis-media.yaml: In lambda function:
/config/esphome/huis-media.yaml:137:33: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'media_url'
           id(song)   = id(player).media_url;
                                 ^~~~~~~~~
*** [/data/huis-media/.pioenvs/huis-media/src/main.cpp.o] Error 1

Is this a bug, or is media_url not the right member name? 

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32

### Board

esp32 board

### Component causing the issue

I2S

### Example YAML snippet

```yaml
- id: song
    type: string
    restore_value: no
    initial_value:  '"leeg"'    
  - id: volume
    type: string
    restore_value: no
    initial_value:  '"0.0"'   
  - id: aktie
    type: string
    restore_value: no
    initial_value:  '"geen"' 
<break>
          id(volume) = to_string((float)(id(player).volume));
          id(aktie)  = to_string((int)  (id(player).state));
          id(song)   = id(player).media_url;
```

ssieb, 2023-06-25:
There's no way to access the url, it's protected.  It's also specific to the i2s media player, it's not an attribute of the base media player class.

RoelHermus, 2023-06-26:
But the i2s media player includes:

           #include "esphome/components/media_player/media_player.h"

and there you can found it.

ssieb, 2023-06-26:
You can find what?

RoelHermus, 2023-06-26:
In the  software found in  i2s_audio_media_player.h , there is an include part that include the media_player.h file.
In the media_player.h file there are reference to get volume, play status and, I hope, the media_url. Volume and play status works fine, but the media_url give an error.

ssieb, 2023-06-26:
The only reference to the media url is in the call for setting it.  There's no option for getting it.

RoelHermus, 2023-06-26:
ok, clear, maybe it's forgotten....I do not understand why volume and status gettable are and the playing media_url not.