esphome/issues #4622: restore_mode: ALWAYS_ON not working in my scenario

bkbartk, 2023-06-26:
### The problem

Hello,

I use `restore_mode: ALWAYS_ON` but whenever the device restarts the switch if always off.
I use this script, and the enable_timer is always turned off after reboot.
The issue exists on both ESP86xx and ESP32-IDF
Haven't checked other devices.

```
switch:
  - platform: template
    id: enable_timer
    name: ${switch_name} Timer
    restore_mode: ALWAYS_ON
    on_turn_on:
      - script.execute: turn_off_timer
    on_turn_off:
      - script.stop: turn_off_timer
    optimistic: true
  - platform: gpio
    pin: GPIO27
    name: ${switch_name}
    id: light_1
    icon: "mdi:lightbulb"
    on_turn_on:
    - light.turn_on: stat_led
    - if:
        condition:
          switch.is_on: enable_timer
        then:
          - script.execute: turn_off_timer
    on_turn_off:
    - light.turn_off: stat_led
    - script.stop: turn_off_timer

light:
  - platform: status_led
    id: stat_led
    pin:
      number: GPIO13
      inverted: yes

script:
  - id: turn_off_timer
    then:
      - delay: 600s
      - switch.turn_off: light_1
```

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32-IDF

### Board

Sonoff DUALR3 and SONOFF TX0EU1C

### Component causing the issue

switch

### Example YAML snippet

```yaml
switch:
  - platform: template
    id: enable_timer
    name: ${switch_name} Timer
    restore_mode: ALWAYS_ON
    on_turn_on:
      - script.execute: turn_off_timer
    on_turn_off:
      - script.stop: turn_off_timer
    optimistic: true
  - platform: gpio
    pin: GPIO27
    name: ${switch_name}
    id: light_1
    icon: "mdi:lightbulb"
    on_turn_on:
    - light.turn_on: stat_led
    - if:
        condition:
          switch.is_on: enable_timer
        then:
          - script.execute: turn_off_timer
    on_turn_off:
    - light.turn_off: stat_led
    - script.stop: turn_off_timer

light:
  - platform: status_led
    id: stat_led
    pin:
      number: GPIO13
      inverted: yes

script:
  - id: turn_off_timer
    then:
      - delay: 600s
      - switch.turn_off: light_1
```

bkaufx, 2023-06-27:
Can you connect this to USB and see what the logs say on boot?  If not, maybe run it on a d1 mini or something temporarily to see what is happening.

bkbartk, 2023-06-27:
thank you,
I prepared a D1 mini 32 bit,
and I hope this is the log you are looking for
```
[D][switch:012]: 'Debug 32 Timer' Turning ON.
[D][switch:055]: 'Debug 32 Timer': Sending state ON
[D][button:010]: 'debug-32 switch Restart' Pressed.
[I][restart.button:012]: Restarting device...
[I][app:134]: Rebooting safely...
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[D][esp-idf:000]: I (108835) wifi:
[D][esp-idf:000]: state: run -> init (0)
[D][esp-idf:000]: 

[D][esp-idf:000]: I (108840) wifi:
[D][esp-idf:000]: pm stop, total sleep time: 93604419 us / 100223115 us

[D][esp-idf:000]: 

[D][esp-idf:000]: I (108844) wifi:
[D][esp-idf:000]: <ba-del>idx
[D][esp-idf:000]: 

[D][esp-idf:000]: I (108857) wifi:
[D][esp-idf:000]: new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
[D][esp-idf:000]: 

[D][esp-idf:000]: W (108872) httpd_txrx: httpd_sock_err: error in recv : 113

[D][esp-idf:000]: W (108874) httpd_txrx: httpd_sock_err: error in recv : 113

[D][esp-idf:000]: I (108899) wifi:
[D][esp-idf:000]: flush txq
[D][esp-idf:000]: 

[D][esp-idf:000]: I (108900) wifi:
[D][esp-idf:000]: stop sw txq
[D][esp-idf:000]: 

ts Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14944
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
I (29) boot: ESP-IDF 4.4.4 2nd stage bootloader
I (29) boot: compile time 21:31:43
I (29) boot: chip revision: v1.0
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (76) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (84) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (91) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (99) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (106) boot: End of partition table
I (111) boot_comm: chip revision: 1, min. application chip revision: 0
I (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=238a0h (145568) map
I (179) esp_image: segment 1: paddr=000338c8 vaddr=3ffb0000 size=036fch ( 14076) load
I (185) esp_image: segment 2: paddr=00036fcc vaddr=40080000 size=0904ch ( 36940) load
I (200) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=95b5ch (613212) map
I (422) esp_image: segment 4: paddr=000d5b84 vaddr=4008904c size=0b7c8h ( 47048) load
I (452) boot: Loaded app from partition at offset 0x10000
I (452) boot: Disabling RNG early entropy source...
I (464) cpu_start: Pro cpu up.
I (464) cpu_start: Starting app cpu, entry point is 0x40081c44
I (450) cpu_start: App cpu up.
I (480) cpu_start: Pro cpu start user code
I (480) cpu_start: cpu freq: 160000000
I (480) cpu_start: Application information:
I (485) cpu_start: Project name:     debug-32
I (490) cpu_start: App version:      2023.6.2
I (495) cpu_start: Compile time:     Jun 27 2023 21:28:17
I (501) cpu_start: ELF file SHA256:  ca387399431d570c...
I (507) cpu_start: ESP-IDF:          4.4.4
I (512) heap_init: Initializing. RAM available for dynamic allocation:
I (519) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (525) heap_init: At 3FFB7468 len 00028B98 (162 KiB): DRAM
I (531) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (538) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (544) heap_init: At 40094814 len 0000B7EC (45 KiB): IRAM
I (552) spi_flash: detected chip: generic
I (555) spi_flash: flash io: dio
I (560) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```
I don't see any usefull information

bkaufx, 2023-06-27:
There's never anything about the switch being set up?  Here's what I get after your logs end:

```
[13:15:10]I (421) cpu_start: Starting scheduler on PRO CPU.
[13:15:10]I (0) cpu_start: Starting scheduler on APP CPU.
[13:15:10][I][logger:262]: Log initialized
[13:15:10][C][ota:469]: There have been 7 suspected unsuccessful boot attempts.
[13:15:10][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:15:10][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[13:15:10][I][app:029]: Running through setup()...
[13:15:10][C][status_led:055]: Setting up Status LED...
[13:15:10][D][esp-idf:000]: I (41) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[13:15:10]
[13:15:10][C][switch.gpio:011]: Setting up GPIO Switch 'test_switch'...
[13:15:10][D][switch:016]: 'test_switch' Turning OFF.
[13:15:10][D][switch:055]: 'test_switch': Sending state OFF
[13:15:10][D][light:036]: 'stat_led' Setting:
[13:15:10][D][light:041]:   Color mode: 
[13:15:10][D][esp-idf:000]: I (74) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[13:15:10]
[13:15:10][D][switch:016]: 'test_switch' Turning OFF.
[13:15:10][C][light:035]: Setting up light 'stat_led'...
```

Then after wifi sets up I get:

```
[13:15:15][C][logger:301]: Logger:
[13:15:15][C][logger:302]:   Level: DEBUG
[13:15:15][C][logger:303]:   Log Baud Rate: 115200
[13:15:15][C][logger:305]:   Hardware UART: UART0
[13:15:15][C][template.switch:068]: Template Switch 'test_switch Timer'
[13:15:15][C][template.switch:091]:   Restore Mode: always ON
[13:15:15][C][template.switch:060]:   Restore State: NO
[13:15:15][C][template.switch:061]:   Optimistic: YES
[13:15:15][C][switch.gpio:068]: GPIO Switch 'test_switch'
[13:15:15][C][switch.gpio:070]:   Icon: 'mdi:lightbulb'
[13:15:15][C][switch.gpio:091]:   Restore Mode: always OFF
[13:15:15][C][switch.gpio:031]:   Pin: GPIO27
[13:15:15][C][light:103]: Light 'stat_led'
```

It appears that always_on isn't working for me either.  It's kind of crazy because the template switch setup seems to just exit and not do anything if the switch is not set to restore, so I don't see how `always on` is even supposed to work.  
https://github.com/esphome/esphome/blob/dev/esphome/components/template/switch/template_switch.cpp#L43

What version were you on before?  This was changed in January, so probably the February release, it looks like.  Is this your first update since then?
https://github.com/esphome/esphome/pull/4280/

Honestly it looks like that check for `this->restore_state_` at that first link shouldn't be there.  It looks like the get_initial_state_with_restore_mode() call in the next line is going to return on if the mode is always_on.

Just made that changed and it fixed it.  Maybe I'll do a pull request later if someone else doesn't get to it first.

bkaufx, 2023-06-27:
Actually that PR I linked to doesn't even change the line I'm talking about, so who knows why the behavior changed.

bkaufx, 2023-06-27:
Now that I'm looking at this a little more, reading documentation, etc., I think this is operating as intended but just documented poorly.  I think you need to set `restore_state: true` to get the restore logic to run.  Then it will restore to `ALWAYS_ON`.  Even though nothing is being "restored", the template switch still considers restoring to always on to be part of the "restore" functionality.

I think documentation needs updated to clarify this or the restore_state option should be removed.  Just making the default restore mode "disabled" would accomplish the same thing.

ZonkSec, 2023-06-28:
Where does `restore_state: true` need to be set? It fails to validate for me when I set just above `restore_mode`.

I'm having a potentially related issue with a switch and restore_mode. I'm seeing inconsistent behavior of `restore_mode` between a power failure restart (unplug, replug power) and a software restart (after flashing or using platform: restart). 

If I set `restore_mode: ALWAYS_OFF`:
* power failure restart results in the switch being `on` by default
* software restart results in the switch being `off` on by default

If I set `restore_mode: ALWAYS_ON`:
* power failure restart results in the switch being `off` on by default
* software restart results in the switch being `on` by default

In my scenario, it seems like the software restart respects the `restore_mode` config, but a power failure restart does the opposite.

bkaufx, 2023-06-28:
This validates fine for me.  I haven't tested it functionally to see if it fixes your issue.  What message are you getting with the validation error?

```
switch:
  - platform: template
    id: enable_timer
    name: ${switch_name} Timer
    restore_state: true
    restore_mode: ALWAYS_ON
    on_turn_on:
      - script.execute: turn_off_timer
    on_turn_off:
      - script.stop: turn_off_timer
    optimistic: true
  - platform: gpio
    pin: GPIO27
    name: ${switch_name}
    id: light_1
    icon: "mdi:lightbulb"
    on_turn_on:
    - light.turn_on: stat_led
    - if:
        condition:
          switch.is_on: enable_timer
        then:
          - script.execute: turn_off_timer
    on_turn_off:
    - light.turn_off: stat_led
    - script.stop: turn_off_timer

```

ZonkSec, 2023-06-28:
Here is my validation error:

```
Failed config

switch.gpio: [source /share/wyzeoutdoor.yaml:38]
  platform: gpio
  name: WyzeOutdoor Relay1
  pin:
    number: GPIO15
    inverted: False
  id: relay1

  [restore_state] is an invalid option for [switch.gpio]. Did you mean [restore_mode]?
  restore_state: True
  restore_mode: ALWAYS_OFF
  on_turn_on:
    - light.turn_on: relay1_led
  on_turn_off:
    - light.turn_off: relay1_led
```

Main difference I see is mine is on `platform: gpio` and yours is `platform: template`. 

Apologizes if my issue is obvious, I'm new to esphome! Appreciate your help so far!

ZonkSec, 2023-06-28:
Ahh, I see. Sorry for throwing the discussion off topic with my misunderstanding.

bkbartk, 2023-06-28:
`I think you need to set restore_state: true`
that indeed seems to work,
I don't know if it ever has worked before, Or I just assumed it did.
still strange that `restore_mode` is totally ignored if you don't set restore_state, Why else would one place it,

bkaufx, 2023-07-17:
This should be fixed in the next update: https://github.com/esphome/esphome/pull/5106