esphome/issues #4628: Waveshare 7.5inV2 does not seem to be supported

TobnacTobi, 2023-06-27:
### The problem

# Problem
The Waveshare E-Ink Display flashes, but does not show/draw anything.  
Seems to be related to https://github.com/esphome/issues/issues/2880

I want to build a E-Ink Dashboard with Home-Assistant, ESPHome and these components:  
- AZ-Delivery ESP32 Dev KitC **V2** (ESP32-WROMM-32) ([Amazon](https://www.amazon.de/gp/product/B074RGW2VQ))
- Waveshare 7.5 Inch E-Paper Display Hat Module V2 ([Amazon](https://www.amazon.de/gp/product/B075R4QY3L))  
- HASS, ESPHome, etc  

Now, I tested multiple GPIO combinations, different board-ids (nodemcu-32s, az-delivery-devkit-v4, esp32dev) and other display model configurations (`7.50in-bV2`, `7.50inV2`, `7.50inV2alt`) and most of them either don't work or show this behavior:  
When updating, the display flickers (as for an full update) but then stays black or shows some "random" pattern that looks like stars in the night sky.

The board is working, I have tested others of the same version. Maybe the board is not supported (since it's V2 instead of the suggested V4).  
I flash it via [ESPHome Web](https://web.esphome.io) and it works without any problems. I can read the logs as well.  
What seems **suspicious**: There is a ~3 pixel border around the black rectangle that appears ( i know it's not the physical border since it flashes black when updating).
This might be some code which tries to draw as well and hides my actual print message behind it.

I am very thankful for any help, this has taken me way too long already.

### Which version of ESPHome has the issue?

2023.6.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.6.2

### What platform are you using?

ESP32

### Board

AZ-Delivery ESP32 Dev KitC V2

### Component causing the issue

Waveshare 7.5 Inch E-Paper Display Hat Module V2

### Example YAML snippet

```yaml
esphome:
  name: helloworld
  friendly_name: HelloWorld

esp32:
  board: nodemcu-32s # also tried az-delivery-devkit-v4, esp32dev
  framework:
    type: arduino

logger: # Enable logging
api: # Enable Home Assistant API

ota:
  password: "***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap: # Enable fallback hotspot (captive portal) in case wifi connection fails
    ssid: "Helloworld Fallback Hotspot"
    password: "WItLm8NwlS07"

captive_portal:

font:
  - file: 'fonts/OpenSans-Regular.ttf' # located in ~/home-assistant-config/esphome/fonts/OpenSans-Regular.ttf
    id: font1
    size: 8

# Example configuration entry
spi:
  clk_pin: 13
  mosi_pin: 14
 
 
# --- Display Layout -----------------------------------------------------------
display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    reset_duration: 2ms
    model: 7.50inV2
    update_interval: 20s
    lambda: |-
      ESP_LOGD("main", "logging01");
      it.print(0, 0, id(font1), "Hello World!"); # here i already tested rectangle, which also does not work
      ESP_LOGD("main", "logging02");
```
```

alexd321, 2023-06-28:
similar here, have a very dark screen, like the contrast is wrong.

rolled back to 2023.6.3 and still the same issue. i dont remember which version i had installed when it was working

TobnacTobi, 2023-06-29:
> rolled back to 2023.5.4 and still the same issue. i dont remember which version i had installed when it was working

Thanks for this hint!  
Desparate for any solution I installed the version `2022.12.2` of ESPHome and it immediately worked!  
It probably broke inbetween. I had a look at the Change-Logs but there's nothing specific for the 7.5inV2 that might break it.

Maybe this will work for you as well.  
I'm glad that I can continue my project now, thanks!

alexd321, 2023-06-29:
Reloading with v2022.12.1 now. I would guess one of these commits broke it. I am not familiar enough with github to look at file history across multiple versions. https://github.com/esphome/esphome/commits/dev/esphome/components/waveshare_epaper

[image]

alexd321, 2023-06-29:
I suspect one of these waveshare files: 

https://github.com/esphome/esphome/compare/f83f1bf..bd6d6ca#diff-062684c9f1a454d53ffc74c0ef373f0f1bef8e954051d4b97536365b28bfdb37

okueng, 2023-10-12:
Works well on my setup (firebeetle, waveshare hat, 7.5inV2) on 2023.9.3
Did you check the inverted: true in the doc? The switches on the hat are correctly set (B,0)?

Here is my yaml setup:
```
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: GPIO14
    dc_pin: GPIO13
    busy_pin: 
      number: GPIO25
      inverted: true
    reset_pin: GPIO26
    model: 7.50inV2
    reset_duration: 2ms
```

fukajan, 2023-11-27:
I am facing the same problem. I tried to revert to as far as 2022.12.1, unfortunately without success. I just realized that I am using a Waveshare e-Papaer ESP32 Driver Board insted of the Driver Hat. No sure what this means and if it is still related or not.

daelsc, 2024-04-29:
Just an update....the yaml posted by OP works fine for me on 2024.4.1.  I am using the 7.5" display with V2 sticker on the back with the Waveshare ESP32 Driver Board (no hat).

briodan, 2024-06-12:
anyone able to solve this getting contrast issues on 2024.4.2 first time i got an epaper display.
working to roll back to and older version but that's going to take a bit.

groemi, 2024-07-03:
I also had the problem that the display was very pale. I had an FPC-8612 v2 panel that worked perfectly with the ESP-32 switches on ON and B. With the FPC-C001 20.08.20, however, the one switch had to be set to A for it to work. In my case, the problem solution was therefore independent of the software.