esphome/issues #4630: esphome 2023.6.2 does not find dallas temperature sensors

herzrasen-cpu, 2023-06-27:
### The problem

After updating from 2023.5.5 to 2023.6.2, no Dallas temperature sensors DS18B20 are found. (Found no sensors!)
After the downgrade to 2023.5.5, the sensors are found again and the temperature values can be determined again.
everything works Fine.

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

dallas

### Example YAML snippet

```yaml
esphome:
  name: temperatur

esp32:
  board: nodemcu-32s

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "fc21e67a0eee06bfd00d47cc9dc64c67"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.0.198

  manual_ip:
    static_ip: 192.168.0.198
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temperatur Fallback Hotspot"
    password: "ffAbqBl7ayfj"

captive_portal:

dallas:
  - pin: 23
    id: keller

sensor:
  - platform: dallas
    address: 0x58030397942bd028
    name: "Heizung Vorlauf Temperatur"
    filters:
      - filter_out: nan
  - platform: dallas
    address: 0x0e0303979424a828
    name: "Heizung Rücklauf Temperatur"
    filters:
      - offset: -0.5
      - filter_out: nan
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/temperatur.yaml...
INFO Starting log output from 192.168.0.198 using esphome API
INFO Successfully connected to 192.168.0.198
[22:31:36][I][app:102]: ESPHome version 2023.5.5 compiled on Jun 27 2023, 08:08:59
[22:31:36][C][wifi:505]: WiFi:
[22:31:36][C][wifi:363]:   Local MAC: C8:C9:A3:C6:FF:58
[22:31:36][C][wifi:364]:   SSID: [redacted]
[22:31:36][C][wifi:365]:   IP Address: 192.168.0.198
[22:31:36][C][wifi:367]:   BSSID: [redacted]
[22:31:36][C][wifi:368]:   Hostname: 'temperatur'
[22:31:36][C][wifi:370]:   Signal strength: -68 dB ▂▄▆█
[22:31:36][C][wifi:374]:   Channel: 2
[22:31:36][C][wifi:375]:   Subnet: 255.255.255.0
[22:31:36][C][wifi:376]:   Gateway: 192.168.0.1
[22:31:36][C][wifi:377]:   DNS1: 192.168.0.1
[22:31:36][C][wifi:378]:   DNS2: 0.0.0.0
[22:31:36][C][logger:301]: Logger:
[22:31:36][C][logger:302]:   Level: DEBUG
[22:31:36][C][logger:303]:   Log Baud Rate: 115200
[22:31:36][C][logger:305]:   Hardware UART: UART0
[22:31:36][C][dallas.sensor:075]: DallasComponent:
[22:31:36][C][dallas.sensor:076]:   Pin: GPIO23
[22:31:36][C][dallas.sensor:077]:   Update Interval: 60.0s
[22:31:36][D][dallas.sensor:082]:   Found sensors:
[22:31:36][D][dallas.sensor:084]:     0x58030397942bd028
[22:31:36][D][dallas.sensor:084]:     0x0e0303979424a828
[22:31:36][C][dallas.sensor:089]:   Device 'Heizung Vorlauf Temperatur'
[22:31:36][C][dallas.sensor:089]:     Device Class: 'temperature'
[22:31:36][C][dallas.sensor:089]:     State Class: 'measurement'
[22:31:36][C][dallas.sensor:089]:     Unit of Measurement: '°C'

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.198 using esphome API
INFO Successfully connected to 192.168.0.198
[22:35:38][I][app:102]: ESPHome version 2023.6.2 compiled on Jun 27 2023, 22:33:54
[22:35:38][C][wifi:543]: WiFi:
[22:35:38][C][wifi:379]:   Local MAC: C8:C9:A3:C6:FF:58
[22:35:38][C][wifi:380]:   SSID: [redacted]
[22:35:38][C][wifi:381]:   IP Address: 192.168.0.198
[22:35:38][C][wifi:383]:   BSSID: [redacted]
[22:35:38][C][wifi:384]:   Hostname: 'temperatur'
[22:35:38][C][wifi:386]:   Signal strength: -69 dB ▂▄▆█
[22:35:38][C][wifi:390]:   Channel: 2
[22:35:38][C][wifi:391]:   Subnet: 255.255.255.0
[22:35:38][C][wifi:392]:   Gateway: 192.168.0.1
[22:35:38][C][wifi:393]:   DNS1: 192.168.0.1
[22:35:38][C][wifi:394]:   DNS2: 0.0.0.0
[22:35:38][C][logger:301]: Logger:
[22:35:38][C][logger:302]:   Level: DEBUG
[22:35:38][C][logger:303]:   Log Baud Rate: 115200
[22:35:38][C][logger:305]:   Hardware UART: UART0
[22:35:38][C][dallas.sensor:075]: DallasComponent:
[22:35:38][C][dallas.sensor:076]:   Pin: GPIO23
[22:35:38][C][dallas.sensor:077]:   Update Interval: 60.0s
[22:35:38][W][dallas.sensor:080]:   Found no sensors!
[22:35:38][C][dallas.sensor:089]:   Device 'Heizung Vorlauf Temperatur'
[22:35:38][C][dallas.sensor:089]:     Device Class: 'temperature'
[22:35:38][C][dallas.sensor:089]:     State Class: 'measurement'

INFO ESPHome 2023.5.5
INFO Reading configuration /config/esphome/temperatur.yaml...
INFO Starting log output from 192.168.0.198 using esphome API
INFO Successfully connected to 192.168.0.198
[22:11:59][I][app:102]: ESPHome version 2023.5.5 compiled on Jun 27 2023, 08:08:59
[22:11:59][C][wifi:505]: WiFi:
[22:11:59][C][wifi:363]:   Local MAC: C8:C9:A3:C6:FF:58
[22:11:59][C][wifi:364]:   SSID: [redacted]
[22:11:59][C][wifi:365]:   IP Address: 192.168.0.198
[22:11:59][C][wifi:367]:   BSSID: [redacted]
[22:11:59][C][wifi:368]:   Hostname: 'temperatur'
[22:11:59][C][wifi:370]:   Signal strength: -69 dB ▂▄▆█
[22:11:59][C][wifi:374]:   Channel: 2
[22:11:59][C][wifi:375]:   Subnet: 255.255.255.0
[22:11:59][C][wifi:376]:   Gateway: 192.168.0.1
[22:11:59][C][wifi:377]:   DNS1: 192.168.0.1
[22:11:59][C][wifi:378]:   DNS2: 0.0.0.0
[22:11:59][C][logger:301]: Logger:
[22:11:59][C][logger:302]:   Level: DEBUG
[22:11:59][C][logger:303]:   Log Baud Rate: 115200
[22:11:59][C][logger:305]:   Hardware UART: UART0
[22:11:59][C][dallas.sensor:075]: DallasComponent:
[22:11:59][C][dallas.sensor:076]:   Pin: GPIO23
[22:11:59][C][dallas.sensor:077]:   Update Interval: 60.0s
[22:11:59][D][dallas.sensor:082]:   Found sensors:
[22:11:59][D][dallas.sensor:084]:     0x58030397942bd028
[22:11:59][D][dallas.sensor:084]:     0x0e0303979424a828
[22:11:59][C][dallas.sensor:089]:   Device 'Heizung Vorlauf Temperatur'
```

RoelHermus, 2023-06-30:
You're not the only one, see close issue:

[Why does no one care that Dallas DS18B20 sensors have not been working with the ESP32 since 6 months?](https://github.com/esphome/issues/issues/4543#top)
#4543

It's a problem with esp32s

RedFlash49, 2023-07-03:
Same problem here. 
I found a hint in the HA forum. Set the logger level to VERBOSE. 
I did that and my two DS18B20 are working again. Don‘t ask me why..

RedFlash49, 2023-07-05:
I updated to 2023.6.4 and its working now again with normal logger level.

ssieb, 2024-06-25:
This should be fixed now.

closed 2024-06-25 by ssieb

tuct, 2024-06-27:
What version fixes this?

ssieb, 2024-06-28:
2024.6.3