esphome/issues #4631: Media_player component not playing longer paths

iondulgheru, 2023-06-27:
### The problem

I am trying to use the media_player component to play music directly from Home Assistant.
Some mp3 files were working, some were not.
I made some debugging, and the conclusion was that the problem is the file path length, including the filename.
I could reproduce this multiple times by renaming mp3 files that didn't work initially so they have a shorter filename, and after that they worked.
An example of path that is not playing is "http://xxxx:8123/media/media/music_free_and_purchased/singles/Van%20Morrison%20-%20%5B1967%5D%20-%20%5BBlowin%20Your%20Mind%5D%20-%2001%20-%20Brown%20Eyed%20Girl.mp3?authSig=xxxx"
After I shorten the filename I can play it, the path being:
"http://xxxx:8123/media/media/music_free_and_purchased/singles/Van%20Morrison%20-%20%5B1967%5D%20-%20%5BBlowin%20Your%20Mind%5D%20-%2001%20-%20Bro.mp3?authSig=xxxx"

The board I am using is a esp32doit-devkit-v1 with a PCM5102 DAC.

There are no errors in logs. In Home Assistant the loading animation is shown instead of player controls when this problem occurs.

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

media_player

### Example YAML snippet

```yaml
i2s_audio:
  - i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO5

media_player:
  - platform: i2s_audio
    id: media_out
    name: ${friendly_name}
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: mono
    mute_pin:
      number: GPIO21
      inverted: true

microphone:
  - platform: i2s_audio
    id: board_microphone
    channel: left
    i2s_din_pin: GPIO35
    adc_type: external
    pdm: false

speaker:
  - platform: i2s_audio
    id: board_external_speakers
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: mono
```

George1422, 2023-06-28:
While I don't have an answer for you, it might be helpful to know the actual string length of the path that works vs one that doesn't work. It may be something the developers might need to know.

iondulgheru, 2023-06-28:
From what I see the url that works, including the auth string, has 471 characters. Without the http parameters it has 157 characters. This is for the url-encoded url.
After url-decode, it has 441 characters including authSig, and 127 without http parameters.

iondulgheru, 2023-06-28:
Maybe this is the problem?
https://github.com/esphome/esphome/blob/dev/esphome/components/media_player/media_player.h#L58
`const optional<std::string> &get_media_url() const { return media_url_; }`
And maybe is related to this issue?
https://stackoverflow.com/questions/34560502/why-is-sizeofstdstring-only-eight-bytes

This is all I could think of, I don't know C++ .

iondulgheru, 2023-09-02:
Also I discovered that it cannot also play media that contains parantheses in the name.

koriwi, 2023-12-24:
the limit is 512. also it cannot play from IPs because it tries to do a DNS resolve for everything which wont work for IPs. Then it throws error 0 which is actually not in the list of known errors for arduino so it silently fails

closed 2025-05-27 by jesserockz