esphome/issues #4633: Custom BinarySensor not including required files in build tree

UniKyrn, 2023-06-28:
### The problem

Using the example code for BinarySensor and TextSensor:

https://www.esphome.io/components/binary_sensor/custom.html
https://www.esphome.io/components/text_sensor/custom.html

Only the TextSensor compiles.  Looking at the build tree for the project, it appears the support code for BinarySensor did not get included, so the compile fails.

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP8266

### Board

D1-mini

### Component causing the issue

Custom Binary Sensor

### Example YAML snippet

```yaml
substitutions:
  name: esp8266-thermal-camera
  text_name: "ESP8266 Thermal Camera"
  
esphome:
  name: ${name}
  comment: WeMos D1 Mini
  name_add_mac_suffix: false
  includes:
    - MyTextSensor.h
    - MyBinarySensor.h
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/esp8266-thermal-camera.yaml...
INFO Detected timezone 'America/Los_Angeles'
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-thermal-camera (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- Wire @ 1.0
|-- SparkFun GridEYE AMG88 Library @ 1.0.2
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 0.8.6
|-- ArduinoJson @ 6.18.5
Compiling /data/esp8266-thermal-camera/.pioenvs/esp8266-thermal-camera/src/main.cpp.o
In file included from src/main.cpp:67:
src/MyBinarySensor.h:3:75: error: expected class-name before '{' token
    3 | class MyCustomBinarySensor : public PollingComponent, public BinarySensor {
      |                                                                           ^
src/MyBinarySensor.h: In member function 'virtual void MyCustomBinarySensor::update()':
src/MyBinarySensor.h:17:5: error: 'publish_state' was not declared in this scope
   17 |     publish_state(state);
      |     ^~~~~~~~~~~~~
*** [/data/esp8266-thermal-camera/.pioenvs/esp8266-thermal-camera/src/main.cpp.o] Error 1
========================= [FAILED] Took 12.74 seconds =========================
```

### Additional information

[image]

ssieb, 2023-06-28:
If that is your entire yaml, then you're missing a lot.  I'm surprised that the text sensor would compile.  Also, "custom" components are deprecated.  You should make an external component instead.

closed 2023-06-28 by ssieb

UniKyrn, 2023-06-28:
No, it's just the first part of the YAML that shows the includes.  If custom components are being deprecated, then why does the documentation still include them, without a warning, or a link to how it should be done going forward?

I didn't write the AMG8833 Thermal Library component, I just ended up resorting to ESPHome docs while trying to figure out why it wouldn't compile.

ssieb, 2023-06-28:
If you want help, then you need to include all of the relevant yaml.