esphome/issues #4637: ESPHome fails to build with esp-idf framework if wire ethernet is used

utemkin, 2023-06-29:
### The problem

Build fails

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32-IDF

### Board

wt32-eth01

### Component causing the issue

components/ethernet

### Example YAML snippet

```yaml
`
esp32:
  board: wt32-eth01
  framework:
    type: esp-idf
    version: 5.0.2
    platform_version: 6.3.2
    # Custom sdkconfig options
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SPEED: y
      CONFIG_GPIO_CTRL_FUNC_IN_IRAM: y
`
`
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
`
```

### Anything in the logs that might be useful for us?

```txt
`
INFO ESPHome 2023.6.2
INFO Reading configuration /config/esphome/climate.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.0.9
INFO Compiling app...
Processing climate (board: wt32-eth01; framework: espidf; platform: platformio/espressif32@6.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50002.230601 (5.0.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling /data/climate/.pioenvs/climate/src/esphome/components/ethernet/esp_eth_phy_jl1101.o
src/esphome/components/ethernet/esp_eth_phy_jl1101.c:22:10: fatal error: eth_phy_regs_struct.h: No such file or directory

*****************************************************************************
* Looking for eth_phy_regs_struct.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:eth_phy_regs_struct.h"
* Web  > https://registry.platformio.org/search?q=header:eth_phy_regs_struct.h
*
*****************************************************************************

   22 | #include "eth_phy_regs_struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/climate/.pioenvs/climate/src/esphome/components/ethernet/esp_eth_phy_jl1101.o] Error 1
========================== [FAILED] Took 4.64 seconds ==========================
`
```

### Additional information

This happens because components\ethernet/esp_eth_phy_jl1101.c unconditionally includes eth_phy_regs_struct.h, which is part of Arduino framework (which is not available when building with esp-idf framework)

HeMan, 2023-06-29:
Is there any specific reason for you to specify `version` and `platform_version`? When I removed them I got it to compile.

utemkin, 2023-06-30:
> Is there any specific reason for you to specify `version` and `platform_version`? When I removed them I got it to compile.

Yes. It uses ESP_IDF v4 by default. I would like to use new API like gptimer and so on. There's lots of improvements in v5 actually.

randybb, 2023-06-30:
There is a loot of breaking changes in IDF v5, so it will not be supported in a near future. Jesse mentioned that probably next year...

closed 2023-07-04 by jesserockz