esphome/issues #4638: The pulse_width sensor is not properly documented

konikvranik, 2023-06-29:
### The problem

I'd like to measure how long is my pump enabled. So I setup pulse width sensor to measure it.

My expectation is that I'll get time of running of the pump, but I got some weird values of unknown time unit.

So is it measuring time in seconds? If true, then it doesn't work properly. After more than minute of run I got value of 0.013.

### Which version of ESPHome has the issue?

2023.6.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

not related

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

pulse_width

### Example YAML snippet

```yaml
- platform: pulse_width
    pin:
      number: GPIO1
      inverted: True
    name: Doba běhu čerpadla
    id: cerpadlo_pulse_width
    on_value:
      then:
        - if:
            condition:
              lambda: 'return x > 20.0;'
            then:
              - globals.set:
                  id: global_pulsewidth_watertotal
                  value: !lambda 'return id(global_pulsewidth_watertotal) + (x * .4);'
```

### Anything in the logs that might be useful for us?

```txt
nope
```

RoboMagus, 2023-10-09:
The pulse width sensor is usefull for PWM applications wich operate in the order of  mili / micro-seconds. Looking at your usecase this may not be the right method for you.

Though not documented, If you take a look at the sensor output in either HomeAssistant, or the ESPHome logs you'll see that the unit of measurement for this sensor is in seconds.

RoboMagus, 2023-10-09:
> So is there any limitation in the pulse length?

Looking at the internals, just over 70 minutes is the largest pulse length this component can measure. To add some clarifications to the documentation I've opened a PR to ensure this information is made available more up-front.
120s should be well within that time frame though.

My personal solution for time tracking is similar to what you mention with manual time delta computation based on On/Off triggers, which is more suitable for this usecase I.M.O.

konikvranik, 2023-10-09:
So what should be the `pulse_width` sensor for? My understanding is that if there are -> ON and -> OFF transitions, the sensor should return the length of this pulse.

RoboMagus, 2023-10-09:
Not quite. As previously mentioned the usecase for pulse_width sensor is for `PWM`. Generally a continuous signal in which relevant data is encoded. And as indicated this is generally using sub-second periods.

Based on my interpretation of the information you provided, one-shot measurements are of interest to you. Which does not match the use-case for the pulse_width sensor. Also, in PWM signals one usually does not have to deal with debounce. I think when turning on / off a pump debounce could cause problems when using a component like this.

konikvranik, 2023-10-09:
Sure, but I'm missing this information on the documentation which leads to this missuse. I still don't understand the use case for this sensor. Oh this should measure HIGH length, then it should apply to my described use case. But it's not true. The documentation is still not complete IMHO.

RoboMagus, 2023-10-09:
See of the PWM sensor as an ADC, their signals are fairly similar in how relevant information is contained. Both are meant for continuous signals (in general). Both are are also intended for _clean_ signals, i.e. no debounce on the line. 
For a switching signals, debounce needs to be taken into account, for which there are no ways to do so in the pulse_width component.

For something long running (larger than sub-second periods), the method you mentioned previously by manually setting values based on On / Off triggers is a valid solution. Looking at the HA forums, others seem to be doing the same.