esphome/issues #4639: doesn't work when wifi enable_on_boot is false and native api enabled

dima-fedotovs, 2023-06-29:
### The problem

I want to create a battery-powered temperature sensor. I'm trying to save as much energy as possible. The idea was to utilize the wifi.enable_on_boot, setting it to false. Then on boot (or after deep sleep), compare the new value from the temperature sensor with the previous value. If the difference is big enough, then enable the WiFi and send a new temperature to HA. If the temperature is changed just a little - do nothing - just enable the deep_sleep. 
I thought it was a trivial task. The main logic is done and tested easily. 
After I tested the logic of manipulating with sensors, I decided to turn enable_on_boot of wifi to false. And there is a problem: after I uploaded the configuration, my device stopped working. I can see non-stopping messages in logs. Logs differ on T7-S3 and T-Display-S3. But both devices don't work.
Seems the reason is that I've enabled native API for communication with HA, and it crashes the system if there is no WiFi connection. Removing native API - everything works except sending sensor's values to HA.

It would be nice to start initializing of native API subsystem only after WiFI is connected.

### Which version of ESPHome has the issue?

2023.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP32

### Board

LilyGo T7 S3, LilyGo T-Display-S3

### Component causing the issue

native api, wifi

### Example YAML snippet

```yaml
esp32:
    board: esp32-s3-devkitc-1
    variant: esp32s3
    framework:
        type: arduino
        
esphome:
    name: temperature-sensor-2
    comment: Temperature Sensor 2
    on_boot:
        priority: -100
        then:
            - delay: 5s
            - wifi.enable:
            - wait_until:
                  condition:
                      wifi.connected:
            # sensor.update: xyz # update the sensor to send data
            - delay: 5s
            - wifi.disable:
            - delay: 5s
            - deep_sleep.enter:
                  id: default_deep_sleep
                  
api:
    encryption:
        key: !secret api_encryption_key

wifi:
    power_save_mode: HIGH
    fast_connect: true
    enable_on_boot: false
    ssid: !secret wifi_ssid
    password: !secret wifi_password

deep_sleep:
    id: default_deep_sleep
    sleep_duration: 15s
```

### Anything in the logs that might be useful for us?

```txt
Logs from T-Display:

[20:23:25]load:0x3fce3808,len:0x43c
[20:23:25]load:0x403c9700,len:0xbec
[20:23:25]load:0x403cc700,len:0x2a3c
[20:23:25]entry 0x403c98d8
[20:23:25]ESP-ROM:esp32s3-20210327
[20:23:25]Build:Mar 27 2021
[20:23:25]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[20:23:25]Saved PC:0x42089186
WARNING Decoded 0x42089186: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[20:23:25]SPIWP:0xee
[20:23:25]mode:DIO, clock div:1
[20:23:25]load:0x3fce3808,len:0x43c
[20:23:25]load:0x403c9700,len:0xbec
[20:23:25]load:0x403cc700,len:0x2a3c
[20:23:25]entry 0x403c98d8
[20:23:25]ESP-ROM:esp32s3-20210327
[20:23:25]Build:Mar 27 2021
[20:23:25]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[20:23:25]Saved PC:0x42089186
WARNING Decoded 0x42089186: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[20:23:25]SPIWP:0xee
[20:23:25]mode:DIO, clock div:1
[20:23:25]load:0x3fce3808,len:0x43c
```

dima-fedotovs, 2023-06-30:
Similar situation with MQTT. The device cannot normally start up if MQTT is enabled, but wifi.enable_on_boot is false.

dima-fedotovs, 2023-07-27:
I've conducted a deeper investigation. I gathered logs using UART and found the following:

```
[E][component:113]: Component mdns was marked as failed.
```

The device is continuously rebooting and producing the same output. Following the logs, I disabled mdns. However, this didn't lead to any success. The device continues to crash, but now with a different message:

```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x42021011
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
[I][logger:262]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][main:262]: Updating sensors
[C][dallas.sensor:032]: Setting up DallasComponent...
[C][adc:044]: Setting up ADC 'internal_battery_level'...
[C][adc:090]: ADC 'internal_battery_level' setup finished!
[C][wifi:038]: Setting up WiFi...

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x40377dce:0x3fce2b600x4037dfb9:0x3fce2b80 0x40383b01:0x3fce2ba0 0x4202d871:0x3fce2cd0 0x4203bafd:0x3fce2d00 0x4203bb4e:0x3fce2d20 0x4202cfc5:0x3fce2d 
```

mkaiser, 2023-08-28:
Similar issue here with following code. As asoon as enable_on_boot is disabled, it crashes

```
esphome:
  name: sleep-tester
  friendly_name: Sleep_tester
 
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Jtjhw7xVaD8weF+y5t2//VViE6Ybf22qKXKiQMB67M4="

ota:
  password: "6faed3b075bf9fa544622e7e8a873f44"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  enable_on_boot: false
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sleep-Tester Fallback Hotspot"
    password: "oRrVfXDjEdek"

captive_portal:
```

log

```
ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382160
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382160
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382160
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382160
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382160
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382160
```

closed 2023-09-11 by jesserockz