esphome/issues #4641: Xiaomi LYWSDCGQ and CGDK2 no longer report Temerature and Humidty.

Petro31, 2023-06-30:
### The problem

I have 5 ESP32's and about 8 to 10 temperature sensors.  All of them being either LYWSDCGQ or CGDK2.

The last update broke them all.  They will only (might) report once after the ESP32 is restarted.  They never report a second time.

### Which version of ESPHome has the issue?

2023.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.0b2

### What platform are you using?

ESP32

### Board

mhetesp32devkit

### Component causing the issue

xiaomi_lywsdcgq, xiaomi_cgdk2

### Example YAML snippet

```yaml
e.g. CGDK2

sensor:
  - platform: xiaomi_cgdk2
    mac_address: "58:2D:34:12:BC:7C"
    bindkey: "f6d5a0db976d1c758a23b24807e0ff2b"
    temperature:
      name: Server Room Temperature
    humidity:
      name: Server Room Humidity
    battery_level:
      name: Server Room Battery Level
```

e.g.LYWSDCGQ 

```
sensor:
  - platform: xiaomi_lywsdcgq
    mac_address: !secret mijia_ble_temperature_3
    temperature:
      name: "Master Bathroom Temperature"
      device_class: "temperature"
      state_class: "measurement"
    humidity:
      name: "Master Bathroom Humidity"
      device_class: "humidity"
      state_class: "measurement"
    battery_level:
      name: "Master Bathroom Temperature Sensor Battery Level"
```
```

### Anything in the logs that might be useful for us?

```txt
here's the logs from one showing the issue, you'll see a single update from one of the 2 configured sensors, and nothing ever again.
[... 141 lines cut ...]
[06:18:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 355.94000 s with 0 decimals of accuracy
[06:19:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:19:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 415.94000 s with 0 decimals of accuracy
[06:20:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:20:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 475.94000 s with 0 decimals of accuracy
[06:21:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:21:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 535.94000 s with 0 decimals of accuracy
[06:22:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:22:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 595.94299 s with 0 decimals of accuracy
[06:22:40][D][esp32_ble_tracker:245]: Starting scan...
[06:23:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:23:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 655.94000 s with 0 decimals of accuracy
[06:24:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:24:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 715.94397 s with 0 decimals of accuracy
[06:25:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:25:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 775.94000 s with 0 decimals of accuracy
[06:26:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:26:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 835.94202 s with 0 decimals of accuracy
[06:27:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:27:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 895.94000 s with 0 decimals of accuracy
[06:27:40][D][esp32_ble_tracker:245]: Starting scan...
[06:28:15][D][sensor:094]: 'Master Bedroom ESP32 Free Memory': Sending state 75424.00000 B with 1 decimals of accuracy
[06:28:36][D][sensor:094]: 'Master Bedroom ESP32 Ethernet Uptime': Sending state 955.94000 s with 0 decimals of accuracy
```
```

Petro31, 2023-06-30:
Here's a timeline of the events from yesterday until today.

Just a recap if you are unware, there was an issue with supervised installations that caused the DNS and Multicast containers to fail.  This means that none of the containers in a supervised installation would talk to each other.  That was resolved and fixed during the beta by the core team.  They pulled both updates from those containers, fixed the issues, and redeployed a new update.

Prior to that event, I was running 2023.6.2 on my esphome devices however the addon was updated to 2023.6.3.  I had repairs on the dashboard to update the firmware, but I hadn't done that yet.

Around 8am I updated to the beta and ran into the DNS issue.  We were able to find the cause and fix it by ~10am.  After HA was happy and working, I proceeded with all the other pending updates for esphome firmware.

Today I noticed my shower fan was not turning on and I reboot the device.  I've been rebooting randomly and all devices exhibit the same behavior.  They will only capture temperature and humidity on reboot 1 time, never again.  Here's an example:

[image]

Petro31, 2023-06-30:
Here's a typical configuration for one of these devices, Everything outside the sensor section is identical on all the ESP Devices
```
esphome:
  name: master_closet_esp32
  platform: ESP32
  board: mhetesp32devkit

wifi:
  ssid: !secret wifi_iot_server_ssid
  password: !secret wifi_iot_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dining Room Esp32"
    password: !secret wifi_fallback_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

bluetooth_proxy:
  active: true

esp32_ble_tracker:

switch:
  - platform: restart
    name: "Master Bedroom ESP32 Restart"

sensor:
  - platform: xiaomi_lywsdcgq
    mac_address: !secret mijia_ble_temperature_3
    temperature:
      name: "Master Bathroom Temperature"
      device_class: "temperature"
      state_class: "measurement"
    humidity:
      name: "Master Bathroom Humidity"
      device_class: "humidity"
      state_class: "measurement"
    battery_level:
      name: "Master Bathroom Temperature Sensor Battery Level"
      
  - platform: template
    name: Master Bathroom Temperature Sensor Number
    lambda: |-
      return 3;
    update_interval: 86400s
      
  - platform: xiaomi_lywsdcgq
    mac_address: !secret mijia_ble_temperature_5
    temperature:
      name: "Master Bedroom Temperature"
      device_class: "temperature"
      state_class: "measurement"
    humidity:
      name: "Master Bedroom Humidity"
      device_class: "humidity"
      state_class: "measurement"
    battery_level:
      name: "Master Bedroom Temperature Sensor Battery Level"

  - platform: template
    name: Master Bedroom Temperature Sensor Number
    lambda: |-
      return 5;
    update_interval: 86400s

  - platform: uptime
    name: Master Bedroom ESP32 Ethernet Uptime

  - platform: template
    name: Master Bedroom ESP32 Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    unit_of_measurement: 'B'
    state_class: measurement
```

Petro31, 2023-06-30:
Alittle update,

I removed 
```
bluetooth_proxy:
  active: true
```
and everything came alive.

It seems you can't have both proxy and esp32_ble_tracker enabled at the same time anymore.

nebman, 2023-07-06:
Can confirm, I have a set of Xiaomi, PVVX and ATC sensors around the house which all stopped working in the latest esphome version until I disabled the bluetooth_proxy setting

chertvl, 2023-07-08:
Same here. Cant use ```xiaomi_miscale```, ```ble_client``` (for Oclean toothbrush battery) at the same time with ```bluetooth_proxy```.

bdraco, 2023-07-09:
I think https://github.com/esphome/esphome/pull/5073 will fix this but I have no way to test it

closed 2023-07-09 by jesserockz