esphome/issues #4643: I want to know the answer to the same question as #745

tonyxforce, 2023-07-01:
### The problem

How do i get the state of a binary light/output component from lambda?

### Which version of ESPHome has the issue?

2023.6.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

nodemcu with arduino uno as an i2c port expander

### Component causing the issue

-

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-ca4a68
  friendly_name: jelzo
  includes:
    - arduino_port_expander.h

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "not showing here"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-Ca4A68"
    password: "i2jDs4mhLgqL"

captive_portal:
    
i2c:
  sda: GPIO21
  scl: GPIO22
  id: i2c_component

custom_component:
  - id: ape
    lambda: |-
      auto ape_component = new ArduinoPortExpander(i2c_component, 0x08);
      return {ape_component};

output:
- platform: custom
  type: binary
  lambda: |-
    return {
      ape_binary_output(ape, 6),
            };
  outputs:
    - id: led1

light:
  - platform: binary
    name: Led 1
    output: led1
    id: light1

display:
  - platform: lcd_pcf8574
    dimensions: 16x2
    address: 0x27
    lambda: |-
       it.printf(id(light1).state ? "BE" : "ki");
```

### Anything in the logs that might be useful for us?

```txt
error: 'class esphome::light::LightState' has no member named 'state'
```

tonyxforce, 2023-07-01:
#745

ssieb, 2023-07-01:
github issues are not for support requests.  Come ask on the esphome discord server or one of the forums.

closed 2023-07-01 by ssieb