esphome/issues #4644: esp-idf - esp.emac: no mem for receive buffer

jensihnow, 2023-07-01:
### The problem

I'm using a GL-S10 as Bluetooth Proxy and it logs the following:

```
...
[22:28:38][D][api.connection:1030]: Home Assistant 2023.6.3 (xxx.xxx.xxx.xxx): Connected successfully
[22:29:14][D][esp-idf:000]: E (55424) esp.emac: no mem for receive buffer
[22:33:34][D][esp-idf:000]: E (25011) esp.emac: no mem for receive buffer
[22:33:41][D][esp-idf:000]: E (32038) esp.emac: no mem for receive buffer
[22:34:20][D][esp-idf:000]: E (70169) esp.emac: no mem for receive buffer
...
```

This happens right after startup as well as every once in a while. It seems to have no impact, at least I didn't notice anything yet.

### Which version of ESPHome has the issue?

2023.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP8266

### Board

esp32doit-devkit-v1

### Example YAML snippet

```yaml
# Instructions on opening and wiring for flashing on https://blakadder.com/gl-s10
# Source of the config is https://github.com/blakadder/bluetooth-proxies/blob/main/gl-s10.yaml 

substitutions:
  name: "gl-s10-bt-proxy"
  friendly_name: "GL-S10 BT Proxy"

esphome:
  name: ${name}
  friendly_name: "${friendly_name}"
  # turn on Power LED when esphome boots
  on_boot:
    then:
      - output.turn_on: power_led

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Configuration fo V1.x hardware revision
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  domain: '...'

# Adds Ethernet Info
text_sensor:
  - platform: ethernet_info
    ip_address:
      name: ESP IP Address
  - platform: version
    name: "${friendly_name} ESPHome Version"

# Enable logging
logger:

# Enable Webserver
web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: !secret ...
ota:
  password: !secret ...

# Enable BLE tracker
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

# Bluetooth LED blinks when receiving Bluetooth advertising
  on_ble_advertise:
    then:
      - output.turn_on: bluetooth_led
      - delay: 0.5s
      - output.turn_off: bluetooth_led

# Enable BLE Proxy in active mode!
bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

## DEVICE SPECIFIC CONFIGURATION
# network LED (white one) configured as status led
status_led:
  pin:
    number: GPIO32
    inverted: true

# button on the side labeled RESET
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO33
      inverted: true
    name: "Reset Button"

# Bluetooth LED on when connected to Home Assistant
  - platform: status
    name: "${name} Status"
    internal: true
    on_press:
      - output.turn_on: bluetooth_led
    on_release:
      - output.turn_off: bluetooth_led
      
# output settings for LED's marked Power and Bluetooth
# power LED use: see code line 12
# bluetooth LED use: see code line 63
output:
  - platform: gpio
    pin: GPIO14
    inverted: true
    id: power_led
  - platform: gpio
    pin: GPIO12
    inverted: true
    id: bluetooth_led

# since these pins are broken out inside and labeled as I2C pins they're configured here
#i2c:
#  sda: 15
#  scl: 13
# scan: true
```

### Additional information

I found this one https://github.com/espressif/esp-idf/issues/9594 but I'm not sure if this is related.

mihsu81, 2023-07-10:
I'm having the same issue and also with the GL-S10/

jensihnow, 2023-07-10:
Do you also have the following messages being logged?

```
2023-07-10 23:08:43.163 WARNING (MainThread) [aioesphomeapi.connection] gl-s10-bt-proxy @ xxx.xxx.xxx.xxx: Connection error occurred: [Errno 104] Connection reset by peer
```

Seems to happen with equal frequency.

mihsu81, 2023-07-13:
> Do you also have the following messages being logged?
> 
> ```
> 2023-07-10 23:08:43.163 WARNING (MainThread) [aioesphomeapi.connection] gl-s10-bt-proxy @ xxx.xxx.xxx.xxx: Connection error occurred: [Errno 104] Connection reset by peer
> ```
> 
> Seems to happen with equal frequency.

No, I don't. The esp.emac errors didn't always break the connection in my case.

What solved the issue for me was replacing my Asus RT-AX88U router with a Asus RT-AX88U Pro.
The RT-AX88U was causing a lot of extra traffic in my network and breaking the connection of other devices as well. Not even factory resetting the previous router solved it.
Even adding it as an AI Mesh node or Media Bridge was destabilizing the network.
My suggestion is to also look at your network setup and see what could cause the issue.