esphome/issues #4645: E1.31 not responding anymore after update to V2023.6.0

daandav, 2023-07-02:
### The problem

I'm using E1.31 on WLAN connected ESP8266 for addressable led strip with Hyperion NG. Updating ESPHome to V2023.6.0 breaks the E1.31 integration, probably by ["Migrate e131 to use socket instead of WiFiUDP arduino library"](https://github.com/esphome/esphome/pull/4832#issue-1710788306).

### Which version of ESPHome has the issue?

2023.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

na

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

E1.31

### Example YAML snippet

```yaml
#problem can be fixed/introduced by adding/removing in V2023.6.0:
external_components:
  - source: github://esphome/esphome@2023.5.5
    components: e131
```

### Anything in the logs that might be useful for us?

```txt
don't see any error in logs.
```

plvoy-de, 2023-07-02:
Im on ESP32's and this change breaks all my e131 devices with HyperionNG and OpenRGB.
I've used neopixelbus and i tried moving one device to esp32_rmt_led_strip with no success. normal effects work, hassio integration works, but the e131 effect does nothing since 2023.6.0
I've tried multicast/unicast variants without success.
no apparent errors during compiling or later via debug.
I can confirm reverting to 2023.5.5 via external_components fixes the lights

daandav, 2023-07-02:
@jesserockz I think you added the changes in v2023.6.0. Could you please have a look at the issue when you have time? :)

plvoy-de, 2023-07-27:
@jesserockz if it helps; I did some digging. It seems the UDP Port is not being opened.
Wireshark reports Destination unreachable / Port unreachable

relevant code is unchanged, commented out external components do not work, getting 2023.5.5 version in, Port Unreachable disappears and lights work.
```
#external_components:
# - source: github://esphome/esphome@2023.5.5
#  components: e131

e131:
  method: multicast

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    chipset: ws2812
    pin: GPIO32
    num_leds: 20
    name: Display
    restore_mode: RESTORE_DEFAULT_OFF
    rmt_channel: 0
    effects:
      - addressable_rainbow:
      - e131:
          name: Display
          universe: 140
          channels: RGB
```
Log / Debug Level:
```
[19:29:58][D][light:036]: 'Display' Setting:
[19:29:58][D][light:085]:   Transition length: 1.0s
[19:29:58][D][light:091]:   Effect: 'None'
[19:29:58][D][e131:099]: Unregistering 'Display' for universes 140-140.
[19:29:58][D][e131:110]: Left 140 universe for E1.31.
[19:30:00][D][light:036]: 'Display' Setting:
[19:30:01][D][light:109]:   Effect: 'Display'
[19:30:01][D][e131:084]: Registering 'Display' for universes 140-140.
[19:30:01][D][e131:092]: Joined 140 universe for E1.31.
``` 

Hope this helps a bit debugging this.

popy2k14, 2023-08-19:
+1 E1.31 effect doesnt work anymore with Hyperion.NG. Nothing changed on the hyprion side. So it seems i am facing the same esphome issue.

A fix would be great :-)

hcoohb, 2023-08-20:
Same problem here after upgrading esphome.
I use neopixelbus with e131.

Thanks @daandav for posting the workaround of using the e131 component from version 2023.5.5

top4ek, 2023-09-22:
Thanks for workaround. Same configuration with hyperion.

plvoy-de, 2023-10-03:
I had to step back one more version now, since   - source: github://esphome/esphome@2023.5.5 does not seem to download anymore, changed to   - source: github://esphome/esphome@2023.5.4 for the latest round of 2023.9.3 Updates.
This gets worriesome.

rraboy, 2023-10-07:
> I had to step back one more version now, since - source: github://esphome/esphome@2023.5.5 does not seem to download anymore, changed to - source: github://esphome/esphome@2023.5.4 for the latest round of 2023.9.3 Updates. This gets worriesome.

This works for me 
```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2023.5.5
    components: e131
```

closed 2023-10-10 by jesserockz

fabienval, 2023-10-19:
I am still having the same problem with esphome 2023.10.1 on esp8266, so I am not sure that this PR truly fixed this issue...

daandav, 2023-10-19:
> I am still having the same problem with esphome 2023.10.1 on esp8266, so I am not sure that this PR truly fixed this issue...

Think currently this is in the dev branch to be released later. But not sure though.

daandav, 2023-10-27:
Seems like also compiling with dev branch, E1.31 component is still not working. 

The earlier [fix](https://github.com/esphome/issues/issues/4645#issuecomment-1751850516) is now throwing a compile error.

`INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/led-plafond.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing led-plafond (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
Compiling .pioenvs/led-plafond/src/esphome/components/e131/e131.cpp.o
Compiling .pioenvs/led-plafond/src/esphome/components/e131/e131_addressable_light_effect.cpp.o
Compiling .pioenvs/led-plafond/src/esphome/components/e131/e131_packet.cpp.o
src/esphome/components/e131/e131_packet.cpp: In member function 'bool esphome::e131::E131Component::join_igmp_groups_()':
src/esphome/components/e131/e131_packet.cpp:72:34: error: invalid 'static_cast' from type 'esphome::network::IPAddress' to type 'uint32_t' {aka 'unsigned int'}
   72 |     ip4_addr_t multicast_addr = {static_cast<uint32_t>(
      |                                  ^~~~~~~~~~~~~~~~~~~~~~
   73 |         network::IPAddress(239, 255, ((universe.first >> 8) & 0xff), ((universe.first >> 0) & 0xff)))};
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/e131/e131_packet.cpp: In member function 'void esphome::e131::E131Component::leave_(int)':
src/esphome/components/e131/e131_packet.cpp:107:9: error: invalid 'static_cast' from type 'esphome::network::IPAddress' to type 'uint32_t' {aka 'unsigned int'}
  107 |         static_cast<uint32_t>(network::IPAddress(239, 255, ((universe >> 8) & 0xff), ((universe >> 0) & 0xff)))};
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/led-plafond/src/main.cpp.o
*** [.pioenvs/led-plafond/src/esphome/components/e131/e131_packet.cpp.o] Error 1
========================== [FAILED] Took 5.20 seconds ==========================`

@jesserockz / @justdaniel-gh could you please check.

justdaniel-gh, 2023-10-27:
`ip4_addr_t multicast_addr = {static_cast<uint32_t>(` was removed in https://github.com/esphome/esphome/pull/5252 and is marked as a Breaking Change. You probably cannot use that "fix" any more.  The latest dev fixes your issue though.

Make sure you don't have that `external_component` with 2023.5.5 still in there.  Everything should compile with latest dev or 2023.10.3

daandav, 2023-10-27:
> `ip4_addr_t multicast_addr = {static_cast<uint32_t>(` was removed in [esphome/esphome#5252](https://github.com/esphome/esphome/pull/5252) and is marked as a Breaking Change. You probably cannot use that "fix" any more. The latest dev fixes your issue though.
> 
> Make sure you don't have that `external_component` with 2023.5.5 still in there. Everything should compile, and have e131 working, with latest dev or 2023.10.3

Yes, so 2023.10.3 is compiling (without 2023.5.5 external components) but problem still exists. Like @fabienval also indicated already ([link](https://github.com/esphome/issues/issues/4645#issuecomment-1769787331).

justdaniel-gh, 2023-10-27:
I will say, I only tested unicast. I'm not sure if multicast is working or not as I don't use it.  I'm watching my lights blink away right now with 2023.10.3 and e131 in unicast mode.

```
e131:
  method: unicast
```

hcoohb, 2023-10-28:
I have HA 2023.10.5 and esphome 2023.10.3 on a esp8266 (controlled by hyperion-ng)

I tried e131 with unicast but still not working...

(Everything worked fine on esphome 2023.5.5 with multicast)

@justdaniel-gh Is that because you tried on esp32 vs esp8266? How can I provide some more useful logs to find out what is going on?

daandav, 2023-10-28:
Same here, not working whit unicast nor multicast.

My full config:

```
esphome:
  name: led-plafond
  platform: esp8266
  board: nodemcuv2

wifi:
  networks:
    - ssid: xxxxxxxxxx
      password: yyyyyyyyy

# Enable logging
logger:
#  level: info

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

e131:
  method: unicast
  
light:
  - platform: neopixelbus
    type: RGB
    pin: GPIO4
    variant: WS2812
    num_leds: 120
    id: "led_plafond"
    name: "Led Plafond"
    effects:
      - e131:
          universe: 1
          channels: RGB
      - addressable_rainbow:
          name: rainbow_effect
          speed: 50
          width: 50
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:

          
# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: Plafond_Led ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: Plafond_Led IP
    ssid:
      name: Plafond_Led SSID
#    bssid:
#      name: Plafond_Led BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Plafond_Led Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: Plafond_Led WiFi Signal
    update_interval: 60s
```
    
I guess we need to reopen this issue? @jesserockz

plvoy-de, 2023-10-28:
Mine works again as well.
For ESP32 it seems rmt_led_strip is preferred now, so i went with this config:

e131:
  method: multicast

```
light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    chipset: WS2812
    rmt_channel: 0
    pin: GPIO32
    num_leds: 99
    name: "SomeLight"
    restore_mode: RESTORE_DEFAULT_OFF
    effects:
      - addressable_rainbow:
      - e131:
          name: Hyperion.NG
          universe: 1
          channels: RGB
```

many thanks for taking care of it to all involved :)

mhentschke, 2023-11-14:
Could this have been fixed only for ESP32 maybe? I'm still seeing this issue with both dev and beta branches as well as 2023.10.6

This is my full config:

`esphome:
  name: backlight
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "SSID"
  password: "pass"
  use_address: XXX.XXX.XXX.XXX

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "SSID"
    password: "pass"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  password: "pass"

mqtt:
  broker: XXX.XXX.XXX.XXX
  username: "username"
  password: "pass"
  
wled:
e131:
  method: unicast

light:
  - platform: neopixelbus
    name: Ultrawide_backlight
    pin: 2
    num_leds: 49
    variant: WS2812
    effects:
      - addressable_color_wipe:
      - addressable_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 1.5%
      - random:
          name: "Slow Random"
          transition_length: 30s
          update_interval: 30s
      - random:
          name: "Fast Random"
          transition_length: 2s
          update_interval: 2s
      - addressable_rainbow:
      - wled:
          port: 21324
      - e131:
          name: "Bias"
          universe: 1
          channels: RGB
      - e131:
          name: "Sound"
          universe: 2
          channels: RGB`
          

I also see nothing related to the e131 component in the logs.

daandav, 2023-11-15:
Hi @mhentschke,

indeed seems to be a fix only on ESP32 platform. I requested to reopen the issue. I see you created a new one. Thanks!