esphome/issues #4647: Incorrect work swing mode in the hair component with smartair2 protocol.

iturtikov, 2023-07-02:
### The problem

AC model: haier AS12NS--ERA-W + 4U30HS3ERA 
Protocol: smartair2
Now in the component swing mode determinate in 25th byte and 29th byte. But in my model swing mode determinate only  in 25th byte. 
0 - CLIMATE_SWING_OFF
1 - CLIMATE_SWING_VERTICAL
2 - CLIMATE_SWING_HORIZONTAL
3 - CLIMATE_SWING_BOTH
Changing the mode in 29th byte breaks the further operation of the component. 

### Which version of ESPHome has the issue?

2023.6.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.3

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

haier

### Anything in the logs that might be useful for us?

```txt
change mode with remote control

CLIMATE_SWING_OFF
[02:03:41][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[02:03:41][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 15 00 38 00 7F 00 00 00 00 00 01 00 03 00 00 00 01 00 20 00 00 00 06
[02:03:41][D][haier.climate:429]: HVAC Mode = 0x1
[02:03:41][D][haier.climate:431]: Fan speed Status = 0x3
[02:03:41][D][haier.climate:433]: Swing Status = 0x0
[02:03:41][D][haier.climate:435]: Set Point Status = 0x6

CLIMATE_SWING_HORIZONTAL
[02:03:46][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[02:03:46][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 15 00 38 00 7F 00 00 00 00 00 01 00 03 00 02 00 01 00 20 00 00 00 06
[02:03:46][D][climate:378]: 'Haier AC' - Sending state:
[02:03:46][D][climate:381]:   Mode: COOL
[02:03:46][D][climate:386]:   Fan Mode: AUTO
[02:03:46][D][climate:392]:   Preset: NONE
[02:03:46][D][climate:398]:   Swing Mode: HORIZONTAL
[02:03:46][D][climate:401]:   Current Temperature: 21.00°C
[02:03:46][D][climate:407]:   Target Temperature: 22.00°C
[02:03:46][I][haier.climate:427]: HVAC values changed
[02:03:46][I][haier.climate:429]: HVAC Mode = 0x1
[02:03:46][I][haier.climate:431]: Fan speed Status = 0x3
[02:03:46][I][haier.climate:433]: Swing Status = 0x2
[02:03:46][I][haier.climate:435]: Set Point Status = 0x6

CLIMATE_SWING_VERTICAL
[02:04:21][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[02:04:21][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 15 00 38 00 7F 00 00 00 00 00 01 00 03 00 01 00 11 00 20 00 00 00 06
[02:04:21][D][climate:378]: 'Haier AC' - Sending state:
[02:04:21][D][climate:381]:   Mode: COOL
[02:04:21][D][climate:386]:   Fan Mode: AUTO
[02:04:21][D][climate:392]:   Preset: NONE
[02:04:21][D][climate:398]:   Swing Mode: VERTICAL
[02:04:21][D][climate:401]:   Current Temperature: 21.00°C
[02:04:21][D][climate:407]:   Target Temperature: 22.00°C
[02:04:21][I][haier.climate:427]: HVAC values changed
[02:04:21][I][haier.climate:429]: HVAC Mode = 0x1
[02:04:21][I][haier.climate:431]: Fan speed Status = 0x3
[02:04:21][I][haier.climate:433]: Swing Status = 0x1
[02:04:21][I][haier.climate:435]: Set Point Status = 0x6

CLIMATE_SWING_BOTH
[02:04:31][D][haier.protocol:019]: Sending frame: type 01, data: 4D 01
[02:04:31][D][haier.protocol:019]: Frame found: type 02, data: 6D 01 00 15 00 38 00 7F 00 00 00 00 00 01 00 03 00 03 00 11 00 20 00 00 00 06
[02:04:31][D][climate:378]: 'Haier AC' - Sending state:
[02:04:31][D][climate:381]:   Mode: COOL
[02:04:31][D][climate:386]:   Fan Mode: AUTO
[02:04:31][D][climate:392]:   Preset: NONE
[02:04:31][D][climate:398]:   Swing Mode: BOTH
[02:04:31][D][climate:401]:   Current Temperature: 21.00°C
[02:04:31][D][climate:407]:   Target Temperature: 22.00°C
[02:04:31][I][haier.climate:427]: HVAC values changed
[02:04:31][I][haier.climate:429]: HVAC Mode = 0x1
[02:04:31][I][haier.climate:431]: Fan speed Status = 0x3
[02:04:31][I][haier.climate:433]: Swing Status = 0x3
[02:04:31][I][haier.climate:435]: Set Point Status = 0x6
```

### Additional information

My ACs work fine with this patch. May be, exist a lot of versions of protocol smartair2.
[for_AS12NS--ERA-W.patch](https://github.com/esphome/issues/files/11931715/for_AS12NS--ERA-W.patch)

iturtikov, 2023-07-02:
Last version of my Hair stick was R_1.0.00/e_2.5.14

paveldn, 2023-07-03:
Hi @iturtikov ,
Yep, this is something new.  Your packet looks completely the same as other packets that I saw except for this strange swing pode byte. I am curious if it is possible to see the difference in answers. If you are willing to help run a couple of experiments please contact me by telegram (I have a feeling that you have an account there :) ) https://pavlodudnytskyi.t.me/