esphome/issues #4650: Problem with IR climate on Whirlpool (Galactic) IR remote control DG11J1-91

TokarevSergey, 2023-07-03:
### The problem

Good afternoon. 
I use a Whirlpool (Galactic) air conditioner with an IR remote control DG11J1-91
[image]

and universal IR remote control Smart9 S08 based on esp8266 [image]

During the first connection and setup, a device malfunction was detected. Which leads to the impossibility of creating any automation and his work in this form.
Namely:
if in ESPHome firmware in ****.yaml delete the line (receiver_id:)

[image]

in this case, when controlling a Whirlpool (Galactic) air conditioner with ESPHome climate integration, everything works well. Exactly until the moment the air conditioner is turned on from the native IR remote control DG11J1-91. After that, if any button for selecting the air conditioner operation mode is pressed in the ESPHome integration, it is physically turned off, and the shutdown button is inverted.

### Which version of ESPHome has the issue?

ESPHome Current version: 2023.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.6.3 Supervisor 2023.06.4 Operating System 10.3

### What platform are you using?

ESP8266

### Board

Universal IR remote control Smart9 S08 based on esp8266

### Component causing the issue

ESPHome IR Remote Climate (Platform name: whirlpool)

juapem, 2023-07-07:
I do have the same problem and I have not been able to solve it yet

TokarevSergey, 2023-08-05:
> у меня такая же проблема и я пока не могу ее решить

Maybe you know how to reach the esp home whirlpool integration developer?

ElyOshri, 2023-08-11:
I also have this issue and I've been able to solve it by just turning off the receiver while I'm transmitting something.
To turn off the receiver I just modified `whirlpool.cpp`, what I did was basically get the time when the transmission starts and if its been less then 500ms the receiving functions just returns. (you might be able to get away with less than 500ms but I didn't want to risk it)
 
```
// defining the time as a global variable
long int t1 = millis() + 500; 
```
```
void WhirlpoolClimate::transmit_state() {
  t1 = millis(); // starting the timer
  ...
}
```
```
bool WhirlpoolClimate::on_receive(remote_base::RemoteReceiveData data) {
  // Check if the esp isn't transmitting the same data
  if(millis() - t1 < 500 ){
    ESP_LOGV(TAG, "Blocked receive because of current trasmission");
    return false;
   }
  ...
}
```
If you want I can upload the full custom component, I just modified it a bit so it would work with my remote (DG11J1-04) which is not supported

TokarevSergey, 2023-08-12:
> У меня также есть эта проблема, и я смог решить ее, просто выключив приемник, пока я что-то передаю. Чтобы выключить приемник, который я только что изменил `whirlpool.cpp`, я в основном получил время начала передачи, и если оно было меньше 500 мс, функции приема просто возвращаются. (возможно, вам удастся обойтись менее чем 500 мс, но я не хотел рисковать)
> 
> ```
> // defining the time as a global variable
> long int t1 = millis() + 500; 
> ```
> 
> ```
> void WhirlpoolClimate::transmit_state() {
>   t1 = millis(); // starting the timer
>   ...
> }
> ```
> 
> ```
> bool WhirlpoolClimate::on_receive(remote_base::RemoteReceiveData data) {
>   // Check if the esp isn't transmitting the same data
>   if(millis() - t1 < 500 ){
>     ESP_LOGV(TAG, "Blocked receive because of current trasmission");
>     return false;
>    }
>   ...
> }
> ```
> 
> Если вы хотите, я могу загрузить полный пользовательский компонент, я просто немного изменил его, чтобы он работал с моим пультом (DG11J1-04), который не поддерживается.

I am very grateful for your help.
 Please reset all code and please tell me to insert it into esphome firmware
[image]
[image]

ElyOshri, 2023-08-12:
In order for you to add the custom component you will have to insert the following folder into `/config/esphome/my_components/`
[whirlpool_updated.zip](https://github.com/esphome/issues/files/12327395/whirlpool_updated.zip) (do not insert it as a zip, only insert it as a folder `whirlpool_updated`)

After you add the custom component you will have to edit the `.yaml` configuration by changing the climate platform to `whirlpool_updated` and adding this code:

```
external_components:
  - source: 
      type: local
      path: my_components
    components: [ whirlpool_updated ]
```
The climate section should look like this:
```
climate:
  - platform: whirlpool_updated
    name: "Bedroom AC"
    receiver_id: rcvr
    model: "DG11J1-04"
    sensor: temp_sensor
```

juapem, 2023-09-08:
@ElyOshri your fix worked like charm! I was thinking on doing exactly the same but had no idea how to implement it. I think this should be included in the main package of ESP @jesserockz 

This has been a common issue in whirlpool and Midea

TokarevSergey, 2023-09-08:
> @ElyOshriваше исправление сработало как шарм! Я думал сделать то же самое, но понятия не имел, как это реализовать. Я думаю, это должно быть включено в основной пакет ESP.@jesserockz
> 
> Это обычная проблема в Whirlpool и Midea.

If only it were possible. :(

the-mentor, 2023-09-28:
@ElyOshri what would it take to upstream this fix?

the-mentor, 2023-09-28:
@TokarevSergey  quick question. 
how did you get 5v inside the AC unit to mount the ir blaster?

TokarevSergey, 2023-09-28:
> @TokarevSergey быстрый вопрос. как вам удалось получить 5 В внутри блока переменного тока для установки ИК-бластера?

[image]
Look in the photo there is a 3.3 volt input, I did this, measured the power and ran the wires there.

the-mentor, 2023-09-28:
@TokarevSergey i think i have the same unit
mind sending me a picture of where you pulled 5v from ?

thanks!

TokarevSergey, 2023-09-28:
> @TokarevSergeyДумаю, у меня такое же мнение, и я прислал мне фотографию того места, откуда ты выдернул 5 В?
> 
> Спасибо!
> 
> PS - Я также подал PR на основе изменений, которые@ElyOshriпредложил надеяться, что это будет распространено и каждый сможет извлечь выгоду из исправления.

Yes, of course, only if possible in the evening, I'm busy right now.

TokarevSergey, 2023-09-28:
> @TokarevSergeyДумаю, у меня такое же мнение, и я прислал мне фотографию того места, откуда ты выдернул 5 В?
> 
> Спасибо!
> 
> PS - Я также подал PR на основе изменений, которые@ElyOshriпредложил надеяться, что это будет распространено и каждый сможет извлечь выгоду из исправления.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

the-mentor, 2023-09-29:
A fix for this was up streamed and should be available in the next release of esphome (2023.9.2)

glmnet, 2023-09-29:
Hi, I just see this as I got @ mentioned in the PR, I was the original contributor to this component, and at the time it was working properly.
I wonder how come now disabling RX during TX is required, I just tested my AC and it is not working (2023.9.0) (not sure how I didn't notice before, but I'm not used to keep my nodes updated, however I updated this recently) 
At first I though this was a problem with ESP32, but then I see you're also having an issue with ESP8266 too, the reason for this reasoning was that I was expecting the remote receiver to be disabled while the remote transmitter was taking place, this used to be the case as the ESP8266 does bit bang the output to generate the 38khz signal so having interrupts enabled could cause inaccurate 38khz signals.

Anyway, did anyone find this working fine until some version of ESPHome?

In any case, a transmit shouldn't cause a receive for any protocol at all, while this might not be a problem for many protocols, it can cause other unexpected behavior and odd log messaging, etc.

glmnet, 2023-09-29:
I found the regression https://github.com/esphome/esphome/pull/2476/files
@CarlosGS changed it so the timing is better, however he though the InterruptLock was no longer necessary. I would like to bring it back if you guys agree, a slight interrupt can cause the whole IR transmission to fail.

For ESP32 (with hw transmitter) I believe disabling remote receiver (at a base level) while transmitting is a much better idea

the-mentor, 2023-09-29:
Hi @glmnet looks like the PR I submitted got merged.
I read what you wrote but I'm not sure if you've suggested a better solution that will work for both esp8266 and esp32 but if you do I'm down to trying it.

glmnet, 2023-09-29:
Your solution should work for ESP32 too (in case it was affected, did not try) 
For ESP8266 this worked before as interrupts were disabled while transmitting, since this is no longer the case this problem showed up here. I propose to disable interrupts again, this should make the transmitter even more reliable and as a bonus mute the receiver.

To test this approach, you have to add `InterruptLock lock;` as it was in the code before the PR I mentioned in the previous post.

Receiving IR or RF signals while transmitting signals might have some useful cases only when the media is different, i.e. different frequencies in RF or different rooms in IR (light)

I guess `remote_transmitter` could have a `mute: ` property to mute remote receivers, this could address more scenarios, in case somebody would like to deal with that.

CarlosGS, 2023-09-29:
@glmnet I see! Though ESP8266's implementation should still be robust in the presence of normal interrupts (checked the ESPHome codebase at the time and there were no long interrupts, and TX adapts to any variations https://github.com/esphome/esphome/pull/2476#issuecomment-952188362), I still don't think the interrupt lock is necessary.
Instead the separate `mute:` parameter would be a great option for more flexibility, will keep an eye for other TX/RX issues where it could help.

But yeah this particular one should now be solved for both platforms, nice debugging job all :+1:

TokarevSergey, 2023-10-03:
Now I don’t have time, I’ll check everything and close it.  Thank you.

TokarevSergey, 2023-10-08:
> @TokarevSergeyвы можете закрыть эту проблему, поскольку исправление теперь доступно в общедоступной версии esphome 2023.9.2.

At this stage the situation is like this.

If the air conditioner is turned off.
and press "SMART" on the remote control
[image]

Then the air conditioner turns on
[image]

And in the Home Assistant integration the following happens: the temperature is set to 23 degrees and the “Heating/Cooling” mode does not turn on.
[image]

after clicking any of these buttons
[image]

Then the air conditioner turns off
[image]

And what happens when you integrate Home Assistant is that it turns on.
[image]

And if you continue to press the buttons on the integration board, the mode changes, but the air conditioner does not turn on until you turn it on from the remote control.
[image]

Please help me figure it out. Below is my configuration, maybe I configured something wrong?

esphome:
  name: s08ir
  platform: ESP8266
  board: esp_wroom_02

wifi:
  ssid: "DomWiFi"
  password: "Dom1WiFi2"
  fast_connect: true

logger:
  level: DEBUG

api:
  password: "s08ir"

ota:
  password: "s08ir"

status_led:
  pin:
    number: GPIO4

remote_receiver:
  id: "s08_receiver"
  pin: 
    number: GPIO5
    inverted: true
  dump: all

remote_transmitter:
  id: "s08_transmitter"
  pin: 
    number: GPIO14
    inverted: false
  carrier_duty_percent: 55%
  

binary_sensor:
  - platform: gpio
    name: "s08_button"
    id: "s08_button"
    pin:
      number: GPIO13
      inverted: true
    internal: true

sensor:
  - platform: homeassistant
    id: current_temperature
    entity_id: sensor.ble_temperature_e72e0172f902
    device_class: temperature
    state_class: measurement

climate:
  - platform: whirlpool
    name: "Кондиционер спальня"
    model: DG11J1-91
    id: "konditsioner_spalnya"
    sensor: current_temperature
    receiver_id: "s08_receiver"
    transmitter_id: "s08_transmitter"
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C

TokarevSergey, 2023-10-08:
> @TokarevSergey you can close this issue since the fix is now available in the public release of esphome 2023.9.2

please help

prokudin07, 2023-10-19:
> At this stage the situation is like this.
>
> If the air conditioner is turned off. and press "SMART" on the remote control [image]
>
> Then the air conditioner turns on [image]
>
> And in the Home Assistant [...]

I confirm the problem.

prokudin07, 2023-10-20:
@TokarevSergey Думаю, что надо закрыть это issue и открыть новое.
То, что на скрине и ниже - написать новое, так как ту проблему решили, а новую создали
и упомяни меня в тексте, поддержу.
[image]

TokarevSergey, 2023-10-21:
Thank you.  For your understanding and help.  Enter how this problem was solved.  But another problem came up, I described it in this topic #5014

closed 2023-10-21 by TokarevSergey